dresden-elektronik/deconz-rest-plugin #6978: Device name

jmedemar, 2023-05-15:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

BabaIsYou, 2023-05-15:
Can you please make a "read" before taking screenshots on each cluster ?

jmedemar, 2023-05-16:
> Can you please make a "read" before taking screenshots on each cluster ?

I'm sorry, I did press read on each attributes screen before the screenshots. It's a battery powered sensor, so I also was pushing the button to wake it. 

It only reports these readings: Battery Percentage value of 200 (100% (value of 200/2)), Temperature Measured Value of 2260 (22.6 degrees C (value of 2260/100)), and Humidity Measured Value of 4400 (44% (value of 4400/100)).

Smanar, 2023-05-16:
Hello can you try this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"Third Reality, Inc",
   "modelid":"3RTHS24BZ",
   "vendor":"ThirdReality",
   "product":"ThirdReality multisensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_TEMPERATURE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0402"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "awake":true,
               "refresh.interval":3600,
               "read":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "fn":"zcl"
               },
               "parse":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "eval":"Item.val = Attr.val / 2"
               },
               "default":0
            },
            {
               "name":"config/offset",
               "default":0
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/temperature"
            }
         ]
      },
      {
         "type":"$TYPE_HUMIDITY_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0405"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "awake":true,
               "refresh.interval":3600,
               "read":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "fn":"zcl"
               },
               "parse":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "eval":"Item.val = Attr.val / 2"
               },
               "default":0
            },
            {
               "name":"config/offset",
               "default":0
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/humidity"
            },
            {
               "name":"state/lastupdated"
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "dst.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":3600,
               "max":7200,
               "change":"0x00000002"
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "dst.ep":1,
         "cl":"0x0405",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x21",
               "min":5,
               "max":1800,
               "change":"0x00000064"
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0402",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x29",
               "min":5,
               "max":1800,
               "change":"0x0000000A"
            }
         ]
      }
   ]
}
```

jmedemar, 2023-05-16:
Made DDF and put in the folder, rebooted Deconz, then re-added the sensor. Everything seems to be working and updating appropriately.

[image]

closed 2023-05-21 by manup