dresden-elektronik/deconz-rest-plugin #6980: IKEA Styrbar remote control - left and right buttons don't work

adrspa, 2023-05-15:
Setup
ConBee 2 connected to RPi 4 running HA
IKEA Strybar running 2.4.5 FW
Deconz v2.21.2
HA 2023.5.3
Supervisor 2023.04.1
Operating System 10.1

Issue
IKEA Styrbar remote control can be paired in Phoscon and is visible in deconz, but only the dim up and dim down buttons work. The left and right buttons don't generate any events, even directly in deconz. 

[image]

Any help would be greatly appreciated!

Best regards,
Adrian

BabaIsYou, 2023-05-15:
Seem duplicate of #6761

Mimiix, 2023-05-16:
Yep, and template not used.

Closing.

closed 2023-05-16 by Mimiix

adrspa, 2023-05-16:
Hi,

Sorry for not following the template. Wasn't aware there was one as this was my first post in this forum. I'll take a look at #6761, thanks.

Best regards,
Adrian