dresden-elektronik/deconz-rest-plugin #6981: Tuya Smoke Sensor not available in the api

MadMonkey87, 2023-05-16:
## Describe the bug
I have a new tuya smoke sensor that seems to be supported by deconz, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6734. I was able to pair it with deconz but it does not appear trough the api. I re-paired it a few times but without success. Clicking on "edit ddf" shows the right ddf file so I guess at least the device was identified correctly. Not also that the name has not changed so I assume something went wrong

## Steps to reproduce the behavior

  1. Go to sensor view in phoscon
  2. Click on 'add sensor', choose 'other'
  3. Press and hold the button on the device until it beeps
  4. A further beep signals that the device has connected to deconz
  5. The phoscon dialog will timeout, saing that the pairing was not successful
  6. Check the deconz Ui and find a new node
  7. Check the sensors api, no new entry

## Expected behavior
 - The name of the device should change to "Photoelectric Smoke sensor"
 - The device should appear as sensor in the api

## Screenshots
[image]
Basic Cluster:
[image]

## Environment

 - Host system: PC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26400500
 - deCONZ version: 2.21.02 / 19.9.2022
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

Smanar, 2023-05-16:
The DDF is 1 month old https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_m9skfctm_smoke_detector.json

You can check if you have it, but it's possible this one was only on recent deconz version.

Mimiix, 2023-05-16:
Hi,

This is clearly not a bug. 

For any furthur help, please use the forums.

Closing.

closed 2023-05-16 by Mimiix