dresden-elektronik/deconz-rest-plugin #6993: Device name

duppe, 2023-05-22:
Hi.

Dont know if this is the correct way but 

## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: Nedis.
- Model identifier: ZBWS10WT
- Device type :  Switch

## Screenshots

BabaIsYou, 2023-05-22:
Hi,

Missing some clusters screenshots.

duppe, 2023-05-22:
[image]
[image]

duppe, 2023-05-22:
Hi.
Call me stupid, but cannot get it working :-)

duppe, 2023-05-22:
Yes, and I tried many things  ... but buttonevent always stays null  :-/ 

{"config":{"battery":80,"on":true,"reachable":true},"etag":"25050342e43c5919486d981a281aa27d","lastannounced":null,"lastseen":"2023-05-22T19:08Z","manufacturername":"_TZ3000_wqcbzbae","mode":1,"modelid":"TS0041","name":"Insiden","state":{"buttonevent":null,"lastupdated":"none","lowbattery":null},"type":"ZHASwitch","uniqueid":"a4:c1:38:f7:86:7b:fb:7d-01-0006"}

BabaIsYou, 2023-05-22:
Well, then it's recognized by DeCONZ with the DDF provided ? That was the first step ;-)

For button you have to modifiy the file `/usr/share/deCONZ/devices/button_maps.json`
and change  
```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu"],
```
by (add your device at the end of the list)

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae"],
```

duppe, 2023-05-23:
ok, sorry.... I just used the things @BabaIsYou provided, and it worked out of the box  :-)

PR #7006 (2023-05-25): DDF add Nedis ZBWS10WT switch
Close #6993
closed 2023-05-25 by manup