dresden-elektronik/deconz-rest-plugin #6995: Add LIDL SilverCrest Plug with Energy Counter

kabieror, 2023-05-22:
## Device

- Product name: SilverCrest Plug
- Manufacturer: _TZ3000_j1v25l17
- Model identifier: TS011F
- Device type : Please remove all unrelated device types. 
  - Plug with energy counter

The device is already supported for switching, but the energy measurement is not available in the REST interface.

## Screenshots

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]

### On/Off
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]

BabaIsYou, 2023-05-22:
Can you try this DDF ? (see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet for help)

{
  "schema": "devcap1.schema.json",
  "manufacturername": " _TZ3000_j1v25l17",
  "modelid": "TS011F", 
  "product": "LIDL SilverCrest Plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "../tuya/tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "../tuya/tuya_swversion.js"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "../tuya/tuya_swversion.js"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {"at": "0x0000", "cl": "0x0702", "ep": 0, "fn": "zcl" },
          "parse": {"at": "0x0000", "cl": "0x0702", "ep": 0, "eval": "Item.val = Attr.val * 10"},
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

kabieror, 2023-05-25:
Thanks for the DDF! I tried placing it at /usr/share/deCONZ/devices, restarted the container and re-paired the plug. But there are no new endpoints in the REST API. I also tried setting DDF mode to "hybrid" which did not change anything. Have I missed anything?

closed 2023-05-25 by manup

BabaIsYou, 2023-05-25:
AFAIK DDF file won't resist at a container restart doing it that way. May be could try some hnts here : https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839 ... or just update Deconz after merge of the PR.

kabieror, 2023-05-25:
Actually it does, until you recreate the container or in case of docker-compose, you run 'down' and 'up' again. Okay I'll try again after the upgrade. Tank you!

BabaIsYou, 2023-05-25:
Better to put it under subfolder`lidl` to have `swversion` correctly parsed. 

Have you tried to load the DDF using DDF editor to see if it kids without error ?

kabieror, 2023-05-26:
Ah sorry, I forgot to mention this. I tried to load the json in die DDF editor via File>Open... Nothing happens, but the footer shows an error message. The log does not contain any helpful information.
[image]

kabieror, 2023-05-26:
Okay my bad. The file seems to have been copied with errors. Now on the second attempt, it worked like a charme. The REST API now has sensor endpoints showing the current power consumption. Thank you for the fast solution!

Pinky007, 2023-08-11:
Hej @BabaIsYou ,
I have the same product but seems that I still don't get the measurement.
I saw https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7013 which seems to be merged on v2.22.1 beta.
I am running v2.22.2 as addon in HA, so I would have expect it to be included
[image]

Device has been added under lights in Phoscon
[image]

But just offer the switch in HA
[image]

I tried to check the different cluster info but outside of minor difference that I don't think it make a big change, I couldn't see why it wouldn't work

Basic
[image]

identify
[image]

groups
[image]

scenes
[image]

on/off
[image]

simple metering
[image]

Tuya specific
[image]

There is some small DDF differences with yours posted in this thread (marked in red):
[image]
[image]
[image]
[image]
[image]

I also let the plug run sometime so that the values where not 0 and where moving (to avoid missing report on only changed value) but this didn't helped.

Let me know if you have any pointer of where this could be coming from or if further data is needed to adapt the integration.

Thanks!

BabaIsYou, 2023-08-11:
The DDF is in « Silver » state … did you change your deconz configuration accordingly ?

Pinky007, 2023-08-11:
That was it, thank you for the Hint.
I saw that it was silver in the DDF:
[image]

But didn't know about enabling the different levels activation mentioned in your link.

Is there a way to bump the integration to gold? Can I help with this?

Thank you for the quick help!

PR #7013 (2023-05-25): DDF HG08673 LIDL SilverCrest smart plug with monitoring
Fix #6995