dresden-elektronik/deconz-rest-plugin #6996: EGLO Crosslink 2 Motion Sensor

jeceka201, 2023-05-22:
A IP44 motion sensor sold under the EGLO Crosslink 2 brand.  

## Device

- Product name: EGLO crosslink 2 motion sensor.
- Manufacturer: Awox.
- Model identifier: TLSR82xx.
- Device type : 
  - Sensor

## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
[image]
  - Node Info panel
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Other clusters that are not mentioned above
[image]

jeceka201, 2023-05-26:
I tried to Frankestein a DDF but my knowledge is lacking. So no luck. Any help?

BabaIsYou, 2023-05-29:
It seems that quitte all Eglo devices use the same manufacture name and modelid ... Then even if we could make a DDF ,  that one would be loaded for all other Eglo devices :-/

jeceka201, 2023-05-29:
> It seems that quitte all Eglo devices use the same manufacture name and modelid ... Then even if we could make a DDF , that one would be loaded for all other Eglo devices :-/

That's unfortunate. And a good reason not to buy any more of their products. If I wanted to make my own DDF, where do I look for al the right values? Or do I need to sniff the Zigbee network to figure out how a device really communicates?

BabaIsYou, 2023-05-29:
As far as I can see, this device seesm not to provide any information about occupancy on standard Zigbee cluster. It seems to provided them using manufacturer specific clusters 0xFF50 and 0xFF551 on Endpoint 0x03. And at this time I didn't find any information about these clusters.

closed 2023-06-28 by github-actions[bot]