dresden-elektronik/deconz-rest-plugin #6998: Tuya Zigbee Human Presence Sensor

DieterKoblenz, 2023-05-23:
## Device

- Product name: Tuya Zigbee Human Presence Sensor
- Manufacturer: _tze204_sxm7l9xa
- Model identifier: TS0601
- Device type : 
  - Sensor

## Screenshots

[image]
[image]

### Basic
[image]
### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
Tuya specific
[image]

BabaIsYou, 2023-05-23:
Another Tuya clone ... another "DPIP set" ... 

104 - Lux
105 - Presence
106 - Motion sensitivity
107 - Max detection distance
108 - Min detection distance
109 - Target distance
110 - Detection delay
111 - Fading time

Could you try this DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_sxm7l9xa",
  "modelid": "TS0601",
  "product": "Tuya Human Presence Detector",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
[... 195 lines cut ...]
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}
```

DieterKoblenz, 2023-05-23:
Thank you. Now this is the first time I am working with DDF's (and it was a pain to do so in the home assistant add-on) but I managed to use it. 

It now recognizes the device, gives it a name, however I don't see any other values in phoscon other than luminance, which is (incorrectly) 0. Is there anything else I need to do? Do I need to restart deconz? In home assistant it's not recognized.

BabaIsYou, 2023-05-23:
Where did you place the DDF file ? As it reference  `../generic/illuminance_cluster/lux_to_lightlevel.js` that could be normal that you don't get any value for luminance. Two way then : 
1. Copy the `lux_to_lightlevel.js` file to `generic/illuminance_cluster` subfolder of where you put your DDF (content is here https://github.com/dresden-elektronik/deconz-rest-plugin/tree/89151afc8f0f0ed01b6cd611f95fb115ea0102a2/devices/generic/illuminance_cluster)

2. You can try to change 
```
        {
          "name": "state/lightlevel"
        },
```
by
```
        {
          "name": "state/lightlevel",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 104,
            "eval": "Item.Val = Math.round(10000 * Math.log10(Attr.val));",
            "fn": "tuya"
          },
```
That way some other attributes values won't be updated, but you can check that all is fine.

DieterKoblenz, 2023-05-24:
I'll try both ways. The docker container used now has only one subfolder where I can access data.

BabaIsYou, 2023-05-24:
Values related to presence sensor seesm to be updated (angle, presence) then I don't know why there are not reproted to HA.

BabaIsYou, 2023-05-31:
I read one more time this thread and think I finally figure out what's happening ... There isnM't any "read" for tuya function in this DDF.
Could you try to change 
```
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
```
by
```
        {
          "name": "state/presence",
          "read": {
            "fn": "tuya"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
```

BabaIsYou, 2023-06-05:
Found another reference where fading time and detection delay are inversed, and unit seems 10th of seconds for both.
[104, 'illuminance_lux', tuya.valueConverter.raw],
[105, 'presence', tuya.valueConverter.trueFalse1],
[106, 'radar_sensitivity', tuya.valueConverter.raw],
[107, 'maximum_range', tuya.valueConverter.divideBy100],
[108, 'minimum_range', tuya.valueConverter.divideBy100],
[109, 'target_distance', tuya.valueConverter.divideBy100],
[110, 'fading_time', tuya.valueConverter.divideBy10],
[111, 'detection_delay', tuya.valueConverter.divideBy10],

You'll have to do some tries.

TC1666, 2023-06-08:
Just bought one of these sensors, updated to deconz  v2.22.1-beta, and unfortunately the sensor is being recognized as smart plug/light.

[image]

Any advice on how to get this working?

BabaIsYou, 2023-06-09:
As this sensor is still not supported (waiting for feedback from users about proposed DDF above) it’s normal. 

You have to copy the DDF into your system to also try it from your side.

wpiman, 2023-06-12:
Ok, moving over from my other thread.  I have never used DDF files but they look like simple JSON files.  

I created a file called _TZE204_sxm7l9xa.json next to my TY_M100 JSON file.  I copied your file above, and then changed that "none" in the read function to "tuya".  Not sure if I needed to buy I rebooted my PI.

I now see the device... but like the other user above, it looks like a light sensor..  I see the light sensor value change when I put my hand over it...

Here is the DDF file I used...  Let me know if my edits to it look ok...  (sorry, I tried putting code comments around it but it really messes up the JSON indenting??)

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_sxm7l9xa",
  "modelid": "TS0601",
  "product": "Tuya Human Presence Detector",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
[... 190 lines cut ...]
        {
          "name": "state/presence",
          "read": {
            **"fn": "tuya"**
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}

[image]

wpiman, 2023-06-12:
Edit: actually that motion does in fact work when I flip the sensor over.

BabaIsYou, 2023-06-12:
Yes i saw on your screenshot lux and motion then it seems that it works. 

And yes this another Tuya « clone » of the first my-100 … and there will be certainly others ;-) That’s Tuya world !

Have you tried to read or modify other attributes via rest-api or third party apps ?

wpiman, 2023-06-12:
So I use something called the JowiHue plugin for Homeseer which talks to the bridge...   

I didn't even know I could change the parameters from it, but I just changed the motion sensitivity from a 7 to an 8.

BabaIsYou, 2023-06-12:
If your tests are significative then I suggest you can create a PR (pull request) to have the DDF included into a next relaease ;-)

kallejoken, 2023-06-28:
@BabaIsYou 
so with me also a plug is still listed under lights. everything else in the sensor also unfortunately does not work as desired
[image]

 {
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_sxm7l9xa",
  "modelid": "TS0601",
  "product": "Tuya Human Presence Detector",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
[... 184 lines cut ...]
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "tuya"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}

BabaIsYou, 2023-06-29:
Have you deleted the device and restarted Deconz ? Without information from Rest-Api it’s difficult to diagnose …

kallejoken, 2023-06-29:
> Have you deleted the device and restarted Deconz ? Without information from Rest-Api it’s difficult to diagnose …

Hello, yes I deleted and re-added the device several times, restarted the deconz addon several times and restarted the complete HA.  

{
     "config": {
         "on": true,
         "reachable": true,
         "tholddark": 12000,
         "tholdoffset": 7000
     },
     "etag": "30b853d89e357ddbd0926779381d1f6f",
     "lastannounced": null,
     "lastseen": "2023-06-29T11:10Z",
     "manufacturername": "_TZE204_sxm7l9xa",
     "modelid": "TS0601",
     "name": "LightLevel 207",
     "state": {
         "dark": false,
         "daylight": false,
         "lastupdated": "none",
         "lightlevel": 0,
         "lux": 0
     },
     "type": "ZHALightLevel",
     "uniqueid": "a4:c1:38:3b:84:c3:53:ce-01-0400"
 }

{
    "config": {
        "delay": 20,
        "duration": 3,
        "on": true,
        "reachable": true,
        "sensitivity": 7,
        "triggerdistance": "6"
    },
    "etag": "ca3ac711b89afe7d315199b28dc80c95",
    "lastannounced": null,
    "lastseen": "2023-06-29T11:13Z",
    "manufacturername": "_TZE204_sxm7l9xa",
    "modelid": "TS0601",
    "name": "Presence 208",
    "state": {
        "angle": 0,
        "lastupdated": "2023-06-29T11:13:17.675",
        "presence": false
    },
    "type": "ZHAPresence",
    "uniqueid": "a4:c1:38:3b:84:c3:53:ce-01-ef00"
}

BabaIsYou, 2023-06-29:
Hmmm I clearly don't understand from where this smart plug come from ! There is nothing as a /lights into the DDF.

1. Into your Phoscon Screenshot the lightlevel sensor is named `LightLevel 205`, but there is no `LightLevel 205` into HA screenshot, neither into REST-API ... Do you have several devices ?
2. Sorry to insist on that point again, but do you have done a delete of the device, a restart of Deconz and a re-inclusion in that order ?
3.  Where did you put your DDF file ? `swversion` should not be `"null"` if the DDF is placed in the right folder. Because `lightlevel`, `swversion`, `dark`,` daylight ` are calculated values by .js files referenced thru relative path from into the DDF, this one has to be placed into the folder where Deconz place all the DDF.

kallejoken, 2023-06-29:
> Hmmm I clearly don't understand from where this smart plug come from ! There is nothing as a /lights into the DDF.
>
> 1. Into your Phoscon Screenshot the lightlevel sensor is named `LightLevel 205`, [...]

I'll get back to you tonight and report, I currently don't have full access to the system.
thank you and see you later

BabaIsYou, 2023-06-29:
The DDF is actually loaded but, as explianed, it's not the "definitive" place where DeConz is storing the released DDF files, it's the custom place where you can put your own DDF files. To have the other values updated you have two choices : 
1. Waiting for the related PR #7051 to be merged in a future release,
2. Copying the two .js files referenced into the DDF to your custom location and changing the reference to these files into the DDF to have them correctly addressed.

kallejoken, 2023-06-29:
> The DDF is actually loaded but, as explianed, it's not the "definitive" place where DeConz is storing the released DDF files, it's the custom place where you can put your own DDF files. To have the other values updated you have two choices :
> 
> 1. Waiting for the related PR [DDF add support for _TZE204_sxm7l9xa mmWave radar sensor #7051](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7051) to be merged in a future release,
> 2. Copying the two .js files referenced into the DDF to your custom location and changing the reference to these files into the DDF to have them correctly addressed.

you mean the following files "tuya_swversion.js" and this "../generic/illuminance_cluster/lux_to_lightlevel.js"? if so where can i find this "tuya_swversion.js" file? they must be with the .json. in which part in the .json do you have to change the path?

andreaburlizzi, 2023-06-30:
Hello guys, i tried the DDF and lux works, angle in the rest api also works but it doesn't switch presence on, no matter the sensitivity parameter i set. Also, i think that delay and duration are inverted (coudn't set delay at less than 20, that should be the minimum possible for duration) and also impossible to change trigger distance (set on 6 by default). any ideas?

BabaIsYou, 2023-06-30:
About delay and duration could invert the dpid as suggested previously https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6998#issuecomment-1577363604

if angle is updated then it could mean that presence has been detected. Don’t understand why presence is not updated accordingly. That’s working for other users.

andreaburlizzi, 2023-06-30:
> About delay and duration could invert the dpid as suggested previously [#6998 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6998#issuecomment-1577363604)
> 
> if angle is updated then it could mean that presence has been detected. Don’t understand why presence is not updated accordingly. That’s working for other users.

thanks for your reply! could it be because i run deconz on Windows? maybe something different in configuration)? also, last version installed

andreaburlizzi, 2023-07-15:
after managing for a while i could make it work, but after the end of duration period the presence goes on false, even when "angle" gives a value different from 0, i could make it true again exiting from the room and entering again, but it's not the purpouse of a presence detection! anyone else had the same problem?

TC1666, 2023-07-15:
I've given up on trying to get this device to work with deconz/ConBeeII. As an experiment I bought a SLZB-06 coordinator and setup ZigBee2MQTT on a RPi, this device added right away with all clusters and controls properly exposed. In fact, I started to add other devices to this and Z2M exposes more cluster and controls for easy user access than deconz. I will say that the ConBeeII still has better transceiver performance.

andreaburlizzi, 2023-07-15:
I'm using the combo deconz- domoticz on a windows server and except for this sensor i am very satisfied with the ease of use of the system... also, original ZY-m100 is very difficult to find! just this clone, that is also very well build (usb c and better case), anyone that was able to make it work correctly on deconz?

zaunermax, 2023-07-18:
> The DDF is actually loaded but, as explianed, it's not the "definitive" place where DeConz is storing the released DDF files, it's the custom place where you can put your own DDF files. To have the [...]

I tried this but to no avail 🤔 The file seems to be loaded when connecting the sensor but deconz still recognizes it as a "smartplug" which is weird. When using VNC I can confirm that the DDF is loaded, but something seems to be off...

You said "copy the files and change the references". Do you know what the current working directory is when those files are loaded? can I just put an absolute path?

Smanar, 2023-07-18:
>but deconz still recognizes it as a "smartplug" which is weird

This will not change with the DDF, the DDF only impact the API, deconz show the reality, and this device is a smartplug from his firmware.

What is your OS, you are using a real OS or a docker ?

If you are using bad path in the DDF, it will only impact swversion and illiminance, all the rest will work.

zaunermax, 2023-07-19:
Thanks for answering 🙌 

I am running deconz in docker and I use it with a third party integration for iobroker. I guess I cannot use it then as the iobroker integration treats every smartplug the same (standard data points), so I have no datapoints for presence there 🤔 

Thanks anyways for helping me 👌

Smanar, 2023-07-19:
On the official deconz docker, custom DDF need to be placed in `opt/deCONZ/devices/`
https://github.com/deconz-community/deconz-docker/issues/163

closed 2023-07-24 by manup

Monofin, 2023-08-14:
FYI: Specifying both config/delay and config/duration together in a DDF will result in the 'motion sensor' behaviour mentioned above, where the motion state is reset after a *fixed* time regardless of further motion.

According to this thread: https://github.com/ebaauw/homebridge-deconz/issues/154

The purposes of these fields are, to quote the thread:

"config.duration is for devices that don’t send restore reports (don’t send a “no more motion detected”) message. There’s no corresponding Zigbee attribute for this item, as deCONZ handles resetting state.presence for these devices.

config.delay has two uses. For “decent” motion sensors (that do report “no more motion detected”), it maps to PIR Occupied to Unoccupied Delay, configuring the delay on the device, after which it sends the “no more motion”. For motion controllers (like the Trådfri), it’s read-only and reports the On Time parameter of the. On with Timed Off command that the controller sends."

I also note that there are several DDFs present in the repo that make this conflation.

I will raise issues on them, as they are causing extremely confusing behaviour when the fields are mis-used for different parameters: They also trigger behaviour in DECONZ itself (like timed resets of motion status) that is indirect and unhelpful.

"status/angle" - I'm looking at that field here too - these sensors report range, not angle.

Monofin, 2023-08-14:
Other DDFs with this config/delay + config/duration as of deCONZ 2.23.0

These will also be experiencing unexpected behaviour:

_TZE200_TS0601_mmWaveRadar.json
_TZE204_sxm7l9xa_mmwave_radar.json

There are also multiple PIR motion sensors that (without needing to) trigger the behaviours associated with 'config/duration'.

BabaIsYou, 2023-08-14:
These fields where "reused" in purpose because that seemed too complicated to ask to add some new fields into C++ code at that time. Obviously the side effect was not wanted, and there is no conflation, at best some confusion, as the use of these fields is descripted into the DDF.

Of course PR #7098 will allow the correction of all that bad workarounds

Smanar, 2023-08-14:
ATM it's possible to create new item dynamically, I puprose to avoid config/duration.
This one is used on c++ code and reserved for device that DON'T return back to false after a detection.

No problem with config/delay.

BabaIsYou, 2023-08-17:
Trying other propositions into #7183 and will adjust DDFs mentionned above if it's conclusive.

Smanar, 2023-08-18:
Yeah I m agree, was just to make test.

PR #7051 (2023-07-24): DDF add support for _TZE204_sxm7l9xa mmWave radar sensor
Fix #6998
PR #7184 (2023-08-31): DDF add support for TS0601_TZE204_qasjif9e & fix other mmWave Radar Presence DDF
Add support for _TZE204_qasjif9e and add two custom items to avoid side effect of reusing `config/duration` for instance [#6998](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6998#issuecomment-1677574695)

Fix #6998  (breaking change)

Close #7183 
Close #6517