dresden-elektronik/deconz-rest-plugin #6999: Issue pairing innr RB 251 C

DieterKoblenz, 2023-05-23:
## Describe the bug
I am trying to pair a previously paired to z2m innr light bulb (rgb), it pairs, however not correctly.

I'm getting (in deCONZ) a hex named device, with all the relevant clusters. I can controle the device from deCONZ, but obviously it won't show up in phoscon/home assistant.

## Steps to reproduce the behavior
1. Reset the light bulb via manufacturer instruction. 
2. Enable permit join.
3. Try to pair the device.

I've also tried to reset the device during the permit join. Same result unfortunately.

## Expected behavior
A functional device.

## Screenshots

## Environment
 - Host system: NAS (Virtual machine)
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26660700
 - deCONZ version: 2.21.2
 - Device: Conbee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes, a zwave stick, almost a meter from the deconz stick

## deCONZ Logs
[image]
[image]
[image]

## Additional context
53 nodes, plenty of routers including dedicated router (zbdongle)

BabaIsYou, 2023-05-23:
I think it's not a bug but a device request for a non yet supported device. Can you please add a screenshot for all clusters (after clicking "read") ?

Mimiix, 2023-05-23:
I am wondering why a light is not supported automatically. Normally, it is.

I'll check with the devs. If they say this is correct behavior, i have to ask you to open an device request for this.

DieterKoblenz, 2023-05-23:
> Not here please :) Lets await the devs first.

Apologies, I responded to BabalsYou message and in the meantime you replied as well ;)

DieterKoblenz, 2023-05-23:
> Hi,
> 
> Can you check in deCONZ under "control" if legacy support is enabled (Basic or either Hybrid).
> 
> Additionally, can you provide some logs with INFO, INFO L2, Error, Error L2, APS, APS L2. Please use pastebin to share :)

I have hybrid support active due to another device I was trying to add. I'll try to provide some logs of the device pairing. However I am unsure how to get those logs from within the HA add-on.

Smanar, 2023-05-23:
Can try with less flag to start, just "info" and "info_l2"

Mimiix, 2023-05-23:
> > Hi,
> > Can you check in deCONZ under "control" if legacy support is enabled (Basic or either Hybrid).
> > Additionally, can you provide some logs with INFO, INFO L2, Error, Error L2, APS, APS L2. Please use pastebin to share :)
> 
> I have hybrid support active due to another device I was trying to add. I'll try to provide some logs of the device pairing. However I am unsure how to get those logs from within the HA add-on.
> 
> Edit: i've tried to get some logs, but it gets spammed so fast I am unable to get the relevant data.

Just uncheck the boxes and it stops :)

Mimiix, 2023-05-23:
> I have hybrid support active due to another device I was trying to add. I'll try to provide some logs of the device pairing. However I am unsure how to get those logs from within the HA add-on.

What if you put it on Basic?

DieterKoblenz, 2023-05-24:
On basic it paired as extendible light, that is nice. So hybrid mode prevents automatic light pairing on unsupported lights?

I've added a INFO/WARN-L2 log here:
https://pastebin.com/fFAnQQRp

Mimiix, 2023-05-24:
It probably created a basic DDF in hybrid mode, which caused this probably. In Basic mode the legacy code is done first and DDF after.

Nevertheless, as this is not a bug but expected behavior i am closing this issue.

closed 2023-05-24 by Mimiix