dresden-elektronik/deconz-rest-plugin #7011: Tongou DIN Rail Circuit Breaker 63A with Power Monitoring

cornim, 2023-05-25:
## Device

- Product name: [Tongou DIN Rail Circuit Breaker 63A with Power Monitoring](https://zigbee.blakadder.com/Tongou_TO-Q-SY1-JZT.html)
- Manufacturer: _TZ3000_qeuvnohg
- Model identifier: TS011F
- Device type :
  - Sensor
  - Switch

## Screenshots

### Node Info
[image]

### Cluster List
[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering
[image]
[image]

### Diagnostics

### Other clusters that are not mentioned above

### Electrical Measurement
[image]
[image]

### Temperature measurement
[image]

### Unknown
<no attributes>

### Tuya specific
[image]

cornim, 2023-05-25:
The DDF seems to work. I get consumption power and temperature as additional endpoints in HA now.

Thanks.

Smanar, 2023-05-25:
>but the power measurement cluster (0x0B04) has 3 report bindings

It's the binded attributes inside this cluster, it's not a zigbee adress.

BabaIsYou, 2023-05-25:
@cornim Do you want to make the pull request (PR) to have this DDF shared into a next release ?

closed 2023-07-15 by github-actions[bot]