dresden-elektronik/deconz-rest-plugin #7014: Tuya Smart ZigBee Energy Meter Single Phase

raduten, 2023-05-28:
## Device

- Product name: Tuya Smart ZigBee Energy Meter Single Phase 80A
- Manufacturer: _TZE200_lsanae15
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Sensor
  - Other: energy/power meter

## Screenshots

### Basic
[image]

[image]

### Groups
[image]

[image]

### Scenes
[image]

[image]

[image]

### Other clusters that are not mentioned above
[image]

[image]

BabaIsYou, 2023-05-28:
Seems duplicate of #6485, also covered by #6530

Mimiix, 2023-05-28:
I agree, how er, I'd like to keep this open as the op never responded there anymore.

BabaIsYou, 2023-05-28:
> I agree, how er, I'd like to keep this open as the op never responded there anymore.

All right ... waiting for DDF beeing tested anyway

BabaIsYou, 2023-05-29:
Could you try this DDF ? i'm not quite sure of what I am doing here regarding to current and voltage ...

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": ["_TZE200_lsanae15", "_TZE204_lsanae15"],
	"modelid": ["TS0601", "TS0601"],
	"product": "Single Phase Din Rail Smart Energy Meter",
	"sleeper": true,
	"status": "Gold",
	"subdevices": [{
			"type": "$TYPE_ON_OFF_OUTPUT",
			"restapi": "/lights",
			"uuid": [
				"$address.ext",
				"0x01"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "state/on",
					"parse": {
						"fn": "tuya",
						"dpid": 16,
						"eval": "Item.val = Attr.val;"
					},
					"write": {
						"fn": "tuya",
						"dpid": 16,
						"dt": "0x10",
						"eval": "Item.val == 1 ? 1 : 0;"
					},
					"read": {
						"fn": "tuya"
					}
				},
				{
					"name": "state/reachable"
				}
			]
		},
		{
			"type": "$TYPE_CONSUMPTION_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0702"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/offset",
					"default": 0
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/consumption",
					"parse": {
						"fn": "tuya",
						"dpid": 1,
						"eval": "Item.val = Attr.val / 100;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "state/lastupdated"
				}
			]
		},
		{
			"type": "$TYPE_POWER_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0b04"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "none"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/lastupdated"
				},
				{
					"name": "state/current",
					"parse": {
						"fn": "tuya",
						"dpid": 6,
						"eval": "const buffer = Buffer.from(Attr.val, 'base64'); Item.val = (buffer[12] | buffer[11] << 8) / 1000"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "state/voltage",
					"parse": {
						"fn": "tuya",
						"dpid": 6,
						"eval": "const buffer = Buffer.from(Attr.val, 'base64'); Item.val = (buffer[14] | buffer[13] << 8) / 10"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "state/power",
					"parse": {
						"fn": "tuya",
						"dpid": 103,
						"eval": "Item.val = Attr.val / 10;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				}
			]
		}
	]
}
```

BabaIsYou, 2023-05-30:
I'm not really sure it will works that way but could you try this DDF ?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_lsanae15", "_TZE204_lsanae15"],
  "modelid": ["TS0601", "TS0601"],
  "product": "Single Phase Din Rail Smart Energy Meter",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
      {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val / 100;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'base64'); Item.val = (buffer[4] | buffer[8] << 8) / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'base64'); Item.val = (buffer[1] | buffer[0] << 8) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

BabaIsYou, 2023-06-07:
If the DDF is in the "right" place (depends on OS) it should be automatically loaded once you rejoin the device, restart DeConz or press F5 after selecting the device into Deconz-GUI. You can also, as you did, open the DDF, save it and hot reload. Just be sure that at the botton of the windows, after hot releoad, the DDF is really loaded with no error message.

But, as @Smanar mentionned, it's possible that DeConz is not able to actually catch raw data thru REST-API.

closed 2023-07-06 by github-actions[bot]