dresden-elektronik/deconz-rest-plugin #7035: Task overflow when many lights are changed individually

poelzi, 2023-06-01:
## Describe the bug

I have many small mr16 spots in a room, 35+. When they are addressed as a group, no problems, but I use "adaptive lighting" with home assistant  to control the lights and they have to be addressed individually for adjustments.
When all lights are set at the same time, a task overflow occures:

## Steps to reproduce the behavior

1. Set 30+ lamps indvidually through homeassistant
2. Not all lamps are set

## Expected behavior

I think the API should return a propper error code, eigther a 503 with a propper json error.

Upstream integrations could then just queue the commands.

## Environment

 - Host system: Raspberry Pi
 - Running method: Docker container
 - Firmware: 26720700
 - deCONZ version: 2.21.02
 - Device:  ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
```
01:23:56:432 delay sending request 188 dt 0 ms to 0x84FD27FFFE92E73B, ep: 0x0B cluster: 0x0300 onAir: 1
01:23:56:433 delay sending request 191 dt 0 ms to 0xA4C1381349489D16, ep: 0x0B cluster: 0x0008 onAir: 1
01:23:56:434 delay sending request 192 dt 0 ms to 0xA4C1381349489D16, ep: 0x0B cluster: 0x0300 onAir: 1
01:23:56:436 send move to color temperature 476 to 0xA4C1386C8FF14945
01:23:56:438 delay sending request 183 dt 0 ms to 0x84FD27FFFE92597F, ep: 0x0B cluster: 0x0300 onAir: 1
01:23:56:439 delay sending request 187 dt 0 ms to 0x84FD27FFFE92E73B, ep: 0x0B cluster: 0x0008 onAir: 1
01:23:56:440 delay sending request 188 dt 0 ms to 0x84FD27FFFE92E73B, ep: 0x0B cluster: 0x0300 onAir: 1
01:23:56:441 delay sending request 191 dt 0 ms to 0xA4C1381349489D16, ep: 0x0B cluster: 0x0008 onAir: 1
01:23:56:442 delay sending request 192 dt 0 ms to 0xA4C1381349489D16, ep: 0x0B cluster: 0x0300 onAir: 1
01:23:56:443 delay sending request 195 dt 0 ms to 0xA4C138B7A47C91EF, ep: 0x0B cluster: 0x0008 onAir: 1
01:23:56:444 delay sending request 196 dt 0 ms to 0xA4C138B7A47C91EF, ep: 0x0B cluster: 0x0300 onAir: 1
01:23:56:446 send move to color temperature 476 to 0xA4C138D7151627D1
01:23:56:447 5 running tasks, wait
01:23:56:449 send move to color temperature 476 to 0xA4C138A3CA5F5418
01:23:56:450 5 running tasks, wait
01:23:56:452 send move to color temperature 476 to 0x84FD27FFFE92E60C
01:23:56:453 5 running tasks, wait
01:23:56:455 send move to color temperature 476 to 0x84FD27FFFE92597B
01:23:56:456 failed to add task 5571405 type: 6, too many tasks
01:23:56:457 5 running tasks, wait
01:23:56:459 failed to add task 5571407 type: 14, too many tasks
01:23:56:460 failed to add task 5571408 type: 11, too many tasks
01:23:56:461 send move to color temperature 476 to 0xCC86ECFFFEC72546
01:23:56:462 failed to add task 5571409 type: 6, too many tasks
01:23:56:463 5 running tasks, wait
01:23:56:465 failed to add task 5571411 type: 14, too many tasks
01:23:56:466 failed to add task 5571412 type: 11, too many tasks
01:23:56:467 send move to color temperature 476 to 0xCC86ECFFFEC724C2
01:23:56:468 failed to add task 5571413 type: 6, too many tasks
01:23:56:469 5 running tasks, wait
01:23:56:471 failed to add task 5571415 type: 14, too many tasks
01:23:56:472 failed to add task 5571416 type: 11, too many tasks
01:23:56:473 send move to color temperature 476 to 0xA4C13873FE53939A
01:23:56:474 failed to add task 5571417 type: 6, too many tasks
01:23:56:475 5 running tasks, wait
01:23:56:480 5 running tasks, wait
01:23:56:618    0x84FD27FFFE92597F force poll (2)
01:23:56:619 failed to add task 5571420 type: 11, too many tasks
01:23:56:620 send move to color temperature 476 to 0xCC86ECFFFEBB308C
01:23:56:621 failed to add task 5571421 type: 6, too many tasks
01:23:56:622 5 running tasks, wait
01:23:56:624 failed to add task 5571423 type: 14, too many tasks
01:23:56:625 failed to add task 5571424 type: 11, too many tasks
01:23:56:626 send move to color temperature 476 to 0xA4C138FB68EB4E4E
01:23:56:627 failed to add task 5571425 type: 6, too many tasks
01:23:56:628 5 running tasks, wait
01:23:56:630 failed to add task 5571427 type: 14, too many tasks
01:23:56:631 failed to add task 5571428 type: 11, too many tasks
01:23:56:632 send move to color temperature 476 to 0xCC86ECFFFEB79332
01:23:56:633 failed to add task 5571429 type: 6, too many tasks
01:23:56:634 5 running tasks, wait
01:23:56:636 failed to add task 5571431 type: 14, too many tasks
01:23:56:637 failed to add task 5571432 type: 11, too many tasks
01:23:56:638 send move to color temperature 476 to 0xCC86ECFFFEC724B7
01:23:56:639 failed to add task 5571433 type: 6, too many tasks
01:23:56:639 5 running tasks, wait
01:23:56:641 5 running tasks, wait
01:23:56:681 failed to add task 5571435 type: 14, too many tasks
01:23:56:682 failed to add task 5571436 type: 11, too many tasks
01:23:56:683 send move to color temperature 476 to 0x84FD27FFFE92E7E6
01:23:56:684 failed to add task 5571437 type: 6, too many tasks
01:23:56:685 5 running tasks, wait
```

## Additional context
~ 35 lamps
~ 3 from a different room
~ 15 other devices

Smanar, 2023-06-02:
You haven't error on API ?
Because It can don't work for all requests, but for exemple if you try to change the brightness of a bulb and the task fail, you will have thoses error and log AND an error return on the API, an "Internal error" " bridge busy" or something like that.

Mimiix, 2023-06-02:
@Robban how does this get handled in HA?

Kane610, 2023-06-02:
If deCONZ sends a bridge busy error it will retry up to 3 times https://github.com/Kane610/deconz/blob/5be462066ebf328d1531b4a96324009aff1bbc2d/pydeconz/gateway.py#L142

Mimiix, 2023-06-02:
@poelzi what does the api return?

BabaIsYou, 2023-06-03:
@poelzi : this may help to understand what could hapenning in your situation : https://forum.phoscon.de/t/delayed-and-missing-bulb-updates-deconz-struggling-to-handle-large-networks/2424/17

manup, 2023-06-03:
Right now there is no good solution for this, handling many lights at once is best done with groups (but this only applies when the group shares the same state). We are transitioning to a different way based on **StateChange** which is roughly described in https://dresden-elektronik.github.io/deconz-dev-doc/modules/fault-tolerance 

The gist of the internal *StateChange* is that currently most light commands are enqueued directly when a REST-API command is received in a fire-and-forget manner, which scales poorly and doesn't handle faults well. With *StateChange* it doesn't matter if 5 or 500 commands are enqueued, it can be seen as a transaction which the REST-API client issues, deCONZ will then try to fulfill the transaction as soon as possible – when there is space in the low-level Zigbee command queue (APS) and a device is awake/reachable – this also recovers any errors automatically. For a bunch of DDF based sensors and configuration tasks we use it already lights are on the todo list, here we need more cleanup of the legacy code base first as the lights stuff is quite messy currently. For the REST-API client this is transparent but there will also be API additions to query the current state of the StateChange transactions.

closed 2023-07-03 by github-actions[bot]