dresden-elektronik/deconz-rest-plugin #7043: Tuya Soil Sensor won't connect.

Maddinthebrain, 2023-06-08:
## Describe the bug
As in V2.22.0 the Tuya Soil ([Link to an example](https://smarthomescene.com/reviews/tuya-zigbee-plant-soil-sensor-gxm-01-review/)) sensor was added to deconz I bought two of them. And well, they won't connect to deconz. They are not even found as an unknown device.

## Steps to reproduce the behavior

## Expected behavior
It should be found as a sensor, but it is found nothing.

## Screenshots

## Environment
Conbee II
Latest deconz 2.22.01 beta
Latest firmware 26780700

Running on raspberry OS.

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26780700)
 - deCONZ version: (2.22.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes 0.5m long, to gain distance to the Raspi, because of disturbance by USB3.0 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2023-06-08:
You probably have another one (different chip) thus not whitelisted.

This is definitely not a bug and that's why im closing this issue.

For help ik debugging, please use the forums.

Kind regards, Dennis

closed 2023-06-08 by Mimiix

BabaIsYou, 2023-06-08:
Or maybe just have a look to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6731#issuecomment-1556634769

Maddinthebrain, 2023-06-23:
the API indeed catches both of the values. Then at some point the value of the moisture gets lost

[image]
[image]

Maddinthebrain, 2023-07-06:
The problem is in fhem. Fhem recieves the value correctly but it isn't processed correctly.