dresden-elektronik/deconz-rest-plugin #7045: Iluminize 5128.10

AndreKoepke, 2023-06-10:
## Device

- Product name: Zigbee 3.0 roller shutter actuator
- Manufacturer: iluminize
- Model identifier: 5128.10
- Device type : 
  - Other: roller shutter actuator

# Special
This device appears to be a normal dimmer for lights. When the dimmer changed to ...
* 0-30%: roller shutter will close
* 30-70%: roller shutter will stop
* 70-100%: roller shutter will open

In deCONZ-gui, it also appears as "window covering" and all functions works like a charm.

## Screenshots

[image]

### Basic
[image]

### Window covering
[image]

Mimiix, 2023-06-10:
Can you please add all requested screenshots :)? 

Thanks!

AndreKoepke, 2023-06-11:
@Mimiix sure. Which ones are missing?

BabaIsYou, 2023-06-11:
> Which ones are missing?

0x0000, 0x003, 0x0004, 0x0006, 0x0008, 0x1000 and 0x0b05

AndreKoepke, 2023-06-11:
Done. I hope that I screenshoted the right things. ^^

BabaIsYou, 2023-06-11:
Seems right thx

AndreKoepke, 2023-07-09:
PR (#7046) is open and waiting for merge. Should the issue be open until it is merged?

Mimiix, 2023-07-09:
Yes.