dresden-elektronik/deconz-rest-plugin #7047: Device name

SheriffDeputy, 2023-06-11:
## Device

- Product name: EARU EAYCBM-Z 1P : MCB- Micro circuit breaker
- Manufacturer:_TZE204_davzgqq0
- Model identifier: TS0601
- Device type  
  - Switch

## Screenshots

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[deconz_logs.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/11714793/deconz_logs.log)

Mimiix, 2023-06-11:
Hi,

Please add screenshots of all clusters.

Smanar, 2023-06-11:
Can start with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_davzgqq0",
  "modelid": "TS0601",
  "product": "EAYCBM-Z 1P",
  "vendor": "Tuya",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {
            "fn": "tuya",
            "dpid": 16,
            "eval": "Item.val = Attr.val;"
          },
          "write": {
            "fn": "tuya",
            "dpid": 16,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;"
          },
          "read": {
            "fn": "tuya"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {
            "fn": "tuya",
            "dpid": 1,
            "eval": "Item.val = Attr.val * 10;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": []
}
```

If I m right you will have on/off and consumption, for values will be harder, it seem they are all in the same dpid https://github.com/zigpy/zha-device-handlers/issues/2061#issuecomment-1575572651

SheriffDeputy, 2023-06-11:
I loaded this JSON DDF in deconz but still the same behaviour

Mimiix, 2023-06-11:
Please provide all screenshots.

SheriffDeputy, 2023-06-11:
> Please provide all screenshots.

What screen shot should I post?

Mimiix, 2023-06-11:
@Smanar why did this change ?

SheriffDeputy, 2023-06-11:
After switching it off and on again it seems to be back as it was
[image]

Mimiix, 2023-06-11:
The ias ace is also missing ?

Smanar, 2023-06-11:
Loool, I think it's a realy stable device.
But from I m reading, all cluster are useless, then can come and leave, they are useless, this device use only the tuya cluster 0xEF00

With or without DDF you will see (at least need to see, but it seem it's random on this device) the same thing.
The only difference will be on the API.
Just add the DDF and re-include the device, don't care at what you can see in deconz, only check the difference in the API.

SheriffDeputy, 2023-06-11:
After removing and re-adding the device in deconz, it still does not work. However home assistant now detects the switch and a sensor for the consumption. None of them works (the sensor is always 0).

Also now in home assistant I can see an error when I toggle the swith. It says the service call failed as the control is not available

SheriffDeputy, 2023-06-12:
> Please provide all screenshots.

are you still waiting for screenshots? if so what exactly?

Mimiix, 2023-06-12:
> > Please provide all screenshots.
> 
> are you still waiting for screenshots? if so what exactly?

Not sure.

@Smanar which ones do we like?

Smanar, 2023-06-12:
Good question, I think for this device you can start direclty with the DDF.
From that I m reading on other zigbee project, all others clusters are useless, and from your tests totally random ^^.

If I m right only the tuya one is usefull, and for this one you can't have attribute.
So I think we can start with that, we will see later if we need the other "garbage" clusters.

SheriffDeputy, 2023-06-12:
> Good question, I think for this device you can start direclty with testing the DDF. From that I m reading on other zigbee project, all others clusters are useless, and from your tests totally random ^^.
> 
> If I m right only the tuya one is usefull, and for this one you can't have attribute using the GUI. So I think we can start with that, we will see later if we need the other "garbage" clusters.

Thank you Smanar for your help.
The DDF  above does not work. It expose the power control (but no value is retrieved) but does not allow for the on/off service.

here the tuya tab (virtually empty)

[image]

SheriffDeputy, 2023-06-13:
> > The DDF above does not work. It expose the power control (but no value is retrieved) but does not allow for the on/off service.
> 
> Wich one entry you have in API ? Visible with Phoscon/help/API Information (then lights and sensors, the DDF create 2 entries) You are using phoscon to test the on/off ? You can use the "previous DDF" entry in API, you have deleted the device before re-including after the DDF ?
> 
> You haven't "tuya debug" with flag "info" on deconz ?
> 
> `"TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",`

1) I am using phoscon deconz on/off feature but not always visible.

2) I have delete and readded the device many times after adding the previous ddf (saved as json file) in the /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices 
please note the filehas been saved there and no folder for manufacture

3)Sorry I am a newbe on phoscon/deconz/zigbee. I do not understand how to enable the TUYA debugging. how can I start deconz with tuya debug flag?

Mimiix, 2023-06-13:
Please keep open until merged.

Smanar, 2023-06-14:
Yep third app are able to act direclty on the API (and not deconz).
If it work we can add later tension/current but need more code in DDF. So better to keep this issue opened.

closed 2023-07-13 by github-actions[bot]