dresden-elektronik/deconz-rest-plugin #7048: Battery reports 0% always in _TZ3000_wkai4ga5 (tuya 4-gang)

josemi73, 2023-06-12:
## Describe the bug

Hi, the device in the title is not reporting battery status, how could I debug it?

[image]

## Steps to reproduce the behavior

Battery status reports 0% always.
## Expected behavior

it should show the real status of the battery.
Buttons are working properly.

## Screenshots

[image]

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian with USB conbee II connected via network from Home Assistent deCONZ Add-on
 - Firmware version: 26780700
 - deCONZ version: 2.22.01
 - Device: Conbee ii
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

18:22:47:265 DB found sensor C_I_4BTOLDOS 74
18:22:47:266 DB skip loading sensor C_I_4BTOLDOS 74, handled by DDF Tuya remote 4 gangs
18:23:35:451 Websocket 192.168.1.222:34182 send message: {"attr":{"id":"74","lastannounced":null,"lastseen":"2023-06-12T16:23Z","manufacturername":"_TZ3000_wkai4ga5","modelid":"TS0044","name":"C_I_4BTOLDOS","swversion":"1.0.2","type":"ZHASwitch","uniqueid":"70:ac:08:ff:fe:65:f4:c5-01-0006"},"e":"changed","id":"74","r":"sensors","t":"event","uniqueid":"70:ac:08:ff:fe:65:f4:c5-01-0006"} (ret = 2124871840)
18:23:35:452 APS-DATA.request id: 23, addrmode: 0x03, addr: 0x70ac08fffe65f4c5, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x04
18:23:49:317 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('74', 'C_I_4BTOLDOS', 'ZHASwitch', 'TS0044', '_TZ3000_wkai4ga5', '70:ac:08:ff:fe:65:f4:c5-01-0006', '1.0.2', '{"buttonevent":1002,"lastupdated":"2023-06-12T16:23:48.976"}', '{"battery":0,"on":true,"reachable":true}', '', 'normal', '1', '2023-06-12T16:23Z', '2023-06-12T16:23:40Z')
18:23:56:949 [INFO] - Button 1004 - TS0044, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 double, payload: 01, zclSeq: 96
18:23:56:950 Websocket 192.168.1.222:34182 send message: {"e":"changed","id":"74","r":"sensors","state":{"buttonevent":1004,"lastupdated":"2023-06-12T16:23:56.949"},"t":"event","uniqueid":"70:ac:08:ff:fe:65:f4:c5-01-0006"} (ret = 2124870992)
## Additional context

josemi73, 2023-06-12:
> As I can see it seems that bindings for battery reports had been removed during a previous DDF modification. Are you able to modify DDF file directly onto your system to give a try to reintroduce the battery report with someting like this ?
> 
> ```
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0x0001",
>       "report": [
>         {
>           "at": "0x0021",
>           "dt": "0x20",
>           "min": 14400,
>           "max": 86400,
>           "change": "0x00000002"
>         }
>       ]
>     },
> ```
> 
> If not, but have access to DeConz GUI, then can be added using DDF editor.

Hi @BabaIsYou 
Thank you for your answer.
I have just modified inserting your lines into  _TZ3000_TS0044_4gang_remote.json, right?
Or should it be another file?
I have rebooted deconz gateway but battery is not reporting yet.

BabaIsYou, 2023-06-12:
In Deconz GUI, if you make a read onto the cluster 0x0001, does it make a n update ?

BabaIsYou, 2023-06-12:
1. It could stay to 0 if device is sleeping. Perhaps you have to awake the device during read.
2. You’re comparing two different things. The bindings are in the « bindings » tab not the « item » one ;-)

josemi73, 2023-06-13:
[image]

@BabaIsYou And Is it normal that in DDF Editor changes in the json in filesystem are not shown?

josemi73, 2023-06-13:
Updated!

[image]

I configured the Read in battery item, and I inserted the lines in .json @BabaIsYou wrote above.

BabaIsYou, 2023-06-13:
Hi. To make it persistent we’ll have to open a pull request with the updated DDF.

do you want to do it ?

Mimiix, 2023-06-13:
Please keep open until pr is merged.

josemi73, 2023-06-13:
> Hi. To make it persistent we’ll have to open a pull request with the updated DDF.
> 
> do you want to do it ?

I do not know github, it could be dangerous. I added your lines and I enabled the Read in the item battery, I suppose your lines did the trick, because in theory battery read does not work.

Smanar, 2023-06-13:
>I have filled these fields, though I have seen in the zigbee2mqtt webpage that for TS0044 battery attribute is not readable:

Yep, take care, the binding was not removed by error, with it device have battery drain https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6744#issuecomment-1427044174

BabaIsYou, 2023-06-13:
I guess that the removed report was too "stressfull" for a battery report (every 60s !?) and then could have drained the battery just for that.

BabaIsYou, 2023-06-13:
> If I see that the battery drains, Could it be changed to reports 100% always?

Yes it can but this seems a kind of nonsense ... isn't it ?

> These lines you told me mean that the device reports every 14000->86400 seconds?

Yes, between 4h and a whole day, or if value changes for .a value of 2

I think that the drain of battery was due from the combination with stressfull values on report, and leak of "read" function (that leads to a flood of battery refresh attempts, already documenented if I remeber in another issue or on Discord.

BabaIsYou, 2023-06-13:
> Yep, take care, the binding was not removed by error, with it devices have battery drain [#6744 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6744#issuecomment-1427044174)

Perhaps that making separated DDF files would have been, and is still, the right way to address this ?

BabaIsYou, 2023-06-13:
> > These lines you told me mean that the device reports every 14000->86400 seconds?
> 
> Yes, between 4h and a whole day, or if value changes for .a value of 2

BTW have to reduce max to 65535 (maximum value for U8 num type)

josemi73, 2023-06-13:
> 

Ok, I will reduce both min 64500, max 65500 then.

BabaIsYou, 2023-06-14:
Ok. I deleted the PR and we will wait further results from @josemi73 regarding battery lifetime.

josemi73, 2023-06-14:
> Ok. I deleted the PR and we will wait further results from @josemi73 regarding battery lifetime.

It is soon to come to a conclusion. 

[image]

Same battery percentage than yesterday, but last report 10 hours ago. Tomorrow I will look again.

My json file:

```
(...)
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 65500,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
(...)        
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 64500,
          "max": 65500,
          "change": "0x00000002"
        }
      ]
    },
  (...)
```

josemi73, 2023-06-16:
> Ok. I deleted the PR and we will wait further results from @josemi73 regarding battery lifetime.

I think that in the next few days battery will be reaching 100% after the modified DDF 🤣😊 (79 -> 91% today)

[image]

josemi73, 2023-06-17:
91% -> 90%

I would approve the change.

[image]

[image]

closed 2023-08-12 by github-actions[bot]

ekuekb, 2024-01-10:
Hi, 
My name is Ekue, I am working on Home Assistance project and I had the same issue

Smanar, 2024-01-11:
Hello, what is your device ? (the manufacture name)

PR #7203 (2023-08-31): Fix battery not reported for Tuya 4-gang  remote _TZ3000_TS0044_4gang_remote.json
Battery % didn't report anymore since last modification, and missing "read" in "config/battery".

Fix #7048

Replace PR #7050 cause of conflicting changes