dresden-elektronik/deconz-rest-plugin #7056: DIY CC2530 with ptvo firmware - Implementationin Deconze?

Maschler, 2023-06-18:
## Describe the bug
Hello, i flashed a CC25301 chip with ptvo firmware, the purpose is to return a on/off switch status. But something is strange. 

After flashing I can pair the cc2530 with my conbee2. I can see the device and also switch the input via the interface of Deconz (conbee2). BUT that's it.

In the web interface of the Conbee2 (Phoscon) the thing is not shown at the devices (light/sensor/switch). At the Node Info there is also only a "cryptic" name. Normally the name of the device is to be seen there like e.g. "Tradfri on/off...". Also in the Node List all the usual information is empty.

Is there something like a device description missing so it shows up in Phoscon as a light/sensor/switch?

Thanks !

## Device

    Product name: DIY CC2530+C## Describe the bug
Hello, i flashed a CC25301 chip with ptvo firmware, the purpose is to return a on/off switch status. But something is strange. 

After flashing I can pair the cc2530 with my conbee2. I can see the device and also switch the input via the interface of Deconz (conbee2). BUT that's it.

In the web interface of the Conbee2 (Phoscon) the thing is not shown at the devices (light/sensor/switch). At the Node Info there is also only a "cryptic" name. Normally the name of the device is to be seen there like e.g. "Tradfri on/off...". Also in the Node List all the usual information is empty.

Is there something like a device description missing so it shows up in Phoscon as a light/sensor/switch?

Thanks !

## Device

    Product name: DIY CC2530 with ptvo firmware
    Manufacturer: N/A
    Model identifier: N/A
    Deconz v2.21.02
    Firmware:0x26780700
    Raspberry
    Device type :
        On/Off output

## Screenshots
[image]
[image]
[image]

BabaIsYou, 2023-06-19:
Your device is not already supported by Deconz (that's why it's name is hexadecimal). Need to add a DDF file for it and then will be recognized by REST-API and perhaps by Phoscon.

Could you provide all the screenshots asked by the template for opening device request issue ?

Maschler, 2023-06-19:
Thank you for the answer. I saw that something similar was done here. 

Since this is not yet my final configuration (wanted to understand the workflow first) it probably doesn't make sense to create a .dff now. 

But can I create something like that myself? Is there an editor or a manual for it?

[dresden-elektronik/deconz-rest-plugin/issues/5496](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5496)

Mimiix, 2023-06-19:
Please add all requested screenshots.

Maschler, 2023-06-19:
[image]

[image]

[image]

Maschler, 2023-06-19:
By the way. The configuration Tool for the ptvo firmware also provides a "custom converter for Z2M". Here is the content of the .js file:

```
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const zigbeeHerdsmanUtils = require('zigbee-herdsman-converters/lib/utils');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;

const ptvo_switch = zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;
fz.ptvo_on_off = {
  cluster: 'genOnOff',
  type: ['attributeReport', 'readResponse'],
  convert: (model, msg, publish, options, meta) => {
      if (msg.data.hasOwnProperty('onOff')) {
          const channel = msg.endpoint.ID;
          const endpointName = `l${channel}`;
          const binaryEndpoint = model.meta && model.meta.binaryEndpoints && model.meta.binaryEndpoints[endpointName];
          const prefix = (binaryEndpoint) ? model.meta.binaryEndpoints[endpointName] : 'state';
          const property = `${prefix}_${endpointName}`;
	  if (binaryEndpoint) {
            return {[property]: msg.data['onOff'] === 1};
          }
          return {[property]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
      }
  },
};

const device = {
    zigbeeModel: ['ptvo test'],
    model: 'ptvo test',
    vendor: 'ptvo info',
    description: '[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
    fromZigbee: [fz.ignore_basic_report, fz.ptvo_on_off,],
    toZigbee: [tz.ptvo_switch_trigger, tz.on_off,],
    exposes: [e.switch().withEndpoint('l1'),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l1: 1,
        };
    },
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
      await endpoint.read('genBasic', ['modelId', 'swBuildId', 'powerSource']);
    },

};

module.exports = device;
```

Is this useable for adopting a ddf?

BabaIsYou, 2023-06-19:
> But can I create something like that myself? Is there an editor or a manual for it?

May be you could try with this : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

BabaIsYou, 2023-06-19:
> **Screenshot of the Basic cluster** 
> 

> 
> 
> 
> **Screenshot of the Node Info Panel** 

Screenshot of the On/Off cluster is still missing.

Maschler, 2023-06-19:
Added at the end.

Mimiix, 2023-06-19:
Please add them to the original post

Maschler, 2023-06-19:
Done.

BabaIsYou, 2023-06-28:
> > But can I create something like that myself? Is there an editor or a manual for it?
> 
> May be you could try with this : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

@Maschler did you get a chance to read this ?

closed 2023-07-28 by github-actions[bot]