dresden-elektronik/deconz-rest-plugin #7057: iCasa ICZB-R12D is able to measure voltage and current

Maschler, 2023-06-19:
## Description
Hello,

I have noticed that the iCasa ICZB-R12D can also measure. This is currently not implemented.

https://www.zigbee2mqtt.io/devices/ICZB-R12D.html

## Device
[ICZB-R12D iCasa](https://www.zigbee2mqtt.io/devices/ICZB-R12D.html)

### Basic and Identify
[image]
[image]
[image]

### Simple Metering
[image]
[image]
[image]
[image]

### Electrical measurement
[image]
[image]
[image]

### Electrical measurement (change between on/off)
**off**
[image]
**on 100%**
[image]
**on ~50%**
[image]

Maschler, 2023-06-19:
Thanks for help. It works!

**I have three questions:**
1. What is the Task of the marked points? Must "reportable change" be bigger then min interval?
[image]

2. What are the bindings for? What the idea behind?
3. What do i have to do to see the values in Phoscon?

### This DDF fiile works fine for me:
```{
  "schema": "devcap1.schema.json",
  "manufacturername": "icasa",
  "modelid": "ICZB-R12D",
  "product": "ICZB-R12D",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/iczb-r12d.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 0,
            "eval": "if (Attr.val != -32768 && Attr.val != 32768) { Item.val = Attr.val /10 ; }"
          },
          "default": 0
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300,
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 0,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val / 10; }"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 3,
          "max": 300,
          "change": "0x00000003"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 10,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "groupcast",
      "config.group": 24,
      "src.ep": 1,
      "cl": "0x2000"
    },
    {
      "bind": "groupcast",
      "config.group": 123,
      "src.ep": 1,
      "cl": "0x2000"
    }
  ]
}
```
100%
[image]

~50%
[image]

0%
[image]

Mimiix, 2023-06-19:
They probably won't show in phoscon as phoscon needs to add support themselves. Phoscon is closed source.

The rest api is what you are achieving here.

BabaIsYou, 2023-06-19:
> 1. What is the Task of the marked points? Must "reportable change" be bigger then min interval?

Min interval : minimum interval in seconds for value of the attribute to be reported
Max interval : Same, but for maximum but maximum.
Reportable change : in between these previous values, if change of the attribute value is equal or upper to this value it will be reported

> 2\. What are the bindings for? What the idea behind?

That's the less documented part of this software to which a user is quickly confronted to ! ;-) some kind of information here https://forum.phoscon.de/t/bindings-documentation/1935 The global idea is to bind the report of an attribute from a cluster to others into the mesh. If this device is supporting reporting. It can be thru a groupcast for a specific group or unicast communication.

> 3\. What do i have to do to see the values in Phoscon?

Minimaly thru Help/API information/sensors into Phoscon. To have it displayed into Phoscon GUI, make a device request into Phoscon forum as it's a closed sotfware with it's own whitelist.

SwoopX, 2023-06-19:
> 1. What is the Task of the marked points? Must "reportable change" be bigger then min interval?
> 2. What are the bindings for? What the idea behind?
> 3. What do i have to do to see the values in Phoscon?

For 1)
You better approach that upside down.
Reportable change: send a report if raw value changes +/- of what you entered
min reporting interval: upon value change, send a report earliest after X seconds
max reporting interval: send a report after a maximum of X seconds, regardless of the value

For 2)
Think of it as telling the device to whom to talk to: Hey device, if you wanna talk about the current power consumption, tell it the coordinator.

For 3)
Answer has already been given. Phoscon is not capable of showing power or consumption data as part of any sensor.

P.S.: I'm afraid the DDF still requires some more tweaking before it's ready for a PR 🙂

Maschler, 2023-06-20:
@SwoopX @Mimiix @BabaIsYou Thanks for answering.

@SwoopX What are your recommendations regarding the DDF?

SwoopX, 2023-06-28:
What I see on a quick glimpse:

- Please remove the group case bindings at the end
- The consumption sensor does not output the correct values. Based on the screenshots, the raw value must be divided by 3600 to provide Wh
- The consumption sensor binding reportable change values appears to be way to small, which could result in rapid fire
- Refresh intervals should be +60 as their respective max value from the binding
- Device should also have a ZHASwitch resource including a (presumably simple) button map
- "description" and "default" values can typically be removed

closed 2023-07-28 by github-actions[bot]