dresden-elektronik/deconz-rest-plugin #7058: Aqara WS-USC02

mattgio22, 2023-06-20:
## Device

- Product name: Aqara WS-USC02
- Manufacturer: LUMI
- Model identifier: lumi.switch.b2laus01
- Device type : 
  - Switch

## Screenshots
[image]
[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2023-06-20:
Can you please add a screenshot of the full cluster list

Mimiix, 2023-06-20:
Can you add all screenshots of all clusters seems some are missing.

mattgio22, 2023-06-20:
My apologies. I believe this is the rest of them.

[image]
[image]
[image]
[image]
[image]

Mimiix, 2023-06-20:
Please add to the first post in this topic. Easier to find it :)

mattgio22, 2023-06-20:
Top comment now has all images.

mattgio22, 2023-06-27:
So - I actually did manage to get this working with the top switch (0x01). The bottom switch (0x02) does not work - and I'm not exactly sure the best way to add it. Either way, here's the code I used:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.switch.b2laus01",
  "vendor": "BackyardSwitch1",
  "product": "lumi.switch.b2laus01",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/lumi.switch.b2laus01.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on/startup",
          "description": "On/Off state on device startup (power on).",
          "refresh.interval": 3600,
          "default": 255
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

Mimiix, 2023-06-28:
@Smanar , @BabaIsYou or @SwoopX any of you able to help out :)?

mattgio22, 2023-06-28:
Slight update! I did manage to get it. Here is the working code for both top and bottom switches. This works great now. If anyone else in the future wants to use it, there are two ways - either use the json below, or in the DDF editor, drag over two "On/Off Light", and put 0x01 and 0x02 for the addresses respectively. Hopefully this helps anyone else who has this switch!

{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.switch.b2laus01",
  "vendor": "BackyardSwitch1",
  "product": "lumi.switch.b2laus01",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/lumi.switch.b2laus01.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on/startup",
          "description": "On/Off state on device startup (power on).",
          "refresh.interval": 3600,
          "default": 255
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on/startup",
          "description": "On/Off state on device startup (power on).",
          "refresh.interval": 3600,
          "default": 255
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

Smanar, 2023-06-28:
Hello, can you make a PR with the DDF for other users ?
No so hard as there is only 1 file.

Just some commment :
- remove "description" field
- the "config/on/startup" is working for this device ?

Mimiix, 2023-06-28:
Please keep open until merged.

BabaIsYou, 2023-06-28:
Hi, Nice try !

A few other comments : 

- `"vendor": "BackyardSwitch1",` Seems better to be `Aquara`
- "`product": "lumi.switch.b2laus01",` is rather  `WS-USC02`
- It's also seems to have a temperature senor inside this device. Have you tried to add a temperature sensor into the DDF like something like this  ? (Don't really know if it will works because it seems you didn't press on the "read" button before taking the screenshots of the clusters)
```
   {
         "type":"$TYPE_TEMPERATURE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0002"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/offset",
               "default":0
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/temperature",
               "refresh.interval":60,
               "read":{
                  "at":"0x0000",
                  "cl":"0x0002",
                  "ep":1,
                  "fn":"zcl"
               },
               "parse":{
                  "at":"0x0000",
                  "cl":"0x0002",
                  "ep":1,
                  "eval":"Item.val = Attr.val"
               },
               "default":0
            },
            }
         ]
      }
```

- May be some bindings were missing to

SwoopX, 2023-06-28:
Let's please stop raping that poor little device 🙂 

I'll prepare a PR for it. No bindings, not temperature sensor and it misses a switch resources and other features.

mattgio22, 2023-06-28:
haha - I don't think it has a temp sensor, but i can try it out. It was my first time working with DDF/json, so my apologies. I was just excited that I got it working :)

SwoopX, 2023-07-02:
@mattgio22 It feels like you're on a rather old firmware, as the typical cluster is missing here (0xfcc0). You might want to update your Conbee/Raspbee firmware and then delete + rejoin the device to make it visible. My expectation would also be that then at least 1 new endpoint would pop up (which would be required to add an additional ZHASwitch resource).

@Smanar I'm surely not denying that mattgio22 is the one actually working hands on with the device, but I'm struggling a bit to see the added value/advantage of your suggestion.

mattgio22, 2023-07-09:
@SwoopX Understood. I've been away, but I'll update firmware and then re-upload photos. I don't know if I said this, but it currently works with both top and bottom, so I don't think the firmware update is 100% necessary here.

Regarding who does the PR, I don't really care - I'm happy to do the work, but I've never done a PR for anything before. That said, it would be good experience for me, but I certainly will need some guidance. Let me update the firmware, and then I'll read more about your process for doing the PR. 

Thank you to all.