dresden-elektronik/deconz-rest-plugin #7061: Xiaomi Aquara Water leak sensor (SJCGQ11LM) falsely reports trouble on deConz 2.22.2 (HA addon)

bennydiamond, 2023-06-26:
## Describe the bug
I'm running deConz HA addon.
I updated addon to version 6.20.0 (which includes deConz 2.22.2). One of my water leak sensor SJCGQ11LM started reporting trouble right away.

I tried to delete and re-pair the device, device will behave normally for a couple of seconds and then will fall back to reporting trouble.

I tried replacing the battery in the device, same issue.

I downgraded deConz HA addon to previous version (6.19.0). The problem went away. SJCGQ11LM sensor behaves as intended.

Only 1 of my 5 * SJCGQ11LM sensor exhibited the problem but I suspect this is due to the fact the others might have been sleeping and thus did not report to deConz while I was witnessing the issue.

## Steps to reproduce the behavior

Pair "SJCGQ11LM" with deConz HA addon 6.20.0
Once paired, press the button on SJCGQ11LM sensor to trigger sending its attributes
Device will report trouble/wet and will stay in that state indefinitely

## Expected behavior

 If not water is physically touching the sensor, the sensor should not report trouble/wet.

## Environment

 - Host system: USFF PC, HP ProDesk Mini G3 400, running Debian 12.
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26400500
 - deCONZ version: (2.22.2)
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## Additional context

I reverted to HA Addon 6.19.0 and did not save any logs. If necessary, I could update again to the faulty version and generate logs.

I highly suspect the issue originates from this change: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6919

Mimiix, 2023-06-27:
@SwoopX can you check?

Larry0ua, 2023-06-27:
Same here. 3 of 4 aquara water sensors started reporting water shortly after update to HA addon 6.20.0 yesterday (deCONZ 2.21.2->2.22.2) running with ConBee II

If additional logs are necessary I can extract them

ebaauw, 2023-06-27:
We changed the DDF in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6919 to use the Xiaomi special attribute report next to the _IAS Zone_.  This PR was merged in v2.22.1-beta.  I’ve been running that version since it’s release.  Haven’t upgraded to v2.22.2 yet, but I don’t think that version has additional changes for the water leak sensor.

Pressing the button causes a special attribute report to be sent.  Could you turn on logging (I think it’s APS level 2) so deCONZ lists the payload of incoming messages on a separate line with `asdu` and list the Xiaomi special attribute payload?  Also, what `swversion` does the sensor report?  Is that the same of the sensors not reporting false positives?

I think I saw some false positives just after installing v2.22.1-beta, but these were caused by an almost empty battery in the sensor.  The value of `config.battery` for these (and many other Xiaomi button cell powered) sensors is utterly useless for detecting this - I think we should expose the voltage as reported by the sensor in addition to the (wrongly) extrapolated level.  Anyway, changing the battery solved my issue, and I haven’t seen any false positives since.

Make sure to test the new battery before installing it to the sensor - it should deliver at least 3.0V.  I’ve seen new button cell batteries that already lost too much power during storage.

S474N, 2023-06-27:
@bennydiamond  same problem as me. but unfortunately I didn't manage to fill out the form, so it was closed to me. By the way, I first reported on Discord a month ago...

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7059

ebaauw, 2023-06-27:
The sensor reports _Model Identifier_ (0x0005) and the special attribute (0xFF01) in one message, in response to pressing the button.
```
attr tp  l                           model identifier
---- -- -- ------------------------------------------
0500 42 15 6c756d692e73656e736f725f776c65616b2e617131
        21  l u m i . s e n s o r _ w l e a k . a q 1

                                   1                       2                       3
            1 2  3 4  5 6  7  8 9  0 1  2 3  4 5  6 7  8 9 0 1 2  3 4  5 6  7 8  9 0  1 2  3
attr tp  l   battery    temp         ?   RSSI dB             LQI  firmware    parent    leak
---- -- -- --------- ------- --------- --------- --------------- --------- --------- -------
01ff 42 22 0121 e50b 0328 1a 0421 a843 0521 1b00 0624 0000000000 0821 0402 0a21 b3f4 6410 01
        34    3.045V    26°C                                    0.0.0_0004              leak
```
deCONZ is working as expected; the sensor does actually report leak detected through the XIaomi special attribute.  Battery should be OK at 3.045V, though.  Note the older firmware (my newest sensor has 0.0.0_0006, however this model doesn't seem to support firmware updates, not even through the Aqara M2 hub).  Also note that the reported length of the special attribute value is wrong (34 bytes reported vs 33 bytes sent), but that's the case on my sensors as well.

Did you try shorting the contacts for a while?  Does the detection state change when you remove the shortcut?  I vaguely remember messing with the sensor for a bit, after replacing its battery, but I haven't kept a log of what I did exactly.

If everything else fails, you could delete the `state/water_bis` item from the DDF, so deCONZ ignores the leak state (tag 64) from the special attribute.

S474N, 2023-06-27:
Look at my closed bug report: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7059

I just wrote here that I have several water leakage sensors. Two of them have a different FW version than the other three (I also added screenshots there). Now older sensors are 0.0.0_0004, but were 20170721. Newest are now 0.0.0_0006, before upgrade 20200310.

As soon as I'm at the device, I'll try artificially shorting the sensor and send the log if there's any difference in it.

I'm sorry, but I don't know what DDF is and how I can delete the given parameter from it, I'm just a user ;)

bcutter, 2023-07-02:
Please see https://forum.phoscon.de/t/aqara-leak-sensor-new-firmware-unavailable-entities-in-ha/3702/14.

Summary: since updating to 2.22.2 (deCONZ addon for Home Assistant) and restarting the container, two entities are missing:
[image]

Where's the old DDF? I'd like to use that.

Oh wait: why has this been removed?!?

> Remove the non-functional config/enrolled, config/pending, **state/lowbattery**, and **state/tampered** that the legacy code exposed. (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6919)

Guys I spent hours trying to bring things back and now I read "ah, well, it has been removed" - which is NOT listed as "breaking change" but as "ENHANCEMENT" (`Refactor Xiaomi special attribute handling in DDF https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6919`) in https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.22.1-beta. **Explanation please.** 😠

SwoopX, 2023-07-02:
Well, imho, it's not a breaking change as the PR doesn't break the two items from the screenshot above. You cannot break what non-existent on the device.

bcutter, 2023-07-02:
> Well, imho, it's not a breaking change as the PR doesn't break the two items from the screenshot above. You cannot break what non-existent on the device.

So there's never been e. g. a low battery feature? What I see is:
- before the update: all entities available and working
- after the update: two entities unavailable

So technically something broke. If it never was provided from the hardware side so why was it provided in the past... than the PR would be a correction which is good - but **_the communication is not sufficient_**, that's the point.

SwoopX, 2023-07-02:
I'm sorry, but again: you cannot break what's not there. Of course, the question is valid to ask why it was included in the first place but you probably will never get a (satisfying) answer: it was added by an inexperienced user just expecting the feature to be available, items were added as defaults to a certain device category automatically, features were assumed based on the relevant cluster and the device following zigbee standards, accidentally, etc... we're all humans in the end. Corrections/updates on devices happen all the time, similarly to other projects.

They way of communication or level of detail, respectively, is certainly always debatable. In my view, this is also not given for other projects, but I might be wrong here. Not sure if mentioning more details in the release notes would be a silver bullet as they are not frequently checked in my experience.

Anyway, we're drifting away from the topic of this issue, so we should keep it focussed.

andys73, 2023-07-03:
The problem is not the tampered and low battery entities, is the most important one which is leak detected! :)
Like I have detailed in https://forum.phoscon.de/t/aqara-leak-sensor-new-firmware-unavailable-entities-in-ha/3702/14 the newer firmware 0.0.0_0006 leak sensor have this problem. The sensors with 0.0.0_0004 firmware are fine.

mirml, 2023-07-03:
I've got same problem.

bennydiamond, 2023-07-03:
> The problem is not the tampered and low battery entities, is the most important one which is leak detected! :) Like I have detailed in [...]

I only have sensors with version 0.0.0_0004 firmware and they exhibit the issue.

pergolafabio, 2023-07-04:
Is it possible this issue is also present on Xiaomi motion sensors, seeing same thing here, I'm not home, and it gives detected state every 2 hours...

[image]

mirml, 2023-07-04:
> The problem is not the tampered and low battery entities, is the most important one which is leak detected! :) Like I have detailed in [...]

I have sensor with version 0.0.0_0003 and that problem.

pergolafabio, 2023-07-04:
I have 4 Xiaomi motion sensors, they all give me now false detections, one almost every hour, the other one like every 6 hours... Seems all happening after the last update (using ha addon )

SwoopX, 2023-07-04:
@pergolafabio please do not highjack this issue, however, feel free to create new one. Thanks!

pergolafabio, 2023-07-04:
> @pergolafabio please do not highjack this issue, however, feel free to create new one. Thanks!

I thought it's maybe related to Xiaomi detection in general?

Mimiix, 2023-07-04:
> > @pergolafabio please do not highjack this issue, however, feel free to create new one. Thanks!
> 
> I thought it's maybe related to Xiaomi detection in general?

Its not. We recommend opening a new one.

ebaauw, 2023-07-04:
Got a (presumably false) notification from my 0.0.0_0004 sensor today.

Looking at the legacy code, we used to ignore the Xiaomi special attribute for this sensor.  So it shouldn't have been added it to the DDF - my bad.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2f021e06df4d5e8f1684971ab6c8db6bcc50de2b/xiaomi.cpp#L888

I'll prep a PR.  In the meantime, as workaround, you can remove the `state/water_bis` item from the DDF.

S474N, 2023-07-07:
I tested on all affected sensors to trigger an artificial alarm by connecting both sensors. Since then, I have had peace and no flood has been reported.

Unfortunately, now I started constantly reporting a problem with the Aqara vibration sensor. 
[image]

Mimiix, 2023-07-08:
> I tested on all affected sensors to trigger an artificial alarm by connecting both sensors. Since then, I have had peace and no flood has been reported.
> 
> Unfortunately, now I started constantly reporting a problem with the Aqara vibration sensor. [image]

Please open another report, following the template, for the vibration.

S474N, 2023-07-09:
> 

...until restart docker. Now is it back.

How can I edit DDF please? Right click, `Edit DDF` and Remove `water_bis` isn't work.

[image]

 Tested also `Save`, but...:
[image]

martind1983, 2023-07-20:
I only have sensors with version 0.0.0_0004 firmware and the same issue since upgrade....

S474N, 2023-07-20:
martind1983: it's only two months from first report :) 

I finally solved it by switching to another HW and Zigbee2MQTT.

closed 2023-07-20 by manup

mirml, 2023-08-04:
deConz 2.22.2 (HA addon)

I don't have this location:
/usr/share/deCONZ/devices/xiaomi/lumi_sensor_wleak_aq1.json

dxa13, 2023-08-31:
> Got a (presumably false) notification from my 0.0.0_0004 sensor today.
>
> Looking at the legacy code, we used to ignore the Xiaomi special attribute for this sensor. So it shouldn't have been added it [...]

Hi, I tried using your fix with the 2.23.0 beta branch, and my one leak sensor with firmware 0.0.0_0004 (all other 9 I have have firmware 6) is still reporting a false leak.  Is that expected after the fix? Is there a way to make the false leak go away without physical interaction of the sensor (i.e. shorting the leads, or re-pair it)?

andys73, 2023-09-19:
Just to mention the Aqara leak sensor does not have a firmware version issue. The issue with missing entities for leak and door sensors is caused by the bad ddf in the Lidl watering system see https://forum.phoscon.de/t/aqara-door-sensors-entities-not-available-in-ha/3949/16
You will have to remove the LIDL watering system and re-add it after the next release where it has been fixed.

SwoopX, 2023-09-19:
Hi and thanks for mentioning. However, we should highlight that a missing property of a different device does not affect the Xiaomi leak sensor and therefore, deconz REST API in any way. If a REST API clients gets confused based on that, it's unfortunate.

martind1983, 2023-10-25:
Any news on this topic? I did a complete new system installation, but still having the issue with my water leak sensors (both with firmare 0004) reporting a false leak. Initially I thought they are working now, but after a reboot of the system, since more than 1 hour false leak reported.
My Gateway 2.24.1 / Firmware 26780700
According to the release notes, this was / should be already included in my 2.24.1 version? or am I wrong?

andys73, 2023-10-26:
I have 10 leak sensors and none of them received any fake alert.
I am running the stable release 
version 2.22.02 / firmware 26780700