dresden-elektronik/deconz-rest-plugin #7064: Silvercrest Prise connectée

KatyPurry, 2023-06-28:
## Device

- Product name: Silvercrest Prise connectée (only written in French, I don't found the English name)
- Manufacturer: _TZ3000_ynmowqk2
- Model identifier:  TS011F
- Device type : Please remove all unrelated device types. 
  - Router

## Screenshots

[image]

[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering
[image]

### Diagnostics

### Electrical Measurement
[image]
[image]
[image]
[image]

BabaIsYou, 2023-06-29:
This device seems already supported :  https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/lidl/hg08673.json.

Do you have some trouble ?

KatyPurry, 2023-06-29:
Thank you for your answer, i already checked Silver but it doesn't change. Is there anything else I need to do? 

> After changing that, you can simply load your DDF and "Hot Reload" to make it work.

What does it means ? How "load your DDF" ? Sorry English isn't my native language

[image]

BabaIsYou, 2023-06-29:
Select your device, then Ctrl + E, ` File` / `Open` / select your DDF file / `File` / `Hot relaod` and verifiy that the DDF is loaded without error at the bootom of the screen ...

KatyPurry, 2023-06-29:
Thank you very much ! It’s working 
[image]

Mimiix, 2023-06-30:
Case closed :)

closed 2023-06-30 by Mimiix