dresden-elektronik/deconz-rest-plugin #7094: Tuya Fingerbot Plus

beyondzvincent, 2023-07-08:
- Product name: Fingerbot Plus
- Manufacturer: Tuya
- Model identifier: Ts0001
- Device type :  Switch

i first add new switch in phoscon, then press pair in device, no response for 3 mins, but when i go to decoz, it appear as 0xXXX, i click on it i can control the on off, but not appear at phoscon so also not in HA.

[image]
[image]
[image]
[image]

Smanar, 2023-07-08:
It's normal, device not supported, but need more information.
Wired or battery device ?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Mimiix, 2023-07-08:
Please add all screenshots.

beyondzvincent, 2023-07-09:
its a battery device
[image]
[image]

Mimiix, 2023-07-09:
We need screenshots of all clusters. Also, please add them in the first post.

beyondzvincent, 2023-07-09:
edited

beyondzvincent, 2023-07-09:
added new pic

Smanar, 2023-07-09:
Hello, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_dse8ogfy",
  "modelid": "TS0001",
  "vendor": "Tuya",
  "product": "Fingerbot",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Support for on/off and battery

Smanar, 2023-07-10:
Ha, my bad the request use neum type and not bool.
Try with replacing

```
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
```

by

```
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val==1;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x30", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
```

Smanar, 2023-07-11:
>00:21:16:516 Tuya write expression: Item.val == 1 ? 1 : 0; --> 0

And you have the reverse when using "on=true" instead of "on=false" ?

Something you can do, is enabling/disabling the device yourself using "manual mode". You will see tuya return, perhaps need to use another dpid.

beyondzvincent, 2023-07-12:
> > 00:21:16:516 Tuya write expression: Item.val == 1 ? 1 : 0; --> 0
> 
> And you have the reverse when using "on=true" instead of "on=false" ?
> 
> Something you can do, is enabling/disabling the device yourself using "manual mode". You will see tuya return (with "info"), perhaps need to use another dpid.

22:55:58:959 SC tick --> StateCallFunction
22:55:58:960 writeTuyaData, dpid: 0x01, type: 0x30, expr: Item.val == 1 ? 1 : 0;
22:55:58:960 Tuya write expression: Item.val == 1 ? 1 : 0; --> 1
22:55:59:249 ZCL attribute report 0x50325FFFFE29D2E3 for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
22:56:04:386 emit event/tick: 0xa4c1385b20a60d38
22:56:04:386 SC tick --> StateCallFunction
22:56:04:387 writeTuyaData, dpid: 0x01, type: 0x30, expr: Item.val == 1 ? 1 : 0;
22:56:04:387 Tuya write expression: Item.val == 1 ? 1 : 0; --> 0

this is the log when i turn it on and off at HA, no response at the device, but when i turn it on and off at deconz page or at the device physical button, the log didnt show up anything about that but the device did work

Smanar, 2023-07-13:
>but when i turn it on and off at deconz page

You know how to do that on deconz ? using the tuya cluster ?

>or at the device physical button, the log didnt show up anything about that but the device did work

Not good, when you use it manualy, the device need to make a return to say his state have moved .....

Edit:
Or pehraps I m wronng and you are using the on/off cluster ? the 0x0006 ? if yes, the DDF need to be.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_dse8ogfy",
  "modelid": "TS0001",
  "vendor": "Tuya",
  "product": "Fingerbot",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 86400
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

Have set a bind for cluster 0x0006 but no reporting (hoping it was native like lot of Aqara device)

beyondzvincent, 2023-07-14:
> 

this works! thanks alot

aldoushx, 2023-07-16:
> > 
> 
> this works! thanks alot

Hi,
Can you tell us  which DDF is working ? 
Is Smanar's DDF (in his last edit) fully working ?

Because as you previously said, for me with this DDF, in deconz the fingerbot is displayed as a light, and in home assistant when you activate the fingerbot, it stays on, and do not go back to off (you have to activate it again so it goes off).

beyondzvincent, 2023-07-16:
> Hi, Can you tell us which DDF is working ? Is Smanar's DDF (in his last edit) fully working ?
>
> Because as you previously said, for me with this DDF, in deconz the fingerbot is displayed as a light, [...]

yes the last one he provided, now at the phoscon app it shows as light, but at ha it shows as switch. The situation that stays on is no response, not work at all, it happens at prevoius ddf but at the last ddf, it works
[image]

Smanar, 2023-07-17:
On the last DDF the device is reconised as "TYPE_ON_OFF_OUTPUT", so it's not realy a light for me, third app can display it as they want.

aldoushx, 2023-08-06:
Any idea on how to customize the up and down positions, and the sustain time with deconz ?

Smanar, 2023-08-06:
For the sustain time can try

```
        {
          "name":"config/delay",
          "read":{
            "fn":"none"
          },
          "write":{
            "dpid":103,
            "dt":"0x2b",
            "eval":"Item.val;",
            "fn":"tuya"
          },
          "parse":{
            "dpid":103,
            "eval":"Item.val = Attr.val;",
            "fn":"tuya"
          },
          "default":1
        },
```

But from that I m reading this device use raw mode, and it's not implemented in deconz (I have see another issue with the same problem today), 0x2b is for s32 value.

aldoushx, 2023-08-07:
No, the code below does not work for the sustain time (i tried 3 and 10 as values, no change on the sustain time)

I am OK with the test, I have a debian test machine available, but I will not have access to the fingerbot in 2 days for some time.

Smanar, 2023-08-07:
I m with another user that have a device that use raw value, but for him it's sure https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7151#issuecomment-1668076062

If you have the GUI, can you try to take a look in logs with flag "info" and "ddf" and try the "config/delay" request ?
We will see the request made by deconz and I hope a return from the device with the correct synthax to use.

aldoushx, 2023-08-07:
OK, I am not familiar with the use of the REST API so maybe I have done it wrong.

I sent the following command through the terminal after hot reloading the updated DDF (with the delay parameter) : 
curl -H 'Content-Type: application/json' -X PUT -d '{"delay": 5}' http://IP:port/api/****/lights/7/config

the following error was returned in the terminal : 
[{"error":{"address":"/lights/7/config/delay","description":"parameter, delay, not available","type":6}}]

Smanar, 2023-08-08:
How have you updated the DDF ? (not something easy on HA)
And the problem is "config" field are hidden on "light" device, so I don't thing they can be visible on the API ...

Else for me your request is fine.

BabaIsYou, 2023-08-17:
There is so much stuff than can be controlled for this device : https://github.com/Koenkk/zigbee2mqtt/issues/17841#issuecomment-1591643703 ;-)

jordibsk10, 2023-08-31:
Hi all,
I'd love to also be able to control this new device, but unfortunately I'm not that expert yet and I don't know exactly what to do to make it work like @beyondzvincent .

Can any one shortly guide me (and for sure others in the near future) on how to make it work?

Thank you all!!

Smanar, 2023-09-01:
Have you installed the DDF, what is your OS ?

jordibsk10, 2023-09-11:
Hello, @Smanar,
Apologies, I don't know what is the DDF :(
I'm using deconz version 6.20.0, with a conbee stick (with latest version). And for the Home Assistant version, I'm using:
Home Assistant 2023.9.1
Supervisor 2023.08.3
Operating System 10.5

As said, I can see it in the deConz application, but i can't see it anywhere else :(
[image]
It looks it doesn't connect to any router or the main zigbee stick.

Hope this helps and thank you very much for your help,
Jordi

Smanar, 2023-09-11:
You need a DDF to use this device, and as the support is not yet official you need to put the DDF yourself.
It's something easy, just need to create a text file, unfortunately you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Smanar, 2023-10-01:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_dse8ogfy",
  "modelid": "TS0001",
  "vendor": "Tuya",
  "product": "Fingerbot",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
   {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name":"config/delay",
          "read":{
            "fn":"none"
          },
          "write":{
            "dpid":103,
            "dt":"0x2b",
            "eval":"Item.val;",
            "fn":"tuya"
          },
          "parse":{
            "dpid":103,
            "eval":"Item.val = Attr.val;",
            "fn":"tuya"
          },
          "default":1
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

New DDF, this one, delete the old one before using it.
This one create the "light" entry but now with a "sensor" one. The ZHAbattery can have all "config" filed, for the moment have added the config/battery and config/delay.

closed 2023-10-31 by github-actions[bot]

KatyPurry, 2024-04-05:
I'm not a very tech/dev guy but I managed to get the device working by updating the DDF!

The strange thing is that it acts as a light in Phoscon / Homebridge. Wouldn't it be better to make it work as a switch? 

Using it as a light is more difficult for automations, because it triggers when I turn the light on, but also when I turn the light off. Whereas if it were used as a switch, one press would be enough to activate it and return it to its original state ?

Smanar, 2024-04-05:
To resume, "light" receive order, "sensor" send information.
You send order to this device for it move, and this one don't send information if you move it manualy, so it's a "light".

Powered switch are "light"
Battery switch are "sensor"

Smanar, 2024-04-05:
> In my case, is it possible, for example, for the arm not to move from ON to OFF?

So you want off to on ?

I m with someone in PM to add some feature to the device. Do you know if the "on with auto off" is working natively ? There is a field config/delay, but for me it miss a command to activate it.

KatyPurry, 2024-04-05:
> > In my case, is it possible, for example, for the arm not to move from ON to OFF?
> 
> So you want off to on ? I don't the see the problem, with "on" it move in a side, with "off" in the other side ?
> 
> I m with someone in PM to add some feature to the device. Do you know if the "on with auto off" is working natively ? There is a field config/delay, but for me it miss a command to activate it.

It doesn't work that way (if I understand what you're saying). For example, here's what happens when I do OFF to ON or ON to OFF in Homebridge / Homekit : 

https://github.com/dresden-elektronik/deconz-rest-plugin/assets/89610869/7d292b13-4d69-4c61-a088-747d1e524317

I want the movement to be from OFF to ON only, and for nothing to happen when it "switches off" (ON to OFF).

> I m with someone in PM to add some feature to the device. Do you know if the "on with auto off" is working natively ? There is a field config/delay, but for me it miss a command to activate it.

Sorry I'm this dumb guy, I'm afraid I can't help you for this question (or explain to me what I have to show you)

Mimiix, 2024-04-05:
> > > In my case, is it possible, for example, for the arm not to move from ON to OFF?
> > 
> > 
> > So you want off to on ? I don't the see the problem, with "on" it move in a side, with "off" in the other side ?
> 
> It doesn't work that way (if I understand what you're saying). For example, here's what happens when I do OFF to ON or ON to OFF in Homebridge / Homekit, the movement is always composed of the descent of the arm and its ascent. :
> 
>  8mb.video-z8M-4mhWsa2H.mp4 
> I want the movement to be from OFF to ON only, and for nothing to happen when it "switches off" (ON to OFF).
> 
> > I m with someone in PM to add some feature to the device. Do you know if the "on with auto off" is working natively ? There is a field config/delay, but for me it miss a command to activate it.
> 
> Sorry I'm this dumb guy, I'm afraid I can't help you for this question (or explain to me what I have to show you)

@ebaauw is this something that needs to be solved in deCONZ or Homebridge?

KatyPurry, 2024-04-05:
To explain, in my case it's for use on a computer screen's power button. I want the screen to turn off when I press a remote control. 

That's what I was explaining earlier, the fact that it's managed like a light isn't obvious, considering it as a "press" button would have been simpler (like the Aqara Mini Switch, for example), 1 press = a back and forth movement of the arm without having to manage the On/Off states like a light.

Smanar, 2024-04-06:
@Mimiix is right.

Can you try with phoscon, instead of Homebridge ?
It's possible the down/up mouvement is software coded. I have someone in PM that want the same working mode than you ^^ for him, he need to make 2 request to have the same result.

KatyPurry, 2024-04-06:
Nothing happens when I press the light on Phoscon.

Yes you're right, I want the "auto" mode :D

Smanar, 2024-04-06:
> Nothing happens when I press the light on Phoscon

Ha ? Nothing at all, In my mind the device need to make "UP" or "Down" but alone.

KatyPurry, 2024-04-06:
Sorry, I said the wrong thing. I don't usually use Phoscon. It gives the same behavior as Homebridge Down/Up on Off and Down/Up on On

Smanar, 2024-04-06:
Can you try this DDF (if you are using the one in this post for the "_TZ3210_dse8ogfy")

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_dse8ogfy",
  "modelid": "TS0001",
  "vendor": "Tuya",
  "product": "Fingerbot",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
   {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
     "meta": {
        "values": {
          "config/mode": {"clic": 0, "switch": 1, "program": 2}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["clic", 0], ["switch", 1], ["program", 2]
          ],
          "parse": {"fn": "tuya", "dpid": 101, "eval": "if (Attr.val == 0) { Item.val = 'clic' } else if (Attr.val == 1) { Item.val = 'switch' } else { Item.val = 'program' }"},
          "write": {"fn": "tuya", "dpid": 101, "dt": "0x30", "eval": "if (Item.val == 'clic') { 0 } else if (Item.val == 'switch') { 1 } else { 3 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name":"config/delay",
          "read":{
            "fn":"none"
          },
          "write":{
            "dpid":103,
            "dt":"0x2b",
            "eval":"Item.val;",
            "fn":"tuya"
          },
          "parse":{
            "dpid":103,
            "eval":"Item.val = Attr.val;",
            "fn":"tuya"
          },
          "default":1
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

You will have a "config/mode" fiel with 3 possibles values 

`["clic", 0], ["switch", 1], ["program", 2]`

KatyPurry, 2024-04-06:
Thank you ! I updated the DDF by your DDF, but now how to change the mode in homebridge/homekit ? 😅

Smanar, 2024-04-07:
Lol, depend of OS, if you have edited the one in the folder /usr/share/deCONZ/devices/ it will be replaced the next time you will update deconz, so no danger to make test with it.
After the change, better to relaod deconz, easier than using the DDF editor.

No need to re-include the device.

If you take a look in the API, for exemple with phoscon/help/debug view/sensor, you will found the new field on the ZHAbattery sensor.

To change the value, it depend of third app, IDK if you have a direct access to all fields, else you will need to use command line.

`curl -H 'Content-Type: application/json' -X PUT -d '{"mode": "switch"}' http://IP:PORT/api/APIKEY/sensors/ID/config`

https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

KatyPurry, 2024-04-07:
Thank you for this valuable information. I made the changes but I have errors. I'm sorry I'm that guy who's not very comfortable with all this.

To sum up, I use unRAID. To use your JSON file. I would connect with deCONZ VNC, locate the Fingerbot sensor > Edit DDF > load JSON file and finally I did a Hot Reload.

Then I ran the command: `curl -H 'Content-Type: application/json' -X PUT -d '{"mode": "switch"}' http://192.168.1.9:8020/api/***/sensors/39/config`, but it returned an error saying that the config field was not found.

I thought maybe I'd done the procedure wrong, so I went straight to updating the json file `/usr/share/deCONZ/devices/tuya/_TZ3210_dse8ogfy_fingerbot.json`. Edited with nano, written then quit. I restarted deCONZ but since then the Fingerbot no longer appears in Phoscon.

That's where I'm at.

Smanar, 2024-04-07:
>Edited with nano, written then quit. I restarted deCONZ but since then the Fingerbot no longer appears in Phoscon.

You probably have done a typo in the file

>but it returned an error saying that the config field was not found.

You are sure you are using the good ID ? You can check direcy in your browser `http://192.168.1.9:8020/api/***/sensors/39` you need to have a "config" part.

KatyPurry, 2024-04-07:
Okay, I've started the procedure again. I removed the Fingebot from DeCONZ VNC, stopped the deCONZ container and unplugged the Conbee stick for a few minutes. 

I added the Fingerbot on Phoscon. I edited the file with nano, this is the start of the file (I wrote the file of course):
[image]

I connected to deCONZ VNC for Hot Reload. Now, If I check directly in my browser, it displays : 

```
{
    "config": {
        "delay": 1,
        "on": true,
        "reachable": true
    },
    "etag": "1d7cc0bdfe6bd79c2c18ea0a4dc57ba8",
    "lastannounced": null,
    "lastseen": "2024-04-07T18:01Z",
    "manufacturername": "_TZ3210_j4pdtz9v",
    "modelid": "TS0001",
    "name": "Battery 44",
    "state": {
        "battery": 0,
        "lastupdated": "none"
    },
    "swversion": "1.1.2",
    "type": "ZHABattery",
    "uniqueid": "a4:c1:38:7e:ce:a3:b0:e8-01-0001"
}
```

If I check the DDF in deCONZ VNC, it doesn't seem to use the version I edited:
[image]

I don't dare restart the deCONZ container. What should I do?

KatyPurry, 2024-04-08:
Yes I'm using docker on [unRAID OS](https://unraid.net/).

By editor, you mean the DDF Editor in deCONZ GUI ? This was just to show you the json wasn't updated. I went through nano on terminal to edit the file.

I restarted the deCONZ container and checked what JSON file was used, this is the correct JSON you provided:
[image]

Checked API, no config mode:
[image]

Smanar, 2024-04-09:
So dockers have special working mode, I think you are using the official deconz docker image ?
So for DDF you need to use the folder "opt/deCONZ/devices/" https://github.com/deconz-community/deconz-docker/issues/163

KatyPurry, 2024-04-09:
> So dockers have special working mode, all is virtual, I think you are using the official deconz docker image ? 
I'm using the official one yes: `deconzcommunity/deconz:stable`

So for DDF you need to use the folder "opt/deCONZ/devices/" [deconz-community/deconz-docker#163](https://github.com/deconz-community/deconz-docker/issues/163)

I have on file in the folder (the one I uploaded via deCONZ VCNC GUI)
[image]

Smanar, 2024-04-09:
Strange the file is called "fingerbot1.json" but on the editor (on the window title" its the file "_TZ3210_dse8ogfy_fingerbot.json" that is used, and the one on the "opt" folder need to be prioritary.

KatyPurry, 2024-04-09:
> Strange the file is called "fingerbot1.json" but on the editor (on the window title) its the file "_TZ3210_dse8ogfy_fingerbot.json" that is used, and the one on the "opt" folder need to be prioritary.

These are not the same files. 
I used `fingerbot1.json` to test load from a file in deCONZ VNC GUI. Forget this file.
`_TZ3210_dse8ogfy_fingerbot.json` is when I modified file in `/usr/share/deCONZ/devices/tuya/_TZ3210_dse8ogfy_fingerbot.json`

And to make it even more difficult to understand, here is what is displayed after the restart of deCONZ container:
[image]

Smanar, 2024-04-09:
Your last picture is like deconz don't found DDF at all.
But on docker not possible to use "/usr/share/deCONZ/devices", this folder not exist in reality. Or you know how works dockers ? But I don't know what you have done. 

For me the better DDF you have is "fingerbot1.json" the one you have in the "opt" folder, but why this one is not picked by deconz ? You can try to load it using the editor, perhaps there is a typo ?
There is special logings in deconz for DDF (on help/debug view)

gabriele-latino, 2024-09-30:
A dumb question, it's possibile to set the fingerbot's arm lenght and timing, just like can be done by Tuya?

Smanar, 2024-09-30:
Can add a delay with for exemple

        {
          "name": "config/delay",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 103,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 103,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 5
        },

With value from 0 to 10.
Problem, from my memory the DDF core don't support raw value, to write, ok to read them.

For the lenght you are talking about the min/max ? And here I have a new problem, I don't see which one fild to use, can create a config/min and config/max, but I think it will cause problems with existing fields.

McElch, 2025-01-06:
Hello, i got a question regarding the way the Fingerbot works. I have a Raspberry with Docker running the official deconz Container and a ConBee II. With the Informations found here and the DDF found [there](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3210_dse8ogfy_fingerbot.json) i managed to get my brand new Zigbee Fingerbot to pair.

Whatever  I klick (On / Off / Toggle ) in the deconz GUI as in my Home Automation Software (FHEM), it works always the way "PowerOn - Auto PowerOff".

In the actual DDF is that delay-snippet included already, right? I modified the default value from 1 to 5, expecting that would cause the Fingerbot to stay longer (5 Seconds) in the On Position before returning to off, but it hast no effect.

I use it for the DoorOpener of my old Intercom. It works for me, but not for Visitors, as the On-Time is very short. To explain, what I want is a zzzzzzzzzzzzzzzzzzzzzt instead of a zzt - you all know that sound :)

This could be accomplished either by a on-for-timer Function (if on would move the Arm down an off would move it up again) of my smarthome, or with a slower return of the Fingerarm. For me it doesnt matter wich way of the two. How would you do that?

Thanks in advance! Great Infos and Support here!

McElch, 2025-01-07:
Thanks for your reply! Yes I wonder why i am the only one. Pretty sure, the Probleme is in front of screen... :)

To be honest, i had a very old deconz Version. Was suprised that it now know Blinds and Thermostaes...  hehe.. I didnd dare to touch it, sorry. Now i have 2.29.1 beta.

But the behavior stays same. And im still confused: With the new Version, it recognizes the Device. It does not use my JSON file, but  one that comes with the update. named tuy/....fingerbot.json. But that hasnt that ["clic", 0], ["switch", 1], ["program", 2] features included.
So i tried yours from here, but that did not work (device did not act, neither in Phoscon nor in my Smarthome).
And when I used yours with that ["clic", 0], ["switch", 1], ["program", 2], where should i look if the Sustain Value changes? And where to look for the used mode? Sorry for my confusion!

Smanar, 2025-01-08:
Ha you are right, the official one is lighter https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3210_dse8ogfy_fingerbot.json

So we can presume you are using native setting, can you take a look on logs when using the config/delay ? (with flag "DDF" and both "INFO").

>. It does not use my JSON file, but one that comes with the update

So ATM you have your json + the official one+ the one from this issue, what is your OS where have you put the DDF ? If you can edit the official one, it's better having only one, and after a change you need to restart deconz (it load DDF only at start)

McElch, 2025-01-08:
Hi, im running Raspian 11 bullseye with Docker (official deconz Image). For further testing, now I renamed the official _TZ3210......json in _TZ3210....bak. So now, deconz should always use my fingerbot3.json in opt/deconz/devices (for persitance).

Now i took the content of the official file into that place, only chaning the delay default to 5. Restarted deConz and checked if it uses my json now - it does.

Behavior is the same: whatever clicked in Phoscon or the GUI, the arm drives down und immidiatly up again. Here is what the Log says:

> 16:32:34:216 emit event/tick: 0xA4C138419B741F5A
16:32:34:235 SC tick --> StateCallFunction
16:32:34:238 SC tick --> StateCallFunction
16:32:37:042 	0xA4C138419B741F5A force poll (2)
16:32:37:076 a4:c1:38:41:9b:74:1f:5a-01/state/on expression: Item.val = Attr.val --> false
16:32:37:079 ZCL attribute report 0xA4C138419B741F5A for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
16:32:37:082 SC state/on: synced
16:32:37:083 SC --> StateFinished
16:32:37:084 SC --> StateFinished
16:32:37:085 SC state change finished: a4:c1:38:41:9b:74:1f:5a-01
16:32:37:086 SC state change finished: a4:c1:38:41:9b:74:1f:5a-01

And here is my json:

> {
  "schema": "devcap1.schema.json",
  "uuid": "aeb64a3c-50c7-45c7-9dca-6378a5004200",
  "manufacturername": [
    "_TZ3210_dse8ogfy",
    "_TZ3210_j4pdtz9v"
  ],
  "modelid": [
    "TS0001",
    "TS0001"
  ],
  "vendor": "Tuya",
  "product": "Adaprox fingerbot (TS0001)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "parse": {
            "fn": "tuya",
            "dpid": 105,
            "eval": "Item.val = Attr.val;"
          },
          "read": {
            "fn": "none"
          },
          "default": 0
        },
        {
          "name": "config/delay",
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 103,
            "dt": "0x2b",
            "eval": "Item.val;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 103,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 5
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}

Smanar, 2025-01-09:
This is fine for me, but I m hopping an error message when you try to use the config/delay , something like "bad format type".
With flag "Info" *2 and "DDF" on deconz/help/debug view, with the flag "DDF" you will be able to locate the command, and with both flag "info" you will see tuya request.

McElch, 2025-01-09:
Hmmm. What do you mean with "when i try to use the config/delay"? I just wrote it to the json, hoping it would use the default? Is that wrong? 

As far as i can see there are no errors, but for me its is extremely hard to follow up that flood of messages, and check if they are from that device / my clicks on it. :(

Smanar, are you working at DD-elektronik? Than i bring you the device, its not far from Laubegast! LOL

Smanar, 2025-01-10:
>Hmmm. What do you mean with "when i try to use the config/delay"? I just wrote it to the json, hoping it would use the default? Is that wrong?

Just use the API to change the value, like you have one for the value "5" before.

And no ^^, I don't work for DD

McElch, 2025-01-11:
Well, I think the Problem is me, not the Device. I didnt use either a API (wich?) nor the VNC-GUI! As written, I just changed the default value in the DDF file. Is that the mistake?

If you mean the deConz GUI, there are some buttons in the on/off Cluster Info. Do you mean this? Sorry for Layer 8 Problems! :)

Smanar, 2025-01-12:
Haaa, so yes ^^.

You need to make API call tochnage the value.
Depend on third app, some of them permit to change field else can use curl, with something like

`curl -H 'Content-Type: application/json' -X PUT -d '{"duration": 900}' http://192.168.1.1:80/api/36E138295E/sensors/16/config`

https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

[5 more replies]