dresden-elektronik/deconz-rest-plugin #7103: GiEX QT06 Smart irrigation valve

bennydiamond, 2023-07-12:
## Device

- Product name: GiEX QT06 smart irrigation valve
- Manufacturer: _TZE200_a7sghmms
- Model identifier: TS0601
- Device type : Irrigation Valve
  - Sensor
  - Switch

## Screenshots

[image]
[image]
[image]

### Basic
[image]

bennydiamond, 2023-07-12:
Device already supported and documented in other Zigbee software suites:
https://zigbee.blakadder.com/GiEX_QT06.html
https://www.zigbee2mqtt.io/devices/QT06_2.html

bennydiamond, 2023-07-12:
Thank, pardon my ignorance but I have no clue on how to do that!
I'm running the deCONZ official addon for Hassio.
I tryed adding a json file in the container at location `/usr/share/deCONZ/devices/tuya` but it is deleted on addon restart...

BEskandari, 2023-07-12:
Try this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6944
A Pull request was submitted https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6947

closed 2023-07-12 by Mimiix

milkplus27, 2023-07-18:
> Thank, pardon my ignorance but I have no clue on how to do that! I'm running the deCONZ official addon for Hassio. I tryed adding a json file in the container at location `/usr/share/deCONZ/devices/tuya` but it is deleted on addon restart...

hey there! I've also just purchased one of these GiEx Zigbee water valves and am running deCONZ in HA via a Conbee II - but seem incapable of getting it to pair. Did you have any luck in the end?

bennydiamond, 2023-07-19:
Follow this video tutorial: https://www.youtube.com/watch?v=rgTQpOVg8w0

If you're using the Official HA deConz addon. Add the DDF file from above in the following relative directory path from the root of your Home Assistant config folder: `addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices/tuya`. You should only have to create the `devices` and `tuya`(optional) folder. All the other ones should already be there.

milkplus27, 2023-07-19:
Awesome, thank you! I will give this a go. Then once the DDF file is added, do you have to load it from within deCONZ? And afterwards do you add the Giex valve as a an 'Other' switch in Phoscon?

milkplus27, 2023-07-19:
> Add the DDF file from above in the following relative directory path from the root of your Home Assistant config folder: `addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices/tuya`

hmm, these folders don't seem to exist. the addons folder in my HA config folder is empty. which tbh... seems a bit odd as i have a bunch of addons

milkplus27, 2023-07-19:
oh, you know what? I managed to figure it out... just has to delete the node from deCONZ and re-add and now it's showing up in HA. excellent. thanks for the assist :)