dresden-elektronik/deconz-rest-plugin #7104: Gira FoH longpress not working properly

noodles321, 2023-07-13:
## Describe the bug
I have a couple of Friends of Hue Switches from Gira. If I longpress them, there will be a log entry in my openhab instance. So I guess the longpress was recognized. But there is rarely a dimming process working. It ist all configured via phoscon, but it does not really work.

## Steps to reproduce the behavior
90% of the time I longpress any Switch. It ist not necessary If the switch ist nearby or not. Openhab log shows me always it is recognized but there is no dimming 

## Expected behavior
Dimming lights as configured in phoscon App.

## Environment
deCONZ Version 2.22.02
Conbee II Firmware 26720700

 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container 
 - Firmware version: 26720700
 - deCONZ version: 2.22.02
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

noodles321, 2023-07-13:
Hey, Here is the result. I switched the light in, then tried top dimm 2 or 3 Times down, but norhing happens. Than I switched the light off.
```
{
    "18:54:27:196": {
        "attr": {
            "id": "4",
            "lastannounced": "2023-07-12T23:17:50Z",
            "lastseen": "2023-07-13T16:54Z",
            "manufacturername": "innr",
            "modelid": "RF 263",
            "name": "Wohnzimmer1",
            "swversion": "2.1",
            "type": "Dimmable light",
            "uniqueid": "60:a4:23:ff:fe:07:2b:41-01"
        },
        "e": "changed",
        "id": "4",
        "r": "lights",
        "t": "event",
        "uniqueid": "60:a4:23:ff:fe:07:2b:41-01"
    },
    "18:54:27:249": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 1000,
            "lastupdated": "2023-07-13T16:54:27.360"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:32:319": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 1003,
            "lastupdated": "2023-07-13T16:54:32.249"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:33:966": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2000,
            "lastupdated": "2023-07-13T16:54:33.988"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:39:193": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2003,
            "lastupdated": "2023-07-13T16:54:39.253"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:40:340": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2000,
            "lastupdated": "2023-07-13T16:54:40.367"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:45:33": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2003,
            "lastupdated": "2023-07-13T16:54:45.058"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:45:742": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2000,
            "lastupdated": "2023-07-13T16:54:45.757"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:51:478": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2003,
            "lastupdated": "2023-07-13T16:54:51.517"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:53:119": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2000,
            "lastupdated": "2023-07-13T16:54:53.202"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:53:273": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2002,
            "lastupdated": "2023-07-13T16:54:53.385"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "18:54:53:291": {
        "e": "changed",
        "id": "4",
        "r": "lights",
        "state": {
            "alert": "none",
            "bri": 134,
            "on": false,
            "reachable": true
        },
        "t": "event",
        "uniqueid": "60:a4:23:ff:fe:07:2b:41-01"
    }
}
```

Mimiix, 2023-07-13:
The 1003 seems to indicate something with long press. Can you confirm?

noodles321, 2023-07-13:
The xxx0 Events are on, the xxx2 are Off and the xxx3 seem top be dimming, but it does not work and I can't be clear for now. I Just noticed that a longpress results in a xxx0 immediately and gives a xxx3 after releasing. Maybe this ist where the Problems begin, because longpress is recognized after i needed it?

Mimiix, 2023-07-13:
Does it keep spamming the xxx0?

noodles321, 2023-07-13:
Here I got it to Work! I am really confused sorry. It appears that a xxx1 event hast top be there.
The light was on and I pressed and held the Button.

```{
    "19:22:05:389": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2000,
            "lastupdated": "2023-07-13T17:22:05.455"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "19:22:11:446": {
        "e": "changed",
        "id": "4",
        "r": "sensors",
        "state": {
            "buttonevent": 2001,
            "lastupdated": "2023-07-13T17:22:11.552"
        },
        "t": "event",
        "uniqueid": "00:00:00:00:01:71:66:99-f2"
    },
    "19:22:11:464": {
        "e": "changed",
        "id": "4",
        "r": "lights",
        "state": {
            "alert": "none",
            "bri": 1,
            "on": true,
            "reachable": true
        },
        "t": "event",
        "uniqueid": "60:a4:23:ff:fe:07:2b:41-01"
    }
}

```

noodles321, 2023-07-13:
> Does it keep spamming the xxx0?

No. I just tried top get it to Work. It is really really rare that it works so that I were on the wrong way as I thought xxx3 is the event I need. Sorry for confusing.

Mimiix, 2023-07-13:
In that case , nothing we can do. The correct event is emitted to phoscon when the device sends it. 

As it's not a bug, but simply a limitation of the device firmware, I'm closing this issue.

noodles321, 2023-07-13:
Oh but it worked on older Versions with as I used it as raspbian Version without docker. I don't know the exact number but it worked

Mimiix, 2023-07-13:
Even if its an isue with phoscon, you need to report it on the forums.

For now, seems that deconz is handling it correctly.

manup, 2023-07-14:
The device should be identical to the Philips FoH (same module). I've forwarded the issue for retesting.

From the long press events perspective the following should be seen:

```
x0001 Hold
x0003 Long release
```

noodles321, 2023-07-14:
Thank you. It still works with a Hue Bridge so I guess it is not the Switch.