dresden-elektronik/deconz-rest-plugin #7106: LCL003 light goes zombie mode

posixx, 2023-07-14:
## Describe the bug

I have 3 LCL003 (philips hue outdoor strip) lights. One of them is not reacting to commands, but still reports as available in deconz / phoscon. it does not react to any command but is still reported: zombie mode
## Steps to reproduce the behavior

after adding the light it works for a couple of hours. After that, on / off / color change etc do not work. the only way to activate the light again is to power cycle or remove from deconz and readd.
## Expected behavior

It should work the same as the other 2 LCL003 lights i have
## Screenshots

## Environment

version 2.22.02 / 19-9-2022
Firmware 26720700

 - Host system: NUC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2023.7.2
 - Device:  ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

I have bought a new LCL003 light and added it to Deconz. After a couple of hours; SAME ISSUE! light is going into zombie mode. so it definitely is related to deconz!

ebaauw, 2023-07-14:
I’ve seen this on a couple of different Zigbee 3 Hue lights, but also with firmware image type 0x0114.  Intermittently, they would enter a state where would continue to react to Zigbee commands as expected, but not react physically.  Power cycling the light cured it.  It’s a bug in (an older previous version of) the light firmware.  Not sure what’s triggering this bug; but it could be related to deCONZ using attribute reporting, where the Hue bridge still polls these lights.

My issues stopped after upgrading to newer light firmware.  Not sure if that was v1.101.10 or v1.104.2 (the [release notes](https://www.philips-hue.com/en-us/support/release-notes/lamps) are not very helpful).

posixx, 2023-07-15:
I have no idea how to update this firmware. Also the lights are listed as bluetooth but i cannot get them to become visible / connect to hue app...

posixx, 2023-07-15:
Ok, found the latest firmware file i can find here: https://github.com/dresden-elektronik/deconz-ota-plugin#hue-firmware (1.93.7). But as i am using deconz as home assistant addon; how to upload the firmware file to the addon storage so i can selet it on the OTAU plugin screen?

posixx, 2023-07-15:
Unfortunately after upgrading the lights, again the one light goes into zombie mode again!

ebaauw, 2023-07-16:
That’s not the latest firmware, see the release notes I linked earlier.  The README of the OTA plugin seems hopelessly outdated.

The Hue app should update to the latest firmware, but this might require several iterations, as older firmwares sometimes cannot be updated to the latest directly.  Some lights actually change _Image Type_ during one of these steps.  I don’t have much experience accessing the lights over BLE, but I think you can only discover and pair them shortly after power cycling them.

The best source for the latest firmware probably is: `https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json`, but you need to do some decimal - hex conversion.

posixx, 2023-07-16:
Yes you're right; all lights which were updated (LCL003 (outdoor strip) and 929003553001 (resonate downward)) to 1.93.7 now list 0x11f as image type. Will try and update them using latest 0x11f image from koenkk source

posixx, 2023-07-20:
Ok, what i tried:

- removed the problematic light from deconz for about 2 days (light goes on after removing so it reacts fine)
- removed the group + scenes where the light was a member of
- readded the light, created the group again and added the scenes

Again, the same problem. This is completely unlogical to me. Going on vacation now to get the frustration out of my head,.

closed 2023-08-18 by github-actions[bot]