dresden-elektronik/deconz-rest-plugin #7107: Humidity no longer retrieved after updating to 2.22.02 (TS0601/_TZE200_bjawzodf)

c-hoeger, 2023-07-15:
## Describe the bug

After updating to 2.22.02, Humidity of sensor with `modelid TS0601`, `manufacturername _TZE200_bjawzodf` does no longer retrieved while temperature is still working.

## Steps to reproduce the behavior

Update to 2.22.02 and retrieve data using curl.
The lastupdated date and time stays at the date and time deconz had been updated to 2.22.02.

Once I downgrade to 2.21.02, it starts working again.

## Expected behavior

Humidity should still be retrieved.

## Screenshots

```json
  "8": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "etag": "38850681968ed435eb5c21813572271f",
    "lastannounced": null,
    "lastseen": "2023-07-15T08:56Z",
    "manufacturername": "_TZE200_bjawzodf",
    "modelid": "TS0601",
    "name": "Humidity 8",
    "state": {
      "humidity": 5160,
      "lastupdated": "2023-07-15T08:57:01.078"
    },
    "swversion": "1.0.6",
    "type": "ZHAHumidity",
    "uniqueid": "a4:c1:38:9f:fd:0f:b2:0f-01-0405"
  },
  "9": {
    "config": {
      "battery": 100,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "etag": "0135be6960f6eff834d4036831fdd392",
    "lastannounced": null,
    "lastseen": "2023-07-15T08:56Z",
    "manufacturername": "_TZE200_bjawzodf",
    "modelid": "TS0601",
    "name": "Temperature 9",
    "state": {
      "lastupdated": "2023-07-15T08:56:55.880",
      "temperature": 2580
    },
    "swversion": "1.0.6",
    "type": "ZHATemperature",
    "uniqueid": "a4:c1:38:9f:fd:0f:b2:0f-01-0402"
  }
```

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26400500
 - deCONZ version: 2.22.02
 - Device: RaspBee
 - Do you use an USB extension cable: no
[... 876 lines cut ...]
        },
        "FF81BB75AD": {
            "create date": "2023-04-11T18:20:44",
            "last use date": "2023-04-11T18:21:15",
            "name": "Phoscon#B1194x731"
        },
        "FF8776F80D": {
            "create date": "2022-08-20T10:09:23",
            "last use date": "2022-08-20T10:10:35",
            "name": "Phoscon#B1785x1288"
        }
    },
    "wifi": "not-configured",
    "wifiavailable": [],
    "wifichannel": "1",
    "wificlientname": null,
    "wifiip": "192.168.8.1",
    "wifimgmt": 0,
    "wifiname": null,
    "wifitype": "accesspoint",
    "zigbeechannel": 15
}
```

## Additional context

Smanar, 2023-07-15:
Hello, ha yes, this device don't have same convertion than other, other use a *100 multiplier and this one a * 10.
Can you try  this DDF (need to use it on "user" folder to overwrite the deconz one)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_bjawzodf",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

c-hoeger, 2023-07-15:
That did the trick, thanks!

Smanar, 2023-07-15:
No problem on temperature ? (values are not  /10 ?)
Because I have do same, but your previous value seem good ?

c-hoeger, 2023-07-16:
> No problem on temperature ? (values are not  /10 ?)
> 
> Because I have do same for temperature and humidity, and your previous value seem good ?

No, the temperature is and was transferred correctly!

Smanar, 2023-07-16:
Thx, have created a PR to correct the issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7109

closed 2023-07-20 by manup