dresden-elektronik/deconz-rest-plugin #7113: Tuya TS0201 _TZ3000_0s1izerx temperature & humidity sensor

JeroenvIS, 2023-07-18:
## Device

- Product name: Temperature & Humidity Sensor (Powered by Tuya)
- Manufacturer: _TZ3000_0s1izerx
- Model identifier: TS0201
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

[image]

[image]

### Basic
[image]

### Identify
[image]

### Device Temperature
(Temperature measurement cluster)
[image]

### Relative humidity measurement
[image]

### Power Configuration
[image]

## Additional comments
Powered by two AAA batteries. Looks the same as the sensor in #6582 .

Smanar, 2023-07-19:
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_0s1izerx",
  "modelid": "TS0201",
  "vendor": "Tuya",
  "product": "Temperature and humidity sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 84600,
          "change": "0x00000002"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

From what I m reading you willl need a convertion for humidity (*10 or / 10), so just give us the value you have with this DDF.

JeroenvIS, 2023-07-20:
Thanks @Smanar . I'm running DeCONZ as an add-on in Home Assistant; how can I best go about to add this file and where can I put it? Before opening this request I tried copying an existing DDF for a similar sensor and adjusting the manufacturername, but it wasn't picked up - hence my question.

Alternatively to making local DDF changes in the running container: is it an option to include this in a new release (possibly with "silver" status)?

Smanar, 2023-07-20:
We generaly prefer to test them.
For exemple for this devive I can't guess if I m right for humidity without test.

On HA :
- short story, the DDF need to be in /data/.local/share/dresden-elektronik/deCONZ/devices/
- long story https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

JeroenvIS, 2023-07-23:
OK, I haven't managed to get SSH access yet, but I did manage to get deCONZ to see the sensor:
- I selected a different sensor (a similar one, identified as "_TZ3000_dowj6gyi"
- Selected "Edit DDF" in the Edit menu
- Did a "Save as" to /data/.local/share/dresden-elektronik/deCONZ/devices/_TZ3000_0s1izerx.json (effectively copying the existing DDF for that other sensor)
- Modified the file in the DDF editor wrt manufacturer name and product, saved it and requested a hot reload

The sensor now looks fine in deCONZ and it shows up in Phoscon. Hasn't appeared in Home Assistant yet, but perhaps I should give that some time.

Smanar, 2023-07-24:
Phoscon and HA are using same the API. And HA have more control than Phoscon, so yes, not possible having something in phoscon and not HA.

JeroenvIS, 2023-08-05:
This is the DDF that I've been using for the past two weeks:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_0s1izerx",
  "modelid": "TS0201",
  "vendor": "Tuya",
  "product": "Temperature and humidity sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

I have two of these sensors and they both look fine in Home Assistant with DeCONZ and this DDF.

Smanar, 2023-08-06:
Nice thx ^^.
Do you want to make the PR or want I make it to add thoses devices ?

BabaIsYou, 2023-08-07:
May be that it could just be merged to this one ? : https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0201_temp_hum_sensor.json

Smanar, 2023-08-08:
Yes, For me the code is exactly the same.

JeroenvIS, 2023-08-08:
I didn't know that the DDFs supported this (but I'm very much a newbie in this area). Then indeed that looks like the cleanest solution. I went ahead and created a PR, hope that it's OK like this,

Smanar, 2023-08-08:
Perfect ^^

closed 2023-08-11 by manup