dresden-elektronik/deconz-rest-plugin #7115: 3 gang scene switch - TUYA - TZ3000_TS0043

Dewcal, 2023-07-19:
## Device

- Product name:  3 gang scene switch -TZ3000_TS0043_Switch with battery
- Manufacturer: _TZ3000_pkeqinnt
- Model identifier: TS0043
- Device type : RFD
  - Switch
 

## Screenshots
[image]

[image]

[image]

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2023-07-19:
So 2 thing to do

Can create this DDF or edit the devices\tuya\_TZ3000_TS0043_3gang_remote.json

{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_gbm10jnj", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_famkxci2", "_TZ3000_w3c7ouru", "_TZ3000_pkeqinnt"],
  "modelid": ["TS0043", "TS0043", "TS0043", "TS0043", "TS0043", "TS0043"],
  "product": "Tuya 3 gangs switch battery",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },        
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },    
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}

And need to edit the button_maps.json file

        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae", "_TZ3000_pkeqinnt"],
            "map": [

Dewcal, 2023-07-19:
Many thanks for these suggestions.  I created a new ….remote.json file (having saved the original as .OLD) and put it in the users/  folder.

 

I then added "_TZ3000_pkeqinnt" to the “maps” file in the same folder.  Exited and restarted deCONZ and am pleased to say that I now have “Switch 87” appearing.

 

Many thanks for this.  I have not tried to programme any of the three switches but did not when I looked at the switch in HomeSeer, it did show four buttons!  Pleased to say that battery percentage is also visible.

 

Once again many thanks for this and I will try programming tomorrow.

 

 

Best regards,

David Williams

Mobile: +447802785789

Dewcal, 2023-07-20:
Many thanks for these suggestions.  I created a new ….remote.json file (having saved the original as .OLD) and put it in the users/   folder.

I then added "_TZ3000_pkeqinnt" to the “maps” file in the same folder.  Exited and restarted deCONZ and am pleased to say that I now have “Switch 87” appearing.

Many thanks for this.  I have now programmed the switch with 9 events / scenes and all are working - despite the fact that there appear to be four switches available!  Pleased to say that battery percentage is also visible.

Dewcal, 2023-07-20:
Many thanks again fro such prompt help.

Smanar, 2023-07-20:
Ha yes BTW, someone want to make the PR ?

closed 2023-08-05 by manup