dresden-elektronik/deconz-rest-plugin #7124: macOS compiled with CMake is crashing on launch

aryelevin, 2023-07-23:
## Describe the bug
Compiling with CMake on macOS leads to a crash duo to "Class XXXXXXX is implemented in both /Users/user/Downloads/deCONZ.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore (0x10631f9e8) and /usr/local/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore (0x11f4cd9e8). One of the two will be used. Which one is undefined." errors which leads to "QObject::moveToThread: Current thread (0x6000015dc050) is not the object's thread (0x6000015e8150).
Cannot move to target thread (0x6000015dc050)

You might be loading two sets of Qt binaries into the same process. Check that all plugins are compiled against the right Qt binaries. Export DYLD_PRINT_LIBRARIES=1 and check that only one set of binaries are being loaded.
QObject::startTimer: Timers can only be used with threads started with QThread" errors and exception of the app.

It is working properly either with the original complied binary (which comes with the app bundle) or if I remove the QTCore.framework from the system.
It seems that it links to the one on the system in addition to the one in the app bundle when building.

## Steps to reproduce the behavior
1. Build the library (tried to do it on release build, no difference) using the instructions on the BUILDING.md file including installing the dependencies.
2. Copy the complied file to the Plugins folder on the app bundle.
3. The app crashes right after launch (Launching it from shell/terminal will reveal the issue described above).

## Expected behavior
Just run without issues

## Screenshots
N/A

## Environment
macOS Ventura 13.4.1 (c) on Intel i7 Mac mini with 32GB Ram - Headless

 - Host system: Mac mini
 - Running method: Native run on macOS Ventura 
 - Firmware version: N/A
 - deCONZ version: (2.22.02)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
% /Users/levinshome/Downloads/deCONZ.app/Contents/MacOS/deCONZ                                 
libpng warning: iCCP: known incorrect sRGB profile
qt.qpa.fonts: Populating font family aliases took 89 ms. Replace uses of missing font family "Monospace" with one that exists to avoid this cost. 
objc[38979]: Class QMacAutoReleasePoolTracker is implemented in both /Users/levinshome/Downloads/deCONZ.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore (0x10631f948) and /usr/local/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore (0x11f4cd948). One of the two will be used. Which one is undefined.
objc[38979]: Class QT_ROOT_LEVEL_POOL__THESE_OBJECTS_WILL_BE_RELEASED_WHEN_QAPP_GOES_OUT_OF_SCOPE is implemented in both /Users/levinshome/Downloads/deCONZ.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore (0x10631f9c0) and /usr/local/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore (0x11f4cd9c0). One of the two will be used. Which one is undefined.
objc[38979]: Class KeyValueObserver is implemented in both /Users/levinshome/Downloads/deCONZ.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore (0x10631f9e8) and /usr/local/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore (0x11f4cd9e8). One of the two will be used. Which one is undefined.
objc[38979]: Class RunLoopModeTracker is implemented in both /Users/levinshome/Downloads/deCONZ.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore (0x10631fa38) and /usr/local/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore (0x11f4cda38). One of the two will be used. Which one is undefined.
QObject::moveToThread: Current thread (0x6000015dc050) is not the object's thread (0x6000015e8150).
Cannot move to target thread (0x6000015dc050)

You might be loading two sets of Qt binaries into the same process. Check that all plugins are compiled against the right Qt binaries. Export DYLD_PRINT_LIBRARIES=1 and check that only one set of binaries are being loaded.
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::connect: Cannot connect (nullptr)::aboutToQuit() to DeRestPlugin::appAboutToQuit()
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::connect: Cannot connect (nullptr)::nodeUpdated(quint64,QString,QString) to zmController::onRestNodeUpdated(quint64,QString,QString)
QWidget: Must construct a QApplication before a QWidget
zsh: abort      /Users/levinshome/Downloads/deCONZ.app/Contents/MacOS/deCONZ

## Additional context
N/A

manup, 2023-07-24:
Interesting, the current test bundle for macOS is build in one go. I have compiled the plugin also standalone but not yet tried to exchange it with the one in the bundle.

From the logs it looks like the library search path needs to be adjusted so the frameworks get resolved within the bundle. The `CMakeLists.txt` contains already everything needed to do this, likely only the install target needs to be called. I'll do some tests and adapt this in the [BUILDING.md#macos](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/BUILDING.md#macos) docs.

manup, 2023-07-24:
Build instructions have been updated. I've added a small script which builds the plugin and fixes the framework locations.

aryelevin, 2023-07-24:
@manup, Issue resolved with your build script.
Thank you very much

closed 2023-07-24 by aryelevin