dresden-elektronik/deconz-rest-plugin #7125: EasyAccess Locks

arevinje, 2023-07-23:
## Device

- Product name: EasyFingerTouch
- Manufacturer: EasyAccess AS
- Model identifier: easyCodeTouch_v1
- Device type : 
  - Lock

## Screenshots
[image]
[image]
[image]
[image]
[image]

Mimiix, 2023-07-24:
Hi

Please make sure the clusters are read and add a screenshot of the cluster list :)

Smanar, 2023-07-24:
Yep, because I m seing this device in the native code.
So need to check why your device haven't native support.

sliann, 2023-07-31:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2023-08-29 by github-actions[bot]