dresden-elektronik/deconz-rest-plugin #7134: IKEA Tradfri LED1924G9 wrong temperature limits

roupenig, 2023-07-30:
## Describe the bug

I have connected the IKEA Tradfri LED1924G9 to deCONZ, but I am getting wrong limits on the temperatures. According to IKEA documentation, I should be able to go from 1780 to 6000 Kelvin (167 to 561) but deCONZ is reporting 250 to 454.
I've checked that the bulb is using the latest firmware: 1.0.021

## Steps to reproduce the behavior

## Expected behavior

Expect to see 167 in Color Temperature Min Mireds
Expect to see 561 in Color Temperature Max Mireds

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Deconz Community Docker container
 - Firmware version: 26780700
 - deCONZ version: 2.22.02
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2023-07-30:
Hi

Afaik this is what the device reports. Not something deconz comes up with.

@ebaauw perhaps you can confirm?

ebaauw, 2023-07-30:
Indeed, from the screenshot, the bulb reports it supports 250 to 454 mireds (or ~2200K to 4000K).  I happen to have the same bulb, on the same firmware version, which afaik is the most recent one.  When issuing a _Move to Color Temperature_ from the GUI, the bulb reports back a minimum of 250 and a maximum of 454, so it behaves consistently with the reported min and max values.

> According to IKEA documentation

@roupenig could you please provide a link to this documentation?

roupenig, 2023-07-30:
Thanks for checking.

> could you please provide a link to this documentation?

[Here](https://www.ikea.com/nl/en/manuals/tradfri-led-bulb-e27-806-lumen-smart-wireless-dimmable-colour-and-white-spectrum-globe__AA-2301498-1-2.pdf) are the instructions taken from the bulb [product page](https://www.ikea.com/nl/en/p/tradfri-led-bulb-e27-806-lumen-smart-wireless-dimmable-colour-and-white-spectrum-globe-70439158/).

ebaauw, 2023-07-30:
The manual doesn’t specify how these whites are generated.  Need to pair the bulb to the Dirigera hub to check (or try if the left/right buttons on the remote still work with direct binding in deCONZ), but they probably use (or mix in?) the RGB channels for the extreme whites.  Almost tempted to cut open the bulb to see the individual LEDs.

If they do mix RGB and white channels, they would likely use the IKEA-specific _Scenes_ commands for this.  Using standard Zigbee commands, you either drive the RGB or the white channels, but not both.

ebaauw, 2023-07-31:
Paired the bulb and the STYRBAR remote to the DIRIGERA hub and took out the Zigbee sniffer.

The STYRBAR sends unicast IKEA-specific _Scenes_ commands to the hub on the left and right buttons.   I think these are the same as the old five-button remote sends.  The hub then sends a groupcast _Move to Color_ command to the light(s).  Even the "whites" are set using `xy` values.  The hub cycles through nine different values, in line with the instructons:

Color | X | Y
-- | -- | --
Cool white | 0.3804 | 0.3753
Warm white | 0.4598 | 0.4098
Warm glow | 0.5003 | 0.4162
Candle light | 0.5486 | 0.4026
Amber | 0.5730 | 0.3832
Dark peach | 0.6210 | 0.3405
Dark pink | 0.5001 | 0.2529
Light purple | 0.3200 | 0.1743
Light blue | 0.2062 | 0.1189

From what I can tell, the IKEA Home Smart app does the same: it sends _Move to Color_ on each of the 40 preset colours (it doesn't provide a colour wheel).  Again, the "whites" are set using `xy` values.  I cannot seem to find the 20 colours from the instructions, but presumably IKEA doubled that in a later firmware version of the DIRIGERA.  Or the 20 colours are in combination with the TRÅDFRI hub.  I think the light and STYRBAR came out well before the DIRIGERA.

Seeing how the light doesn't seem to support a vibrant red nor a vibrant green, I would theorise that it's actually a three- channel light, without separate warm white / cool white channels.  Like the old gamut-B Hue lights, it would have RGB LEDs optimised to display whites, at the expense of vibrant colours.  You'd have to cut open the bulb and look at the actual LEDs to confirm.  Tempted, but won't be doing that anytime soon.

closed 2023-07-31 by ebaauw

roupenig, 2023-07-31:
Thanks for the investigation, very helpful!

Mimiix, 2023-07-31:
Thanks @ebaauw  :)