dresden-elektronik/deconz-rest-plugin #7135: Moes Tuya Thermostat BTH-002 - TZE204_aoclfnxz

scrap84, 2023-07-31:
## Device

- Product name: Moes Tuya Thermostat BTH-002
- Manufacturer: TZE204_aoclfnxz
- Model identifier: TS0601
- Device type : 
  - Thermostat

## Screenshots

### Basic
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### Other clusters that are not mentioned above
[image]

Smanar, 2023-08-01:
Hello, can try this prototype

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_ztvwu4nk", "_TZE204_5toc8efa", "_TZE200_5toc8efa", "_TZE200_ye5jkfsb", "_TZE204_aoclfnxz", "_TZE200_u9bfwha0"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Moes TRVs",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"off": 0, "heat": 1}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["off", 0], ["heat", 1]
          ],
          "parse": {"fn": "tuya", "dpid": 1, "eval": "if (Attr.val == 0) { Item.val = 'off' } else { Item.val = 'heat' }"},
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "if (Item.val == 'off') { false } else { true }"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/preset",
          "parse": {"fn": "tuya", "dpid": 2, "script": "tuya_trv_preset.js"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "script": "tuya_trv_preset_set.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/valve",
          "parse": {"fn": "tuya", "dpid": 36, "eval": "Item.val = Attr.val > 5;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

Value can need to be converted (for exemple * 10 or /10)
"mode" is to enable / disable the device
"preset" is for .... hu .... will be a surprise, not sure it will work, try to chane something on the device to see reaction on the API.

scrap84, 2023-08-05:
i loaded the ddf, but the device is not recognized or displayed as thermostat within deconz ui; preset does not change anything on the device side, so the temp is not going up or down!?

Is there anything that must be considered by loading the ddf via deconz? I even tried to remove and add the devices again, but the type did not change and no additional value/fields have been added within deconz;

Smanar, 2023-08-06:
On deconz, the type will not change, the device firmware say it's a "Smart plug".
Difference will be only visible on the API.

If the node name change for "thermostat X" it mean the DDF is used, to be sure it's the good one, can select the node and make a "edit DDF"

To see api value can use Phoscon / help / API Information / Sensors

scrap84, 2023-08-06:
The thermostates are shown up as you mentioned and the temperatur can be read out via HA and within Phoscon API; I see events now like that:

`
"23:01:04:593": {
        "e": "changed",
        "id": "106",
        "r": "sensors",
        "state": {
            "lastupdated": "2023-08-06T21:01:00.506",
            "temperature": 2550,
            "valve": 0
        },
        "t": "event",
        "uniqueid": "a4:c1:38:48:95:d8:ec:e3-01-ef00"
    }
`
    
What is currently  not possible to set the temperature or get the event about the temperature change, like with the other theromstate; with the other thermostate i get events like these:

`
    "22:47:55:815": {
        "config": {
            "heatsetpoint": 1700,
            "locked": false,
            "offset": 0,
            "on": true,
            "preset": null,
            "reachable": true,
            "schedule": {},
            "schedule_on": false
        },
        "e": "changed",
        "id": "101",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:3c:84:ff:fe:e2:62:5b-01-0201"
    }
`

I read out the DDF from the correct working device as well and tried to set it for the other devices, but they don't seem to match; Because the events did also not shown up in the event list of the device;

` 
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_aoclfnxz",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/tuya_thd_bth-002_thermostat.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0xEF00",
          "0x0000",
          "0xEF00"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/productid",
          "public": false,
          "description": "Product identifier of the device.",
          "refresh.interval": 86400
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/locked",
          "description": "Locks or unlocks the device. This item has to be understood as child-lock.",
          "default": false
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/preset",
          "description": "Sets the operating mode for Tuya thermostats."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/schedule",
          "description": "On-device heating schedules."
        },
        {
          "name": "config/schedule_on",
          "description": "Determines if on-device schedules for setting the heatsetpoint are currently used or if the thermostat is operated manually.",
          "default": false
        },
        {
          "name": "config/setvalve",
          "public": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/temperature",
          "description": "The current temperature in °C &times; 100.",
          "default": 0
        }
      ]
    }
  ]
}
`

scrap84, 2023-08-06:
After adding the devices once more the devices are sending the updates with the DDF posted from me in the previous post; The only thing, which is currently not working is setting the value for the heatsetpoint. If i try to set the heatsetpoint via HA i does not reach the thermostat.

Smanar, 2023-08-07:
If you have the GUI, just go on log in deconz / help / debug view with flag "DDF" and "INFO" and change the heatpoint manualy.
You will see on log the return from the device (can check the dp) the value and the convertion done by deconz.

scrap84, 2023-08-08:
Did that now, but it looks like from the other working device:

Npot working device ... ECE3
[image]

Working device ... 625B
[image]

Both use the same description within the DDF as posted before, here a screenshot from the config/heatsetpoint
[image]

If I send something via HA to deconz over the UI the log contains for the device the following information:
[image]

For the working device it only prints:
[image]

closed 2023-09-08 by github-actions[bot]