dresden-elektronik/deconz-rest-plugin #7136: Tuya TS0203 support in deconz & phoscon

sharkwsk, 2023-08-01:
## Device

- Product name: Tuya TS0203 Zigbee door sensor
- Manufacturer: Tuya
- Model identifier: TS0203
- Device type : Sensor

### Cluster Info

[image]

### Node Info

[image]

### Background Info:

Using Conbee II stick and integrated to HA, I am using the Phoscon GUI to pair the sensor and unable to do so, Apologies if I have missed anything, I am pretty new in this space and if you need any specific details, let me know

Thank you

Mimiix, 2023-08-01:
Hi,

Can you please add all screenshots as requested in the template?

Please add them to your opening post.

Kind regards,

sharkwsk, 2023-08-01:
Updated original post, could not find some clusters, Let me know if I am doing something wrong

Smanar, 2023-08-01:
Hello, can try with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_rgchmad8", "_TZ3000_bzxlofth","_TZ3000_au1rjicn","_TZ3000_4ugnzsli","_TZ3000_decxrtwa", "XXXXXXX"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203","TS0203","TS0203","TS0203", "TS0203"],
  "vendor": "Tuya",
  "product": "Smart Zigbee Door Window Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

With replacing XXXXXXX by your manufacture name, there is 3 captures with it, and the complete name is not visible on all of them.

sharkwsk, 2023-08-02:
Hi @Smanar 

Thank you, I modified the DDF for Manufacture Name, loaded it and did a "Hot Reload" via VNC. 

Does this mean now the sensor should be able to be paired via Phoscon ?
Sorry If I am asking some basic questions, need some guidance on my original problem statement is: Unable to pair it via Phoscon GUI in HA

https://forum.phoscon.de/t/phoscon-support-for-tuya-ts0203-zigbee-door-sensor/3831

Smanar, 2023-08-02:
I have no impact on phoscon, it can choose to display a device or not, but yes I think it need to be visible in phoscon.

The node name need to change from 0x796f to something readable, else it mean the DDF installation have failed.

If you have HA will not be simple, here you have the way to install DDF https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

It's perhaps possible if you try to load the native DDF `devices\tuya\_TZ3000_TS0203_door_sensor.json` and try to edit it usng the editor, If i m right HA will make deconz save it in `/data/.local/share/dresden-elektronik/deCONZ/devices/`
(Select the device, right clic, edit DDF, load DDF, then add the model id and the manufacture name, then save it)

sharkwsk, 2023-08-03:
Hi @Smanar, Thank you for your feedback.
I was trying to pair it via Phoscon GUI but it did not connect. I left it overnight and somehow now I see the sensor showing up in the Phoscon GUI. Maybe the way I loaded DDF required sometime :)

Below is the snapshot I see. The sensor name "OpenClose 8", I assume this is something Phoscon generates and nothing to do with deconz/DDF file ?

[image]

I also restarted the deconz plugin and the DDF/changes are preserved and sensor is paired/connected

Thank you for your help

Smanar, 2023-08-03:
>Below is the snapshot I see. The sensor name "OpenClose 8", I assume this is something Phoscon generates and nothing to do with deconz/DDF file ?

Phoscon generate nothing, I think now on deconz the node name was "openclose 8" instead of 0x796f 
But seriously IDK how you have succesfully done that on the HA docker ... 

>I also restarted the deconz plugin and the DDF/changes are preserved and sensor is paired/connected

A mystery for me ...

But as all is working, do you want to make a PR to add the device officialy or want I make it ?

traschke, 2023-08-03:
I happened to have a very similar device in the mail today, the `manufacturername` is slightly different: `_TZ3000_6zvw8ham`. Model is `TS0203` as well. Used the ddf @Smanar posted earlier and added the new `manufacturername` and a new `modelid` entry as well. Works with my device as well partly. On/Close detection works, but battery shows always 0 %.
Maybe I have time to look into this further in the next days.

Modified DDF:
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_6zvw8ham", "_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_rgchmad8", "_TZ3000_bzxlofth","_TZ3000_au1rjicn","_TZ3000_4ugnzsli","_TZ3000_decxrtwa", "XXXXXXX"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203","TS0203","TS0203","TS0203", "TS0203"],
  "vendor": "Tuya",
  "product": "Smart Zigbee Door Window Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

sharkwsk, 2023-08-04:
> Phoscon generate nothing, I think now on deconz the node name was "openclose 8" instead of 0x796f But seriously IDK how you have succesfully done that on the HA docker ...

Is this something I can change in Deconz, the node name ?

> > I also restarted the deconz plugin and the DDF/changes are preserved and sensor is paired/connected
> 
> A mystery for me ...

I copied the DDF file into the docker directory, Open/Hot Reload in deconz VNF, nothing else I did

> But as all is working, do you want to make a PR to add the device officialy or want I make it ?

I am not sure how to do that, I will let you do that

Smanar, 2023-08-04:
>Is this something I can change in Deconz, the node name ?

Yes but deconz will use the name used in the API, so if you realy want to change it, you can just use phoscon.

>I am not sure how to do that, I will let you do that

K I wil add the @traschke device in same time.
The battery can need 24h to be displayed, else can check if values exist using the GUI.

PR added https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7145

traschke, 2023-08-04:
> K I wil add the @traschke device in same time. The battery can need 24h to be displayed, else can check if values exist using the GUI.

You are right, the battery value is now read correctly.

closed 2023-08-05 by manup