dresden-elektronik/deconz-rest-plugin #7139: Device name

llugo, 2023-08-02:
## Device

- Product name: Zigbee Radiator Actuator
- Manufacturer: _TZE200_cpmgn2cf
- Model identifier: TS0601
- Device type : 
  - Thermostat

## Screenshots
  - Endpoints and clusters of the node
[image]

  - Node Info panel
[image]
[image]

- Cluser info panel

[image]

## Additional Notes

Since there is a "Generic Tuya TRV  DDF" (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6829) which I unsuccessfully tried to load on the device, I thought I'd give it another chance here. 

It was bought here: https://www.aliexpress.com/item/1005002212605653.html?spm=a2g0o.order_list.order_list_main.5.54a85c5fRUkOms&gatewayAdapt=glo2deu

llugo, 2023-08-02:
Apparently this is a dupe of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6381#issue-1393857813

Mimiix, 2023-08-02:
@Smanar still not possible

Smanar, 2023-08-03:
Can try this DDF 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_7fqkphoq", "_TZE200_xby0s3ta", "_TZE200_cpmgn2cf"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Tuya TRV (MOES)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 2}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 110, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 21, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 7, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 7, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 18, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 18, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 2]
          ],
          "parse": {"fn": "tuya", "dpid": 4, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 4, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 2 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

The dpid for mode can be 106 instead of 4.

llugo, 2023-08-03:
This is what i get from the API after loading the DDF, tried both dpid values (106 and 4) in "parse" and "write" lines in config/mode section...
It somehow got the name somewhat right...

 "19": {
        "config": {
            "heatsetpoint": null,
            "locked": true,
            "mode": "off",
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "b58befe403f8cda28fe96dc5b8a54867",
        "lastannounced": null,
        "lastseen": "2023-08-03T18:37Z",
        "manufacturername": "_TZE200_cpmgn2cf",
        "modelid": "TS0601",
        "name": "Thermostat 19",
        "state": {
            "lastupdated": "2023-08-03T16:23:11.232",
            "lowbattery": null,
            "on": false,
            "reachable": null,
            "temperature": 0
        },
        "type": "ZHAThermostat",
        "uniqueid": "8c:f6:81:ff:fe:a8:86:34-01-ef00"
    },

llugo, 2023-08-05:
> Can you enable log in deconz / help / debug view with flag "info"

`13:42:24:941 TY_DATA_RESPONSE: seq 269, dpid: 0x02, type: 0x02, length: 4, val: 240
13:42:24:942 TY_DATA_RESPONSE: seq 269, dpid: 0x02, type: 0x02, length: 4, val: 240
13:42:24:943 TY_DATA_RESPONSE: seq 269, dpid: 0x02, type: 0x02, length: 4, val: 240
13:42:24:944 TY_DATA_RESPONSE: seq 269, dpid: 0x02, type: 0x02, length: 4, val: 240
13:42:24:945 TY_DATA_RESPONSE: seq 269, dpid: 0x02, type: 0x02, length: 4, val: 240
13:42:24:946 TY_DATA_RESPONSE: seq 269, dpid: 0x02, type: 0x02, length: 4, val: 240
13:42:24:947 TY_DATA_RESPONSE: seq 269, dpid: 0x02, type: 0x02, length: 4, val: 240
13:42:25:159 TY_DATA_RESPONSE: seq 270, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:161 TY_DATA_RESPONSE: seq 270, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:161 TY_DATA_RESPONSE: seq 270, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:162 TY_DATA_RESPONSE: seq 270, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:162 TY_DATA_RESPONSE: seq 270, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:164 TY_DATA_RESPONSE: seq 270, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:165 TY_DATA_RESPONSE: seq 270, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:368 TY_DATA_RESPONSE: seq 271, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:369 TY_DATA_RESPONSE: seq 271, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:370 TY_DATA_RESPONSE: seq 271, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:371 TY_DATA_RESPONSE: seq 271, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:371 TY_DATA_RESPONSE: seq 271, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:372 TY_DATA_RESPONSE: seq 271, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:373 TY_DATA_RESPONSE: seq 271, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:25:578 TY_DATA_RESPONSE: seq 272, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:25:579 TY_DATA_RESPONSE: seq 272, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:25:580 TY_DATA_RESPONSE: seq 272, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:25:581 TY_DATA_RESPONSE: seq 272, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:25:581 TY_DATA_RESPONSE: seq 272, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:25:582 TY_DATA_RESPONSE: seq 272, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:25:582 TY_DATA_RESPONSE: seq 272, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:29:306 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205
13:42:29:307 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205
13:42:29:308 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205
13:42:29:310 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205
13:42:29:311 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205
13:42:29:312 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205
13:42:29:313 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205
13:42:29:568 TY_DATA_RESPONSE: seq 274, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:29:569 TY_DATA_RESPONSE: seq 274, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:29:570 TY_DATA_RESPONSE: seq 274, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:29:571 TY_DATA_RESPONSE: seq 274, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:29:571 TY_DATA_RESPONSE: seq 274, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:29:572 TY_DATA_RESPONSE: seq 274, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:29:573 TY_DATA_RESPONSE: seq 274, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:34:701 TY_DATA_RESPONSE: seq 275, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:34:703 TY_DATA_RESPONSE: seq 275, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:34:704 TY_DATA_RESPONSE: seq 275, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:34:705 TY_DATA_RESPONSE: seq 275, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:34:706 TY_DATA_RESPONSE: seq 275, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:34:708 TY_DATA_RESPONSE: seq 275, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:34:709 TY_DATA_RESPONSE: seq 275, dpid: 0x04, type: 0x04, length: 1, val: 2
13:42:45:616 ZCL attribute report 0xF0D1B80000194F9E for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:42:46:575 ZCL attribute report 0xF0D1B80000194F9E for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:42:48:718 TY_DATA_RESPONSE: seq 276, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:48:719 TY_DATA_RESPONSE: seq 276, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:48:720 TY_DATA_RESPONSE: seq 276, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:48:720 TY_DATA_RESPONSE: seq 276, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:48:721 TY_DATA_RESPONSE: seq 276, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:48:722 TY_DATA_RESPONSE: seq 276, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:48:723 TY_DATA_RESPONSE: seq 276, dpid: 0x03, type: 0x02, length: 4, val: 230
13:42:48:946 TY_DATA_RESPONSE: seq 277, dpid: 0x6D, type: 0x02, length: 4, val: 75
13:42:48:947 TY_DATA_RESPONSE: seq 277, dpid: 0x6D, type: 0x02, length: 4, val: 75
13:42:48:948 TY_DATA_RESPONSE: seq 277, dpid: 0x6D, type: 0x02, length: 4, val: 75
13:42:48:949 TY_DATA_RESPONSE: seq 277, dpid: 0x6D, type: 0x02, length: 4, val: 75
13:42:48:950 TY_DATA_RESPONSE: seq 277, dpid: 0x6D, type: 0x02, length: 4, val: 75
13:42:48:951 TY_DATA_RESPONSE: seq 277, dpid: 0x6D, type: 0x02, length: 4, val: 75
[... cut ...]
13:43:21:485 ZCL attribute report 0xF0D1B800001953A4 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:43:22:405 ZCL attribute report 0xF0D1B800001953A4 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:43:26:306 trigger rule 40 - pir-fsm-reset-after-off
13:43:26:309 rule event /sensors/13/state/lastupdated: 0 -> 0
13:43:26:311 rule event /sensors/14/state/status: 3 -> -1
13:43:27:805 Set sensor check interval to 100 milliseconds
13:43:36:306 trigger rule 43 - pir-fsm-state-wait
13:43:36:309 rule event /sensors/14/state/status: -1 -> 0`

Smanar, 2023-08-05:
`13:42:29:306 TY_DATA_RESPONSE: seq 273, dpid: 0x02, type: 0x02, length: 4, val: 205`

This one need to be setheatpoint, and I m using same

`"parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},`

It need to work ....

Nothing special in logs using "DDF" and "info" ?

llugo, 2023-08-05:
>You are not using HA or docker ?

no just deconz v2.22.2.

Somehow the device seems to pick up a wrong ddf when connecting...
Loading and hot reloading the DDF posted above doesn't seem to work. Whenever i close the editor and open it again there are wrong DPID entries.

Screenshot after connecting the node:
[image]

Screenshot after hot reloading
[image]

Sreenshot after closing the DDF editor an opening it again
[image]

DDF location ist /home/[USER]/.local/share/dresden-elektronik/deCONZ/devices/tuya/

Smanar, 2023-08-05:
Ok, I think you have more than 1 DDF for the same device.

You have one called tuya_ts601.json and one called _TZE200_cpmgn2cf.json

llugo, 2023-08-05:
Oh good point. 
After restarting deconz it's picking the right DDF!
I thought just deleting the 'wrong' file was sufficient.

llugo, 2023-08-05:
This is what the API response now.
Heatsetpoint looks good. But temperature should be nonzero.

```
"20": {
        "config": {
            "battery": 0,
            "heatsetpoint": 2000,
            "locked": true,
            "mode": "off",
            "on": true,
            "reachable": true,
            "windowopen_set": false
        },
        "etag": "f3c5ce323d00fb21d47b5212f5f86085",
        "lastannounced": null,
        "lastseen": "2023-08-05T16:48Z",
        "manufacturername": "_TZE200_cpmgn2cf",
        "modelid": "TS0601",
        "name": "Thermostat 20",
        "state": {
            "lastupdated": "2023-08-05T16:30:44.783",
            "lowbattery": null,
            "temperature": 0
        },
        "type": "ZHAThermostat",
        "uniqueid": "8c:f6:81:ff:fe:a8:86:34-01-ef00"
    },
```

Smanar, 2023-08-05:
`18:54:07:820 TY_DATA_RESPONSE: seq 357, dpid: 0x03, type: 0x02, length: 4, val: 260`

```
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
```

So 260 * 10 give 2600, nothing strange ? Still no value for temperature ?

llugo, 2023-08-05:
>Still no value for temperature ?

After doing some more API requests, it's reporting temperature correctly now!

```
    "20": {
        "config": {
            "battery": 0,
            "heatsetpoint": 2350,
            "locked": true,
            "mode": "off",
            "on": true,
            "reachable": true,
            "windowopen_set": false
        },
        "etag": "ec61d6dfe51d6299b394b7e477f01928",
        "lastannounced": null,
        "lastseen": "2023-08-05T17:13Z",
        "manufacturername": "_TZE200_cpmgn2cf",
        "modelid": "TS0601",
        "name": "Thermostat 20",
        "state": {
            "lastupdated": "2023-08-05T17:12:08.825",
            "lowbattery": null,
            "temperature": 2650
        },
        "type": "ZHAThermostat",
        "uniqueid": "8c:f6:81:ff:fe:a8:86:34-01-ef00"
    },
```

Great!
Finally this device is going to be used as intented!

@Smanar Thanks a thousand once again!

Smanar, 2023-08-06:
I m just seing this device have native support in c++ code

`    {"_TZE200_cpmgn2cf", "TS0601", "MOES", "Tuya_THD HY368 TRV"},`

IDK why it haven't worked ... Perhaps because device detection is harder without DDF.

And this device haven't battery support, so bad you have `"lowbattery": null,` you haven't used battery to test this field ?

llugo, 2023-08-07:
1)
 I changed batteries, API response now looks like that:

```
    "19": {
        "config": {
            "battery": 0,
            "heatsetpoint": 1500,
            "locked": false,
            "mode": "off",
            "on": true,
            "reachable": true,
            "windowopen_set": false
        },
        "etag": "d806a25c45a93b11cc05562c9b2b9fdc",
        "lastannounced": null,
        "lastseen": "2023-08-07T18:41Z",
        "manufacturername": "_TZE200_cpmgn2cf",
        "modelid": "TS0601",
        "name": "Thermostat 19",
        "state": {
            "lastupdated": "2023-08-07T18:41:30.298",
            "lowbattery": false,
            "temperature": 2550
        },
        "type": "ZHAThermostat",
        "uniqueid": "8c:f6:81:ff:fe:a8:86:34-01-ef00"
    },
```

2) 
Just to double check for "integral" deCONZ support I removed the DDF file then deleted the node then restarted deCONZ and reconnected the node.
It wasn't detected as a TRV.
So i deleted the node.
Then i moved the DDF back, restarted deconz and reconnected the node again.
It was then recognized as a TRV like before.

llugo, 2023-08-07:
>I changed batteries, API response now looks like that [...]

Interestingly, altough "lowbattery" is reported as false deCONZ shows a low battery symbol:
[image]

Smanar, 2023-08-07:
deconz can't get battery level, this device is a tuya one, nothing standard.

`"lowbattery": false,`

Ha nice the value was updated (it's no more "null'), it was good or bad battery ?

llugo, 2023-08-08:
>deconz can't get battery level, this device is a tuya one, nothing standard.

Oh, I see.

>Ha nice the value was updated (it's no more "null'), it was good or bad battery ?

I'm not sure what you mean with good or bad battery?
The device is using 2 AA alkaline batteries.
When it was showing `"lowbattery": null,` there was also a blinking battery symbol on the devices display so the battery was indeed drained.
After changing batteries with a new pair, blinking stopped and now API shows `"lowbattery": false,`.

Smanar, 2023-08-08:
>I'm not sure what you mean with good or bad battery?

If you have used full battery it's ok (because we had `"lowbattery": false,) but if you have used empty battery (to simulate a low battery), the value is reversed ^^.

llugo, 2023-08-16:
Sorry but apparantly the case isn't closed i'm afraid.
When setting the heatsetpoint through API the device for some reason returns the value to the factory default after about 5 seconds. Also the device isn't doing anything anymore when i change the heatsetpoint.

Besides connecting it, and changing battereies i never changed anything on the device itself.
Any idea what could cause that behaviour?

```
command:
curl -X PUT -H "Content-Type: application/json" -d '{"heatsetpoint":'3100'}' http://localhost:8080/APIKEY/sensors/19/config

API response after 1s:
{
    "config": {
        "battery": 0,
        "heatsetpoint": 3100,
        "locked": false,
        "mode": "off",
        "on": true,
        "reachable": true,
        "windowopen_set": false
    },
    "etag": "35718a8ed5289073f25bf58c3e7d8e69",
    "lastannounced": null,
    "lastseen": "2023-08-16T18:56Z",
    "manufacturername": "_TZE200_cpmgn2cf",
    "modelid": "TS0601",
    "name": "Thermostat_Bad",
    "state": {
        "lastupdated": "2023-08-16T18:27:47.669",
        "lowbattery": false,
        "temperature": 2850
    },
    "type": "ZHAThermostat",
    "uniqueid": "8c:f6:81:ff:fe:a8:86:34-01-ef00"
}

API respone after 5s:

{
    "config": {
        "battery": 0,
        "heatsetpoint": 1500,
        "locked": false,
        "mode": "off",
        "on": true,
        "reachable": true,
        "windowopen_set": false
    },
    "etag": "34afd4accc31ec731f9b1421980aa6ee",
    "lastannounced": null,
    "lastseen": "2023-08-16T18:56Z",
    "manufacturername": "_TZE200_cpmgn2cf",
    "modelid": "TS0601",
    "name": "Thermostat_Bad",
    "state": {
        "lastupdated": "2023-08-16T18:57:11.039",
        "lowbattery": false,
        "temperature": 2800
    },
    "type": "ZHAThermostat",
    "uniqueid": "8c:f6:81:ff:fe:a8:86:34-01-ef00"
}
```

Mimiix, 2023-08-16:
Wasn't the max 2700?

llugo, 2023-08-16:
> Wasn't the max 2700?
Yeah i can only say that when I initially tested it temps where about 2400 or so, so that would explain it.
From the docs delivered with the device it says factory default max temp would be 35°C.

Smanar, 2023-08-17:
`"mode": "off",`

You haven"t other mode to try ?

llugo, 2023-08-17:
> `"mode": "off",`
> 
> You haven"t other mode to try ?
> Something you can try is set it manually on the device, you will see if it work or if the device don't change the mode in same time.

Ok, i tried to change things on the rather complicated UX of the device itself.
After doing that i was able to set heatsetpoint without it jumping back,  but no changes in 'mode' where observed.

> And this device probably support "preset"
> 
>         {
>           "name": "config/preset",
>           "parse": {"fn": "tuya", "dpid": 4, "script": "tuya_trv_preset.js"},
>           "write": {"fn": "tuya", "dpid": 4, "dt": "0x30", "script": "tuya_trv_preset_set.js"},
>           "read": {"fn": "none"}
>         },
> The DDF need to be in the same folder than "tuya_trv_preset_set.js"

After having a look at `tuya_trv_preset_set.js` I think maybe it sets itself from manual to auto (with 'auto' maybe not accepting heatsetpoint changes) after a hardcoded time?

I'll try adding the preset lines to the DDF.

llugo, 2023-08-17:
> I'll try adding the preset lines to the DDF.

Looks like this worked!
I get the following API response now:

```
{
    "config": {
        "heatsetpoint": 1500,
        "locked": false,
        "mode": "off",
        "on": true,
        "preset": "eco",
        "reachable": true,
        "windowopen_set": false
    },
    "etag": "3b4532bbd46d39a3e6ce8182e909c1ff",
    "lastannounced": null,
    "lastseen": "2023-08-17T19:06Z",
    "manufacturername": "_TZE200_cpmgn2cf",
    "modelid": "TS0601",
    "name": "Thermostat 19",
    "state": {
        "lastupdated": "2023-08-17T19:06:25.881",
        "lowbattery": false,
        "temperature": 2700
    },
    "type": "ZHAThermostat",
    "uniqueid": "8c:f6:81:ff:fe:a8:86:34-01-ef00"
}
```
It is possible to change the preset to 'manual' via an API request. So if my hypothesis about 'auto' mode, which i hereby expand to 'eco' mode, is correct it should work now.  We'll see!

Smanar, 2023-08-18:
I will be happy if you can describe the working mode ^^, not something evident for tuya.

In my mind you need to have mode=auto (for the device auto regulate the temperature) and preset=auto too ?
But why this device have mode and preset working ? There is not one of them useless ?

llugo, 2023-08-18:
So, after waiting for a day just to see if anything changed, the preset 'manual' is still set and i can set the desired heatsetpoint.
I'll now use a basic bash script with api requests and some cron entries to do weekday/weekend/day/night 'presets'.
>But why this device have mode and preset working ? There is not one of them useless ?

'Mode' seems useless yes. While trying different things on the device again I still didn't get any other value for 'mode' than 'off'.
So config/mode can be pruned from the DDF just like config/battery?

Smanar, 2023-08-19:
>So config/mode can be pruned from the DDF just like config/battery?

Yes, surely, so the actual DDF need to be

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_7fqkphoq", "_TZE200_xby0s3ta", "_TZE200_cpmgn2cf"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Tuya TRV (MOES)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 110, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 7, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 7, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 18, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 18, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/preset",
          "parse": {"fn": "tuya", "dpid": 4, "script": "tuya_trv_preset.js"},
          "write": {"fn": "tuya", "dpid": 4, "dt": "0x30", "script": "tuya_trv_preset_set.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

llugo, 2023-09-01:
It's probably bad since it's my first attemp at writing a bash script with more than 5 lines, but anyways:
I published a script you can use to control this devices (and probably other TRVs) settings.
Currently there is support for 
- listing TRV devices (assigned as ZHAThermostat in deCONZ)
- (re)naming TRV devices
- setting the desired temperature indefinitely or for a specified duration

It was only tested with OPs device on Raspian buster. Needs curl and jq.
You can find the script here:
[https://gist.github.com/llugo/ca4a64c2e29e3278e0a7a696faade832](https://gist.github.com/llugo/ca4a64c2e29e3278e0a7a696faade832)

closed 2023-09-30 by github-actions[bot]

bchhabra, 2024-02-04:
@Smanar my bad all the time i was using above json only, sorry for the confusion.
I modified the above json to use the DPID 109 for valve
now I see the value in the valve as 1. 
However on the trv itself I check the valve and it says 50 which is 50% open. According to the manual value of valve could go from 0 to 99 (0 = fully closed and 99 0= fully open)

> {
          "name": "state/valve",
          "parse": {"fn": "tuya", "dpid": 109, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },

Smanar, 2024-02-04:
State/valve is a bool, so it can be only true or false, the DPID 109 is a bool too.
From z2m for the _TZE200_cpmgn2cf

`e.binary('valve', ea.STATE, 'CLOSED', 'OPEN'),`

bchhabra, 2024-02-04:
@Smanar is there any way to get this Information? I am happy to do some tests if you want to try out something

Smanar, 2024-02-04:
But you need to have it using

```
{
"name": "state/valve",
"parse": {"fn": "tuya", "dpid": 109, "eval": "Item.val = Attr.val != 0"},
"read": {"fn": "none"}
},
```

This value don't move from true to false ?

bchhabra, 2024-02-04:
This works fine as you described, it is 0 when closed and 1 when open.

I just checked the zigee2mqtt docs and found that valve open percentage is not readable. They have 3 values for this as described below 

If all you need is to control on and off, you can set "force" with topic zigbee2mqtt/FRIENDLY_NAME/set.

The payload values are:
open -> fully opens valve and stays there
close -> fully closes valve and stays there
normal -> normal valve operation

Can we achieve something similar?

Smanar, 2024-02-05:
You have the z2m doc page ? (so much clone with tuya)
I m checking again, and I m seing only 2 thing abut valve, the valve return, and a "valve detection" request, but this one is a bool too.

Smanar, 2024-02-05:
So on your page I can read there is a valve detection but it don't mean "valve position" ?
The valve position is the dpid 109 too, it seem we are just checking if this value is 0 or other value. And good new I was wrong again, this field is not a bool but an int so can try

```
{
"name": "state/valve",
"parse": {"fn": "tuya", "dpid": 109, "eval": "Item.val = Attr.val"},
"read": {"fn": "none"}
},
```

bchhabra, 2024-02-06:
@Smanar 
I tried a few more tests.

**for both config/locked and config/windowopen_set**
- set on trv and try to access via API (not able to see the changed values in API response)
- set via API reflects in get on API but does not reflect on trv.

Smanar, 2024-02-06:
Do you have the GUI ?
If yes, just enable log in deconz (help/debug view with flag info  and info_l2 )set the feature manualy on the device, and look in log the dpid used by the device, because I m cheking again and dpid are good 7 and 18
.

bchhabra, 2024-02-07:
for windowopen_set

**via api**
11:53:06:923 writeTuyaData, dpid: 0x12, type: 0x10, expr: Item.val;
11:53:06:931 Tuya write expression: Item.val; --> true
11:53:17:249 SC state change failed: a4:6d:d4:ff:fe:1f:99:18-01-ef00
11:53:35:537 0x0017880106278811 found group 0x0004
11:53:35:540 0x0017880106278811 found group 0xFFF0
11:53:43:046 sensor 8 (TRADFRI motion sensor): disable presence
11:53:43:446 Set sensor check interval to 1000 milliseconds

**change on trv**
11:54:05:376 TY_DATA_RESPONSE: seq 691, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:05:379 TY_DATA_RESPONSE: seq 691, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:05:381 TY_DATA_RESPONSE: seq 691, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:05:385 TY_DATA_RESPONSE: seq 691, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:05:388 TY_DATA_RESPONSE: seq 691, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:05:392 TY_DATA_RESPONSE: seq 691, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:17:798 TY_DATA_RESPONSE: seq 692, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:17:800 TY_DATA_RESPONSE: seq 692, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:17:803 TY_DATA_RESPONSE: seq 692, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:17:805 TY_DATA_RESPONSE: seq 692, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:17:808 TY_DATA_RESPONSE: seq 692, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:17:811 TY_DATA_RESPONSE: seq 692, dpid: 0x03, type: 0x02, length: 4, val: 190
11:54:18:817 0x0017880106278811 error APSDE-DATA.confirm: 0xE9 on task
11:54:50:199 0x0017880106278811 error APSDE-DATA.confirm: 0xE9 on task
11:54:56:175 ZCL attribute report 0xEC1BBDFFFEAB7C09 for cluster: 0x0006, ep: 0x0B, frame control: 0x18, mfcode: 0x0000

Smanar, 2024-02-07:
When you have changed it on the TRV, nothing was returned, dpid 0x03 is only temperature report.
And when you have set it using the API, all seem fine, but you have an error message

```
11:53:06:923 writeTuyaData, dpid: 0x12, type: 0x10, expr: Item.val;
11:53:06:931 Tuya write expression: Item.val; --> true
11:53:17:249 SC state change failed: a4:6d:d4:ff:fe:1f:99:18-01-ef00
```

You can enable APS log (need APS+APS_l2), you will see something 

`17:52:56:269 	asdu (length: 5): 10C8000000`

Will be the raw request, but I m almost sure it will finish by 00 01 12 01 00/01

Your device probably don't support it, or use a different dpid.

bchhabra, 2024-02-07:
I shared here the logs, not sure if this helps.
[logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/14197229/logs.txt)

Smanar, 2024-02-07:
Those TRV are not designed to work like that, to prevent battery drain, it's lazy device. You set a heatpoint and the device work alone.

To know if this device support for exemple the "lock" feature, You need to see it in logs when you use it manualy.

>11:54:05:376 TY_DATA_RESPONSE: seq 691, dpid: 0x03, type: 0x02, length: 4, val: 190

It's the dpid 0x12 for the "lock"

We can use too the 'config/unoccupiedheatsetpoint' to set a second heatpoint, but IDK how work this device. Do you want to use comfort or eco ? it work with preset ?

bchhabra, 2024-02-07:
In the post above for z2m it is the same trv being used and they have implemented the force opening/closing the valve. How do you conclude that trv is not designed to work like that?

I used preset (set via API )and they work as expected, no issues there.
Setting a schedule is something I was looking for, and setting a temperature for eco/comfort.

Anyways the main question here is to control the valve forcefully.

If all you need is to control on and off, you can set "force" with topic zigbee2mqtt/FRIENDLY_NAME/set.

The payload values are:
open -> fully opens valve and stays there
close -> fully closes valve and stays there
normal -> normal valve operation

{
    "force": "open"
}

Smanar, 2024-02-08:
>How do you conclude that trv is not designed to work like that?

Just make a try yourself.

The "force" mode is the dpid 106, type enum, value 0 - normal, 1 - open, 2 - close
So for test can try
```

        {
          "name": "config/interfacemode",
          "parse": {"fn": "tuya", "dpid": 106, "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 106, "dt": "0x30", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
```

I m agree "config/interfacemode" is ugly, but mode and preset are already used, and the one resting are not famous, make a test, create a new field will be easier after, on last DDF version you can create "custom" field.

craybill-greater, 2024-02-29:
I'm also interested to get the _TZE200_cpmgn2cf working with deconz. I also have the Tuya Zigbee hub running with the tuya adapter on iobroker. If relevant for support I can share data points generated by the tuya dapter in iobroker...

Smanar, 2024-02-29:
Hello can try this DDF
It''s an updated one, with the "mode" feature to enable auto/manu/off

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_7fqkphoq", "_TZE200_xby0s3ta", "_TZE200_cpmgn2cf"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Tuya TRV (MOES)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 2}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 110, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 7, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 7, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 18, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 18, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/preset",
          "parse": {"fn": "tuya", "dpid": 4, "script": "tuya_trv_preset.js"},
          "write": {"fn": "tuya", "dpid": 4, "dt": "0x30", "script": "tuya_trv_preset_set.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 2]
          ],
          "parse": {"fn": "tuya", "dpid": 106, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 106, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 2 }"},
          "read": {"fn": "none"}
        },
        {
        "name": "state/valve",
        "parse": {"fn": "tuya", "dpid": 109, "eval": "Item.val = Attr.val"},
        "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

craybill-greater, 2024-11-13:
Thanks this DDF Works quite well so far.

Would it be possible to add some more DPIDs?

Here is the list of DPIDs I gathered from TUYA hub for this TRV. Especially the  room_temp calibration 44 would be interesting, but the others as well.
[image]

Smanar, 2024-11-13:
Tell me wich one dpid you want, we can see what is free as command.
The calibration is useless, just replace

```
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
```

by
```
        {
          "name": "config/offset"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val;"},
          "read": {"fn": "none"}
        }
```

You can use "config/offset" after that

craybill-greater, 2024-11-13:
The dpids without lock symbol in the screenshot are r/w, locked ones are ro.Von: Smanar ***@***.***>Gesendet: Mittwoch, November 13, 2024 6:29:33 PMAn: dresden-elektronik/deconz-rest-plugin ***@***.***>Cc: craybill-greater ***@***.***>; Comment ***@***.***>Betreff: Re: [dresden-elektronik/deconz-rest-plugin] Zigbee Radiator Actuator / _TZE200_cpmgn2cf / TS0601 (Issue #7139)
Tell me wich one dpid you want, we can see what is free as command.
The calibration is useless, just replace
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }

by
        {
          "name": "config/offset"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val;"},
          "read": {"fn": "none"}
        }

You can use "config/offset" after that

Smanar, 2024-11-14:
You want the "wind" for exemple ? you know what is it ?

Checkdat, 2025-02-18:
Hi Smanar,

Is it possible to add Windowopen_set?
I tried to set zhe value (on/off) via FHEM 
And the response was: 
Invalid value ,on,off, for parameter windowopen_set.

Smanar, 2025-02-18:
Hello, try with true or false, but without quote, it's a bool value.

Checkdat, 2025-02-18:
So testet from fhem: set DVICENAME windowopen_set false
set DVICENAME windowopen_set true
set DVICENAME windowopen_set ON
set DVICENAME windowopen_set OFF

EVERYTIME invalid value 🙈 i have no clue

Checkdat, 2025-02-20:
Hi again,

In response of my post obove:

windowopen_set is not working (yet)

offset value is very strange 😅
Set in 000 format not in °C 
0-1°C in strange steps from 0-99 
100 will set an offset of 2°c 
200 is 4°C ….
Offset for 1.1-1.9 is not possible

My opinion was to „calibrate“ with external temp reading ond set the differenz as offset 
No clue if this will work …

Smanar, 2025-02-21:
I don't see the offset in the DDF ?
You have coded a special one ?