dresden-elektronik/deconz-rest-plugin #7147: IKEA RODRET Kabelloser Dimmer/Schalter

TheGreatNemo, 2023-08-04:
## Device
- Product name: RODRET Kabelloser Dimmer/Schalter - E2201
- Manufacturer: IKEA of SWEDEN
- Model identifier: RODRET Dimmer
- Device type :
  - Switch

## Screenshots
Cluster Info
[image]
[image]
[image]
[image]

Node Panel
[image]
[image]

DIYtechie, 2023-08-18:
I also have this request. Will not seem to pair.

Mimiix, 2023-08-18:
Guys, it's not supported yet. That's why it's not showing up in phoscon. As long as it has the hex names it wont work. If you can't get it in deconz, it's a different issue and perhaps you need to ask for help on the forums with that.

BabaIsYou, 2023-08-24:
As suggested by ebaauw in Discord, will begin by reusing DDF for Styrbar (adapted of course)

```
 {
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_IKEA",
  "modelid": "RODRET Dimmer",
  "product": "RODRET Dimmer - E2201",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x1000"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0820",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x1000"
        ],
        "out": [
          "0x0006",
          "0x0008",
          "0x0005"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0001",
            "at": "0x0021",
            "eval": "Item.val = Math.round(Attr.val / 2)"
          },
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0005"
    }
  ]
}
```

and adding these lines into `button_maps.json`
```
        "ikeaRodretMap": {
            "vendor": "IKEA",
            "doc": "Rodret dimmer remote",
            "modelids": ["RODRET Dimmer"],
            "buttons": [
                {"S_BUTTON_1": "Dim Up"},
                {"S_BUTTON_2": "Dim Down"},
            ],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Move down"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"]
            ]
        },
```

BabaIsYou, 2023-08-24:
> Looks good, except for the non-existent _Scenes_ client cluster (0x0005) in the fingerprint and in the bindings. On the STYRBAR, that cluster is used by the left/right buttons.

Thx ! Corrected in the above proposed DDF.

TheGreatNemo, 2023-08-25:
> As suggested by ebaauw in Discord, will begin by reusing DDF for Styrbar (adapted of course)
> [...]
> and adding these lines into `button_maps.json`
> 
> ```
>         "ikeaRodretMap": {
>             "vendor": "IKEA",
>             "doc": "Rodret dimmer remote",
>             "modelids": ["RODRET Dimmer"],
>             "buttons": [
>                 {"S_BUTTON_1": "Dim Up"},
>                 {"S_BUTTON_2": "Dim Down"},
>             ],
>             "map": [
>                 [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
>                 [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
>                 [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
>                 [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
>                 [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Move down"],
>                 [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
>                 [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"]
>             ]
>         },
> ```

Thanks a lot.
This seams to work 👍 I can add the switch, they show up in Phoscon App and look like they work pretty much as intendet 👍 

I only changed one small thing:
The line "{"S_BUTTON_2": "Dim Down"}," shoudn't have comma at the end.

BabaIsYou, 2023-08-25:
Sorry about that typo ;-/
Do you want to make the pull request to have this included in a next release ?

TheGreatNemo, 2023-08-25:
> Do you want to make the pull request to have this included in a next release ?

Done. 
Thanks again 👍

x2005254, 2023-08-28:
Dear All,

I include the DDF and the button Map config in my deconz conf, as other I'm able to discover it with Phoscon now, But I cannot use it as group Binding to affect to a light without phoscon. the inclusion didn't create the proper group.

Second issue, it shown as empty battery : 
[image]
[image]
by deconz.

[image]

ebaauw, 2023-09-09:
Needed to visit IKEA to get new filters for my STARKVIND, so picked up a RODRET along the way.

The device does send groupcasts by default.  Group ID is picked randomly (on device reset) and reported through _ZLL Commissioning_.  I'll do a PR with updated DDF.

BurningTheSky, 2023-09-10:
> Thanks a lot. This seams to work 👍 I can add the switch, they show up in Phoscon App and look like they work pretty much as intendet 👍
>
> I only changed one small thing: The line "{"S_BUTTON_2": "Dim [...]

Mind if I ask how did you added the switch? Can it be done via Phoscon or does it have to go another way? I'm new to deCONZ so excuse my simple question!

closed 2023-09-16 by ebaauw