dresden-elektronik/deconz-rest-plugin #7149: LXN59-2S7LX1.0 doesn't updata update relay status

LeoeLeoeL, 2023-08-05:
3A Smart Home DE model LXN59-2S7LX1.0 (2 gang switch) doesn't update relay status when using wall-switch

## Describe the bug
When I switch on/off the device using the wall-switch the status is not updated in Phoscon  and Home Assistant

## Steps to reproduce the behavior

## Expected behavior
I expect an update of the status

## Screenshots

## Environment

 - Host system: Raspberry P)
 - Running method: Raspbian
 - Firmware version: 0x26780700
 - deCONZ version: v2.23.0-beta and previous versions
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? APC UPS

## deCONZ Logs

## Additional context
Here there is the same problem but in Zigbee2Mqtt
https://github.com/Koenkk/zigbee2mqtt/issues/4245

Mimiix, 2023-08-06:
Can you add a screenshot of the basic cluster so we know what we are dealing with :)?

LeoeLeoeL, 2023-08-06:
[image]

BabaIsYou, 2023-08-09:
Seems that this device is already suppoted by legacy code (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3764).
Could you please provide the DDF auto-generated by legacy code to have a look at it ?

LeoeLeoeL, 2023-08-09:
This one?

{
  "schema": "devcap1.schema.json",
  "manufacturername": "3A Smart Home DE",
  "modelid": "LXN59-2S7LX1.0",
  "product": "LXN59-2S7LX1.0",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

BabaIsYou, 2023-08-10:
May be could try to add some bindings into a new DDF 
```
{
  "schema":"devcap1.schema.json",
  "manufacturername":"3A Smart Home DE",
  "modelid":"LXN59-2S7LX1.0",
  "product":"LXN59-2S7LX1.0",
  "sleeper":false,
  "status":"Gold",
  "subdevices":[
    {
      "type":"$TYPE_ON_OFF_LIGHT",
      "restapi":"/lights",
      "uuid":[
        "$address.ext",
        "0x01"
      ],
      "items":[
        {
          "name":"attr/id"
        },
        {
          "name":"attr/lastannounced"
        },
        {
          "name":"attr/lastseen"
        },
        {
          "name":"attr/manufacturername"
        },
        {
          "name":"attr/modelid"
        },
        {
          "name":"attr/name"
        },
        {
          "name":"attr/swversion"
        },
        {
          "name":"attr/type"
        },
        {
          "name":"attr/uniqueid"
        },
        {
          "name":"state/alert",
          "default":"none"
        },
        {
          "name":"state/on",
          "refresh.interval":5
        },
        {
          "name":"state/reachable"
        }
      ]
    },
    {
      "type":"$TYPE_ON_OFF_LIGHT",
      "restapi":"/lights",
      "uuid":[
        "$address.ext",
        "0x02"
      ],
      "items":[
        {
          "name":"attr/id"
        },
        {
          "name":"attr/lastannounced"
        },
        {
          "name":"attr/lastseen"
        },
        {
          "name":"attr/manufacturername"
        },
        {
          "name":"attr/modelid"
        },
        {
          "name":"attr/name"
        },
        {
          "name":"attr/swversion"
        },
        {
          "name":"attr/type"
        },
        {
          "name":"attr/uniqueid"
        },
        {
          "name":"state/alert",
          "default":"none"
        },
        {
          "name":"state/on",
          "refresh.interval":5
        },
        {
          "name":"state/reachable"
        }
      ]
    }
  ],
  "bindings":[
    {
      "bind":"unicast",
      "src.ep":1,
      "dst.ep":1,
      "cl":"0x0006"
    },
    {
      "bind":"unicast",
      "src.ep":2,
      "dst.ep":1,
      "cl":"0x0006"
    }
  ]
}
```

BabaIsYou, 2023-08-18:
If you are not comfortable with handling DDF files, the binding can be added via the DDF editor of the DecONZ user interface and saved in a specific DDF file specific to your environment.

LeoeLeoeL, 2023-08-18:
Sorry for not answering but I am far from home so I cannot test your DDF yet.
Anyway, you are right. I'm not comfortable with DDF. I had to make a new file using the "notepad" in Raspbian :-)

BabaIsYou, 2023-08-18:
You can also just save the proposed DDF made by legacy code and change Draft by Gold. Then you can also add bindings from DDF editor and save DDF again and do a hot reload.

LeoeLeoeL, 2023-08-20:
It works!
Thank you very much, tou solved me a problem.

P.S. Is it possible to put this DDF in the new releases of Deconz?

BabaIsYou, 2023-08-20:
> It works! Thank you very much, tou solved me a problem.
> 
> P.S. Is it possible to put this DDF in the new releases of Deconz?

We have to make a pull request for that. Do you want to make it ? Else I’ll do it ;-)

LeoeLeoeL, 2023-08-20:
It's better if you do that.
Thank you

PR #7192 (2023-08-21): DDF add support for lxn59-2s7lX1.0
Close #7149
closed 2023-08-21 by manup