dresden-elektronik/deconz-rest-plugin #7151: Tuya _TZE204_ac0fhfiq TS0601 BiDirectional ZigBee Energy Meter

decrouxtom, 2023-08-06:
## Device

- Product name: BiDirectional ZigBee Energy Meter with 150A Clamp Current Sensor
- Manufacturer: _TZE204_ac0fhfiq
- Model identifier: TS0601
- Device type : Sensor

- HA quirk: https://github.com/zigpy/zha-device-handlers/issues/2420

## Screenshots

[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]

### Other clusters that are not mentioned above

Smanar, 2023-08-06:
Ha nice so much informations.

Have started a DDF but effectively there is a problem for the dp = 0x06. But it's not from the device, as we can see all the needed data on the payload.
The problem is more how deconz manage Raw value.

```
        switch (dataType)
        {
        case TuyaDataTypeRaw:
        case TuyaDataTypeString:
            return result; // TODO implement?
```

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_ac0fhfiq",
  "modelid": "TS0601",
  "product": "BiDirectional ZigBee Energy Meter",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "state/consumption_2",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 0, "eval": "Item.val = Attr.val / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 0, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 0, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

If I make a code update to suppport RAW value, are you able to compile it to test ? You just need a linux machine with full OS ?

decrouxtom, 2023-08-06:
Wahoo, so fast! Thanks a lot :)
I only have a raspberry running Raspbian 5.10.103, but if I just need to install GCC or equivalent, I should be able to do that. I will need some guidance to test but I'm sure there is a wiki somewhere!

Smanar, 2023-08-07:
If value or not hexa, we will try with code 
`
chain = chain.toHex()`

decrouxtom, 2023-08-07:
Great, I will try in the coming days.
Thanks

Smanar, 2023-08-07:
Sure, tell me if you need more explanations.
The code compile, but I realy have no idea what can be the result ^^

decrouxtom, 2023-08-07:
Hi!

Smanar, 2023-08-08:
It's consumption and power sensor, so not supported by phoscon.
You can see them on domoticz or in phoscon/help/Api Information/sensors.

It's not so bad

`TY_DATA_REPORT ....... dpid 0x06 type 0x00 ... lenght :8, val 09 28 00 09 a6 00 01 fc`

0928 =>234 V
0009a6 => 2.4A
0001fc =>508W

P = U * I * cos @ => 234 * 2.4 = 561 not so bad.

Now the bad new ^^, I m cheating on c++ code 
```
            DBG_Printf(DBG_INFO, "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %s\n",
                       rt, seq, dpid, dataType, dataLength, qPrintable(chain.toHex()));
```

I convert the value to hex before display it, so in reality the value can be not converted.
We can check direclty on the DDF

Try replacing 

```
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 0, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

by
```
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = (Attr.val[0] << 8) | Attr.val[1]) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

I m trying to get the 2 first bytes for tension.
No need to re-include the device, just restart deconz or make a "hot reload".
Checking logs will be usefull, if the value is not a string but a number it will cause a JS error, and same you will see value to compare with result, to check if the value is in hexa or not.

decrouxtom, 2023-08-08:
Thanks for the explanation !
About qt, yes, I've got an error, missing package or something similar. Installing qt5 fixed the issue.

With what you suggest (adding missing bracket ;)), result is 0.
I've tried just Attr.val[1] but JS TypeError.
If I understand, Attr.val is a string. We cannot index it, as Attr.val[i] is a char, not a byte. Maybe something to extract substring, not a bitshift?

About domoticz, I get for consumption 1 & 2, 2 different types of sensor, whereas if I still understand, they are both consumption, not energy ?
[image]

decrouxtom, 2023-08-08:
Try : 
`"Item.val = parseInt(Attr.val.substring(0,4),16) / 10;"`
but returns nan :(
I let the expert!

Smanar, 2023-08-08:
The first one is power (WIP)
The second one if energy, and this one is working if I m right. BTW now I remember the domoticz plugin support state/consumption but not state/consumption_2. So You can't see it yet in widget (will add it later, for the moment the raw values)

You have tried to check logs ? with flag "info" and "ddf", we can see the value in entry. I m not sure if deconz handle it correclty.

Smanar, 2023-08-08:
I have an idea, if the logging haven't worked.

>"Item.val = Attr.val.length;"

The result will an int, and can check if the string is empty or not.

There is a flag "JS" as debug too.

decrouxtom, 2023-08-09:
> You have tried to check logs ? with flag "info" and "ddf", we can see the value in entry. I m not sure if deconz handle it correclty.
Sorry, I forgot to paste a snapshot!
Value in entry seems ok, but DDF output was 0 or nan.
I will do the snapshot with the other tests soon...

BabaIsYou, 2023-08-09:
> BTW now I remember the domoticz plugin support state/consumption but not state/consumption_2. So You can't see it yet in widget (will add it later, can make a widget with 2 energy on same widget, but for the moment the raw values)

In the mean time may be you could just create an second "pseudo"   consumption sensor for reverse current by adding this into DDF 
```
  {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
```

decrouxtom, 2023-08-12:
Hi,
I'm doing some tests...
[image]
and with JS:
[image]

decrouxtom, 2023-08-12:
Some issue with the 3 bytes info:
[image]

decrouxtom, 2023-08-12:
Looks good!!
[image]

decrouxtom, 2023-08-12:
> > In the mean time may be you could just create an second "pseudo" consumption sensor for reverse current by adding this into DDF
> 
> I will see later, not the more important, better to have power/tension/current first, making the domoticz widget will be easier. But there is a widget where you can have consumption AND production on the same widget (I m using it on my side to have HP and HC on the same graphic)
> 
> I don't have production on my side, so idk what kind of widget is the more usefull

Maybe we can used this widget ?
https://github.com/guillaumezin/DomoticzLinky
production & consumption with HP/HC !

Smanar, 2023-08-12:
>Maybe we can used this widget ?

Lol, yes, it's exactly the one I m speaking

>BTW, power seems to be in W, not kW

Yes, it's the defaut unit in deconz and the one used on the DDF no ?

>In domoticz, do I need to modify the DDF to also create one sensor for each voltage/current?

Nope, there is an hidden setting "ENABLEMORESENSOR" https://github.com/Smanar/Domoticz-deCONZ#configuration
But it will enable it for all your devices.

>Other question, with my old RFLink, I get a "General kWh" device in domoticz, with power and consumption on the same device. Is it possible to change type in DDF, or do I need to manage that a domoticz level ?

Here it's a good question, as your device use only the tuya cluster we can easily mix all fields on the same ZHASensor, but the problem will be for "standardization".
If one day, someone, domoticz dev or deconz dev, change something on ZHAConsumptionn it will impact all thoses sensors type, and same if this happen on ZHAPower, if a sensor "mix" the 2 types it will probably need special stuff.

Handling consumption and consumption_2 on the same widget can be usefull for me because you can easily compare 2 dailies consumptions, but not sure the widget that mix consumption and power will be usefull for all users. So I prefer you manage that on domoticz level.

On your debug we can see the data have a lenght = 2, data is 08f80019c8 ....
So for me it's a string, not hexa values, why ?
- the 3tr byte is 0x00 so it s a "end string" value, it will explain the len = 2
- others values are 08 and f8, but in decimal so not in 0-9 A-f range (they need to be in 0x30>0x39 0x97>0x66), and this can explain the JS error.

So have updated the c++ code, to make a try (will be faster)
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

So now I think Attr.val will be an hexa string, so no more JS error.

>It could also be useful to get power direction ;)

I m checking all field possible and I don't see one we can use .....
But from this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7098 it's possible to add them dynamically, so let's try

On the DDF
```
        {
          "name": "state/direction",
          "parse": {"fn": "tuya", "dpid": 102, "if (Attr.val == 0) { Item.val = 'consuming' } else { Item.val = 'producing' };" },
          "read": {"fn": "none"}
        }
```

On the folder `devices\generic\items` the "system" one, not the "user", create the file state_direction_item.json
```
{
	"schema": "resourceitem1.schema.json",
	"id": "state/direction",
	"datatype": "String",
	"access": "R",
	"public": true
}
```

I m checking the domoticz code to mix the 2 consumption ? but you have production too ?

decrouxtom, 2023-08-13:
Hi!
I added missing ";" and compiled:
[image]

I've created _state_direction_item.json_ in _/usr/share/deCONZ/devices_. Is it the right place ?
[image]
> but you have production too ?

Not yet !

Do you understand why power is both in dpid 101 and 6 ? looks the same value, execept 1 byte more for 101. Should we use this power ?

Smanar, 2023-08-13:
>Do you understand why power is both in dpid 101 and 6 ? looks the same value, except 1 byte more for 101. Better to use this field, no ? Or one is for consumption, the other for consumption-production ?

Good question, perhaps it s the more usefull value, and easier to read on a dedicated dp ?
But on z2m there are not using the 101 too.

So state/direction is working too ?

I think all will work with this code.

```
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = (Attr.val[0] << 8) | Attr.val[1]) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = parseInt('0x' + Attrval.substr(0, 4)) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = parseInt('0x' + Attrval.substr(10, 6)) / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

BTW if you can share the final DDF pls.

>Not yet ! I understood that power is only for producing. I can try inverting the plug if you want.

No, nothing urgent, I have edited the domoticz plugin to support consumption_2 (on the beta branch), was to test it, but can wait someone need it.

decrouxtom, 2023-08-13:
Answer with plug inverted...
`state/direction expression: if (Attr.val == 0) { Item.val = 'consuming' } else { Item.val = 'producing' }; --> producing`
power is the same on dpid 101 and 6:
`dpid: 0x06, type: 0x00, length: 8, val: 08fe003d41000e12`
`dpid: 0x65, type: 0x02, length: 4, val: 3602` (0xe12=3602)

So I'll keep dpid=101 in DDF
[_TZE204_ac0fhfiq_Energy_meter.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/12329476/_TZE204_ac0fhfiq_Energy_meter.json.txt)

decrouxtom, 2023-08-13:
> So state/direction is working too ?

Yes!

But with substr, I cannot see anything in the log with INFO+DDF+JS :(

Smanar, 2023-08-13:
It work on my side, but I m using a JS console. Perhaps

`prototype. substr(…) is not strictly deprecated (as in “removed from the Web standards”), it is considered a legacy function and should be avoided when possible. It is not part of the core JavaScript language and may be removed in the future. If at all possible, use the substring() method instead.`

Try with replacing `substr `by `substring`

decrouxtom, 2023-08-13:
> It work on my side, but I m using a JS console. Perhaps
>
> `prototype. substr(…) is not strictly deprecated (as in “removed from the Web standards”), it is considered a legacy function and should be [...]

Same :(

Smanar, 2023-08-13:
This one is working on my console

```
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = parseInt('0x' + Attr.val.substr(10, 16));" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = parseInt('0x' + Attr.val.substr(0, 4)) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = parseInt('0x' + Attr.val.substr(4, 10)) / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

>Same :(

And no error to explain the "no support" ?

Smanar, 2023-08-13:
And with 

```
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "s=String(Attr.val);Item.val = parseInt('0x' + s.substr(0, 4)) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

decrouxtom, 2023-08-13:
`Item.val = parseInt('0x' + Attr.val.substring(4, 10)) / 1000;`
or
`s=String(Attr.val);Item.val = parseInt('0x' + s.substring(0, 4)) / 10;`
are not working.
BUT
` Item.val = parseInt('0x' + Attr.val.toString().slice(4, 10)) / 1000; ` works ;)

decrouxtom, 2023-08-13:
deCONZ UI seems to not handle "," correctly. It adds quotes and carriage returns when using slice(x,y)
Editing file manually solves the issue
We do not need the toString() conversion:
[image]

[_TZE204_ac0fhfiq_Energy_meter.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/12329635/_TZE204_ac0fhfiq_Energy_meter.json.txt)

Smanar, 2023-08-13:
>deCONZ UI seems to not handle ","

Ha yes, I forget to say that, there is same issue if you want to add more than one manufacture name (they are separated by a comma too)

So all is working (at least on deconz side) ? You want to make the PR to add the device officialy or want I make it ?

decrouxtom, 2023-08-13:
> So all is working (at least on deconz side) ?

YES!!!
Please, could you do the PR and check my DDF ? I've tried to push in your branch unsuccesfully :(

Smanar, 2023-08-13:
PR done here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7182

>And about Domoticz, do I need to create another request in another github ? Do you want I test the beta version ?

If you can yes, but idk how will work the "production mode" on your device.
You can use the domoticz plugin github if you have problem (it don't depend on deconz), you know the procedure to install the beta version ?

decrouxtom, 2023-08-13:
> PR done here #7182

Thanks!

> > And about Domoticz, do I need to create another request in another github ? Do you want I test the beta version ?
> 
> If you can yes, but idk how will work the "production mode" on your device. You can use the domoticz plugin github if you have problem (as this part don't depend on deconz), you know the procedure to install the beta version ?

I will have a look. For production, it looks like power is always the absolute difference between production and consumption. Better to wait someone having solar panels, not in the coming months for me! 
In the meantime, I've done a script based on papoo's one to merge energy & power in domoticz. If someone needs, I can share it for sure.

> BTW, thx for help, have just added swversion on the DDF. it's a pleasure with users with your knowledge.

It was a pleasure to work with you ;)

Smanar, 2023-08-14:
>it looks like power is always the absolute difference between production and consumption

Ha yes, there is only 1 "magnetic ring", so you measure the difference, so it mean you need to produce more than you consume to increase the `consumption_2`, else you will just have lower `consumption`

closed 2023-09-13 by github-actions[bot]