dresden-elektronik/deconz-rest-plugin #7183: Tuya Wifi MmWave

Rihan9, 2023-08-16:
## Device

- Product name: Tuya MmWave presence sensor
- Manufacturer: _TZE204_qasjif9e
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Luminance Sensor
  - Presence Detection

## Screenshots

[image]

### Basic
[image]

Group,scene and Tuya specific clusters don't contains anything useful.

This device should be related to #6998, maybe one is the ceiling sensor and the other is the wall sensor

Mimiix, 2023-08-16:
Hi,

Please add all screenshots. Even if they seem to be useless.

Rihan9, 2023-08-16:
Ok:

## Groups
[image]
[image]

##  Scenes
[image]
[image]
[image]

## Tuya Specific
[image]

Mimiix, 2023-08-16:
Can you please add them to the original post?

Rihan9, 2023-08-16:
Done :)

Mimiix, 2023-08-16:
Great!
Thanks :)

Rihan9, 2023-08-16:
@Mimiix I was following the guide font here to open a new device request:
https://forum.phoscon.de/t/read-me-device-request-read-me/23

I can't complete the second step (I can't complete the registration process on the blog because the confirmation mail doesn't arrive and I can't access using github because the page replies with a 502 error). I was wondering if I could skip this step.

Mimiix, 2023-08-16:
Doesn't matter that much. Just leave it now. Il look into the issue.

BabaIsYou, 2023-08-17:
Could you try with this DDF. If it works we may be optimise it later with "custom" items to avoid some side effects.

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "__TZE204_qasjif9e",
	"modelid": "TS0601",
	"product": "Tuya MmwRadar Mini human breathe sensor",
	"sleeper": false,
	"status": "Gold",
	"subdevices": [{
			"type": "$TYPE_LIGHT_LEVEL_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0400"
			],
			"items": [{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"refresh.interval": 86400,
					"read": {
						"at": "0x0001",
						"cl": "0x0000",
						"ep": 1,
						"fn": "zcl"
					},
					"parse": {
						"at": "0x0001",
						"cl": "0x0000",
						"ep": 1,
						"fn": "zcl",
						"script": "tuya_swversion.js"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/lastupdated"
				},
				{
					"name": "state/dark"
				},
				{
					"name": "config/tholddark"
				},
				{
					"name": "config/tholdoffset"
				},
				{
					"name": "state/daylight"
				},
				{
					"name": "state/lightlevel"
				},
				{
					"name": "state/lux",
					"description": "The current light intensity in Lux (max is 2000)",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 104,
						"script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
						"fn": "tuya"
					},
					"default": 0
				}
			]
		},
		{
			"type": "$TYPE_PRESENCE_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0xef00"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"refresh.interval": 86400,
					"read": {
						"at": "0x0001",
						"cl": "0x0000",
						"ep": 1,
						"fn": "zcl"
					},
					"parse": {
						"at": "0x0001",
						"cl": "0x0000",
						"ep": 1,
						"fn": "zcl",
						"script": "tuya_swversion.js"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/delay",
					"description": "Detection delay : target confirmation delay time in 10th of second (from 0 to 100). Default 0.1s",
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 101,
						"dt": "0x2b",
						"eval": "Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 101,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 1
				},
				{
					"name": "config/duration",
					"description": "Fading time : target disapperance delay time in second (from 0 to 15000). Default 90s",
					"range": [0, 15000],
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 102,
						"dt": "0x2b",
						"eval": "Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 102,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 90
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "config/sensitivity",
					"description": "Radar sensitivity (from 1 to 9). Default 7",
					"range": [1, 9],
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 2,
						"dt": "0x2b",
						"eval": "Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 2,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 7
				},
				{
					"name": "config/triggerdistance",
					"description": "Far detection sensitivity",
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 4,
						"dt": "0x2b",
						"eval": "Far = ['0','1','2','3','4','5','6','7','8','9']; Attr.val = Math.max(0,Far.indexOf(Item.val)*100);",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 4,
						"eval": "Item.val = String(Attr.val/100);",
						"fn": "tuya"
					},
					"values": [
						["\"1\"", "1m"],
						["\"2\"", "2m"],
						["\"3\"", "3m"],
						["\"4\"", "4m"],
						["\"5\"", "5m"],
						["\"6\"", "6m"],
						["\"7\"", "7m"],
						["\"8\"", "8m"],
						["\"9\"", "9m"],
						["\"0\"", "Off"]
					],
					"default": "10"
				},
				{
					"name": "state/angle",
					"description": "Distance from detected target in cm (from 0 to 1000).",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 9,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "state/lastupdated"
				},
				{
					"name": "state/presence",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 1,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": false
				}
			]
		}
	],
	"bindings": [{
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0xEF00"
	}]
}
```

Rihan9, 2023-08-17:
Hi @BabaIsYou, it seems to be an extra underscore on the "manufacturername" param.
If I remove that it seems to load correctly.

Is not yet perfect:

[image]
[image]

but at list is a presense sensor and not a smart plug :D.

I'm seeing a strange behavior on the presence sensor. It constantly changing between detected and not detected, but I'm right in front o f the sensor (1m or so).

The light sensor instead does not work: it stuck on 0 lux

BabaIsYou, 2023-08-17:
Hi,

The behavior you mention about changing from detected and not detected even if angle is different of 0 is exactley what I referenced as "side effects". That's because we reuse the delay item that is used for something else in legacy code.
It's documented here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6998#issuecomment-1677574695

Regarding light sensor, it's normal because the DDF is not in it's definitive place. To have it operational you may copy /usr/share/deCONZ/devices/generic/illuminance_cluster/lux_to_lightlevel.js to the subfolder /generic/illuminance_cluster of were you put your DDF (/home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices ?)

BabaIsYou, 2023-08-17:
I suggest to add these files into `/usr/share/deCONZ/devices/generic/items/
`
1. state_targetdistance_item.json

```
{
	"schema": "resourceitem1.schema.json",
	"id": "state/targetdistance",
	"description": "Distance from detected target",
	"datatype": "Int16",
	"access": "R",
	"public": true,
	"default": 0
}
```

2. config_fadingtime_item.json

```
{
	"schema": "resourceitem1.schema.json",
	"id": "config/fadingtime",
	"datatype": "UInt16",
	"access": "RW",
	"public": true,
	"description": "Target disapperance delay time in second (from 0 to 15000). Default 90s",
	"range": [0, 15000],
	"default": 0
} 
```

I modified the previous DDF accordingly

Rihan9, 2023-08-17:
Hi @BabaIsYou I applied the changes mentioned by you on the previous comments.
Now the light sensor seems to work correctly: I don't have another sensor to use for the reference so I don't know if the value is correct, but at least is responding to environmental changes.

[image]

But I don't see the new configuration on home assistant. I can see them loaded on the DDF:
[image]
[image]

The presence sensor seems more stable (the value 2lux on the light sensor is because I faced the sensor to you wall to check the presence sensor). I'm seeing a different value for the lux sensor between home assistant and phoscon app, but I don't thing is the right place to discussing it

BabaIsYou, 2023-08-17:
Well that's a good beginning ! ;-)
This kind of devices are not a reference about lux measure accuracy ... But that's not what is usually wanted about these kind of device. It's more about is there a light or not ! 

I also realize that no `"read": {"fn": "tuya"}` is present (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-support-Tuya-manufacturer-specific-cluster-&-How-to-know-which-datapoints-a-Tuya-device-provides%3F) then I updated one more time the DFF (see `presence` item).

Don't really know about HA, but Phoscon values are from REST-API (based onto this DDF) then you could check using Phoscon/Help/API information/Sensors and then you could get what REST-API returns as json for your sensors.

Rihan9, 2023-08-18:
Hi @BabaIsYou I applied the last DDF and the presence sensor seems to be worst. 

I checked on the Phoscon/Help/API information/Sensors as you suggested:
[image]

"Presence" seems to be stuck on "false" now, but the target distance seems to respond to the environmental changes (me approaching/moving away). The other configurations variables still don't show on HA, but on the api explorer (screenshot above) seems to be readable. 
Maybe I but the files "state_targetdistance_item.json" and "config_fadingtime_item.json" on the wrong path?

This is my current opt/deCONZ folder tree:
[image]

Rihan9, 2023-08-18:
Ok I:
- deleted the device from phoscon app
- purged the old device from home assitant
- re-added the sensor on phoscon app
- re-synced HA with Deconz

Now the situation is:
[image]
[image]

It's again a smart plug. Why this thing hate me? The positive aspect is that now the "presence" sensor works fine

BabaIsYou, 2023-08-18:
Ouch ! Weird ... First of all, can you see if the DFF has been loaded into Deconz GUI with the right items. It has been done one time (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7183#issuecomment-1682952740) so I think that the json and js files were at the right place. because they reported some values even for the custom items.

Rihan9, 2023-08-18:
Hi,
here a bunch of screenshots:
[image]
[image]
[image]
[image]
[image]

I posted 2 screenshots of the DDF to show you the last change applied on "Presence". For me it seems to be loaded correctly.

BabaIsYou, 2023-08-18:
All seems to be fine regarding API view. Just to try something. Could add "`refresh.interval": 5`, line to the DDF : 

```
				{
					"name": "state/presence",
					"refresh.interval": 5,
					"read": {
						"fn": "tuya"
					},
					"parse": {
						"dpid": 1,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": false
				}
```

Rihan9, 2023-08-18:
Here the smart plug 4 screenshot:
[image]
please note: this screenshot is before applying the changes on the DDF

Mimiix, 2023-08-18:
So the device works, the only issue is the name?!

BabaIsYou, 2023-08-18:
> So the device works, the only issue is the name?!

Not really, we're supposed to have 2 sensors (light level and presence) and no smart plug. If I understand we currently have 1 sensor (presence) and 1 smart plug

Rihan9, 2023-08-18:
It's not so simple. For summing up:
- the light sensor is working fine
- the presence sensor seems a little instable for now
- a bunch of configuration entities don't show up on HA throught the official integration (need to check if it is the HA integration with a problem or something still missing on deconz
- The device is still viewed as smart plug, and a smart plug entities is created both on deconz and in HA

BabaIsYou, 2023-08-18:
> the light sensor is working fine

I miss that one sorry ! 

> * the presence sensor seems a little instable for now

It's still instable now ?

Mimiix, 2023-08-18:
> It's not so simple. For summing up:
>
> * the light sensor is working fine
> * the presence sensor seems a little instable for now
> * a bunch of configuration entities don't show up on HA throught the [...]

Stability is not with the DDF. That being said it's probably the device itself.

Be advised: Home assistant uses the same api as phoscon. If phoscon shows correctly, home assistant is not up to date with the api info. Best is to disable and re enable the integration  / restart home assistant.

Rihan9, 2023-08-18:
@BabaIsYou We currently have 1 presence sensor, 1 light sensor and 1 smart plug.

I'm cheching for the stability after the last change (refresh.interval). I think I need to adjust the distance and sensitivity parameters, but the rest seems to working ok.

I'm uploading a history graph for the presence sensor. The room is 4 meter large and the sensor is placed on the corner. A person was sleeping on the most further wall. The graph is yellow when a presence is detected, grey otherwise. Black when I'm resetting the device to apply a DDF change. As you can see is a bit unstable, but I don't think is something to correct on deconz but something to adjust on sensitivity setting.
[image]

Rihan9, 2023-08-18:
@BabaIsYou I cleaned HA and deconz from the device and removed the orphan device from ha, restart all the system and now the smart plug is not here anymore. I'm going to check for an update on HA, but it should be already the latest stable release 2023.8.0

BabaIsYou, 2023-08-18:
Good news !

Rihan9, 2023-08-18:
Can you please let me know how to do it? Even a link to a reference guide is ok

Mimiix, 2023-08-18:
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Here we go

Rihan9, 2023-08-18:
It was easy than I thought.
Here the screenshot:
[image]
[image]

The previous sensitivity value was 7, so I think is working just fine.

Rihan9, 2023-08-18:
@BabaIsYou I noticed now that the parameter "fadingtime" is not showing up on the API with the latest DDF

BabaIsYou, 2023-08-18:
WTF ! The DDF is valid ! Are the .json files for these items still into permament Deconz folder ?

Rihan9, 2023-08-18:
I was checking and I think something went wrong on my instance. I loaded a DDF file with some change and then reload the DDF of this thread, after reupload the other files. Now they are visible correcly. I don't know what went wrong but I think I have done so many tests that I'm unable to find the root cause. I don't thing is something worth investigating though.

BabaIsYou, 2023-08-18:
It's already documented somewhere in forum that DDF editor won't always reload DDF as expected ...

Kane610, 2023-08-19:
Some comments from what home assistant is currently expecting 

trigger distance has previously been expected to be strings (far, medium, near) but now reported as a number in string "10" with unknown max value 
sensitivity has previously been expected to be between 1-3 but now reported as a 7 also with unknown max value

Smanar, 2023-08-19:
I m agree for distance, need to use a string, can use a table to convert.
But sensitivity haven't maximum value, it's like setting a maximum for a temperature, it's not possible.

Kane610, 2023-08-19:
> Sensitivity haven't maximum value for me, it's like set a maximum for a temperature, it's not possible.

It's a configuration property and there is an additional property called sensitivitymax not often used and I don't rely on it. Of course there has to be a known max and min for setting sensitivity.

Smanar, 2023-08-19:
I can't make test ATM, but it seem it's coded on deconz.
`	"range": [0, 255],`

So if the user set a value out of range, the API will just refuse it and the value will not change, so on HA you can just let the user use the value he want ?

Rihan9, 2023-08-19:
Can I suggest to create a new item called, for example "config/triggerdistance_m" to use when the sensor have this setting expressed in meter instead of a generic "far", "medium", "near"?

Smanar, 2023-08-19:
But it make more work to plugin dev, need to support all new item, and with tuya can have new one every day ^^

Rihan9, 2023-08-19:
> But it make more work to plugin dev, need to support all new item, and with tuya can have new one every day ^^

yeah, you are not wrong. But in this case I think is worth considering it, because this is a new generation type of presence sensor and I think is going to replace the old PIR sensor in the long run and using "far", "medium" and "near" is going to limiting it. And I think is more complex to try to use 2 difference value domains in the same item for the third app plugin dev insteand implementing a mew item, what do you think @Kane610 ?

Smanar, 2023-08-19:
```
        {
          "name": "config/sensitivitymax",
          "static": 2
        },
```

It seem this is already used, so can take 0 as min and this one as max.

For distance, it's another story ....

BabaIsYou, 2023-08-19:
> Can I suggest to create a new item called, for example "config/triggerdistance_m" to use when the sensor have this setting expressed in meter instead of a generic "far", "medium", "near"?

It’s quite a workaround between precision of this kind of device, the genuine use of triggerdistance that was not enough precise. And if future device calculates it in inches ? Do we need to create another item ? 🤔
But I like the idea of a numeric item for that purpose. 

> It's a configuration property and there is an additional property called sensitivitymax not often used and I don't rely on it. Of course there has to be a known max and min for setting sensitivity.

The discussion about sensitivity max has already been done 7 months ago https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6621#issuecomment-1374477988
It’s a readonly item and has vocation to stay that way.

I suggest all of you to have a look at issue #6621 to understand from where we came with the tenth of this kind of device that we already support on the same pattern.

That beeing said … all is always perfectible !😀

Kane610, 2023-08-20:
> But I like the idea of a numeric item for that purpose.

Sure but you shouldn't break the current api definition just because it feels good.

> The discussion about sensitivity max has already been done 7 months ago [#6621 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6621#issuecomment-1374477988)
> It’s a readonly item and has vocation to stay that way.

Of course it should be read only, what Im asking for is providing it with the device, so it's possible for different devices to provide different ranges and clients to easily support those variations.

> I suggest all of you to have a look at issue #6517 and PR #6621 to understand from where we came with the tenth of this kind of device that we already support on the same pattern, and avoid discussions that have already been for some of them.

Im giving feedback on earlier established API behaviours, just because there are more examples of not considering the original intents of the properties doesn't make this implementation right. From my point of view they all seem to break API promise.

What Im really after here is consistency within the API given a specific type of sensor (scalability, I don't want to need to re-implement the same sensor multiple times). I don't want to need to consider if trigger distance are strings or a value range. That should be abstracted on deconz side. For me you can change far/medium/near to 3/2/1 and be consistent with these newer sensors.

BabaIsYou, 2023-08-20:
> Of course it should be read only, what Im asking for is providing it with the device, so it's possible for different devices to provide different ranges and clients to easily support those variations.

For this device the need is about a configuration item, configurable by the user then it can’t be readonly. But it can be added to give a max value dependant of device, but what for ?

> What Im really after here is consistency within the API given a specific type of sensor (scalability, I don't want to need to re-implement the same sensor multiple times). I don't want to need to [...]

Reducing the precision value returned in cm or m to only 3 options seems not a good solution IMO.

Kane610, 2023-08-20:
> Reducing the precision value returned in cm or m to only 3 options seems not a good solution IMO.

I mean the other way around, make the device with far/medium/near support configuration between 1-3 and this other device 1-10 and provide a trigger distance max property.

> For this device the need is about a configuration item, configurable by the user then it can’t be readonly. But it can be added to give a max value dependant of device, but what for ?

If devices support different ranges then you need to explain what the upper limit is.

> For me API aim is to present/interact with value of a device with the less transformation, nearest to the raw but human readable value.

It needs to be some layer of abstraction as deconz provides many different devices and they need to fit within the same template.

> But may be I’m wrong, thus be my guest to propose changes into the DDF.

I shouldn't need to be involved with DDF work, agreeing on REST API output should be enough.

Rihan9, 2023-08-23:
Hi All,
since I don't think you have decided what to do with this new type of device, I'm migrating my sensor to another platform. If you find a definitive solution and want to test it, please tag me and let me know.

TC1666, 2023-08-26:
> Hi All, since I don't think you have decided what to do with this new type of device, I'm migrating my sensor to another platform. If you find a definitive solution and want to test it, please tag me and let me know.

I did exactly the same thing when dealing with one of these Tuya variants, gave up with deconz. Installed Z2M and another ZigBee transceiver and the device was recognized instantly with all the parameters properly exposed.

Moving forward I'll just keep deconz/ConbeeII for my legacy battery sensors.

closed 2023-08-31 by manup

thespawnat, 2023-12-18:
hi, is there any update on the device-issue? I'm currently facing the same problem (default integrated as light without custom DDF) and have tried huge amount of customized DDF-Files, but it's still not working.  with this [DDF-File](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7184/commits/426c288732ed396b0610c4343ca1f46580294ff8), the Sensor was detected as "LightSensor" + "Presence-Sensor", but no proper values are shown. 1,0lx on the light-sensor (it should be now 34lx) and no presence detected so far. In HomeAssistant is only the "Delay"-Config possible, but not as much settings like in ZHA. in "Phoscon" the sensor is only presented as "light-sensor", not as presence sensor. 

I've updated my conbee2 firmware to the latest available on the weekend, deconz container is up2date from HA-Addons.

Removed the device from Deconz, called HA-Service to refresh deconz devices / remove old devices, restarted deconz, restarted HA, re-paired the sensor, still same behavior after multiple attempts.

I've attached the "currently used" DDF File as TXT [DDF_File.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13701459/DDF_File.txt)

[image]
[image]
[image]
[image]

regards

PR #7184 (2023-08-31): DDF add support for TS0601_TZE204_qasjif9e & fix other mmWave Radar Presence DDF
Add support for _TZE204_qasjif9e and add two custom items to avoid side effect of reusing `config/duration` for instance [#6998](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6998#issuecomment-1677574695)

Fix #6998  (breaking change)

Close #7183 
Close #6517