dresden-elektronik/deconz-rest-plugin #7185: TS110E dimmer  _TZ3210_k1msuvg6

pittivale, 2023-08-18:
## Device

- Product name: 1CH Zigbee Dimmer Module (bought as GIRIER zigbee dimmer)
- Manufacturer: GIRIER? ( _TZ3210_k1msuvg6 on basic cluster )
- Model identifier: TS110E   ( _TZ3210_k1msuvg6  )
- Device type :  Light (dimmer)

I observe a strange behaviour. If i turn on the light, after about 6 minutes (every single time) the light stays on, but the device reports the off state (OnOff attribute is “false” and the light appears off in the Phoscon App and Home Assistant).

Furthermore, I have noticed that doing a read of the OnOff attribute sets it to “false”, even if it was “true”, before doing the read. Nevertheless, the light stays on.

I should say also that I have a previously bought version of the same dimmer ( _TZ3210_zxbtub8r ) which works without issues.

## Screenshots

### Basic
[image]
[image]
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
![onoff1](https://github.com/dresden-elektronik/deconz-rest-plugin/assets/142609759/eff908d3-fd69-43bf
[image]
-b218-4bebc22ed54f)

### Level Control
[image]

Smanar, 2023-08-18:
It seem this device need so much hack to work, at least for dimmer part.
One of them is the use of tuya unlock sequence.

Will be usefull to save the DDF for this device and add inside
```
        {
          "name": "config/tuya_unlock"
        },
```

Them re-include the buggy device.
All the hack concern the dimmer part, so I hope the tuya unlock sequence is enought to solve the on/off part.

pittivale, 2023-08-18:
Thank you for your answer!

Unfortunately, adding the tuya unlock sequence just changed the problem to the complementary one: the device now stops reporting the off position after about 6 minutes, (or after any read of te OnOff attribute), even though the light is actually off.

Do you know anything else I could try?

Smanar, 2023-08-19:
So it mean now it return an "on" request even the light is off, and before was the reverse ?

I m checking on other zigbee project, only the set level feature is crazy.

pittivale, 2023-08-19:
I also did check the other zigbee project threads, but I seem to have a different problem than the one discussed there. What happens to me is that every read of the OnOff attribute, has the unwanted effect of setting said attribute to a value, even if it is not the real state of the light. It was the "false" value before adding the tuya unlock sequence, and it is now the "true" value.
During Normal Use, I am able to send commands to the light correctly, but if i do nothing, the regular polling of the device (which on my network happens about every 6 minutes) can change the reported state to the wrong one (when the light is on, before changing the ddf, and when the light is off after changing it).

This is the DDF now (I added the unlock sequence to the ddf of the working dimmer version _TZ3210_zxbtub8r ):

`
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_k1msuvg6",
  "modelid": "TS110E",
  "product": "TS110E",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock",
          "public": false,
          "description": "Tuya sequence to unlock some device.",
          "refresh.interval": 86400,
          "default": false
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
`

Smanar, 2023-08-19:
So the attribute 0x0000 in cluster on/off 0x0006 was false when the light was on before the tuya unlock sequence.
And now the attribute is true when the light are off ?

BTW your DDF is not used, you have "draft" as status.

Here you have a DDF with the gold status and a part to set report/bind in the DDf itself.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_k1msuvg6",
  "modelid": "TS110E",
  "product": "TS110E",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/bri",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 1800
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 5,
          "max": 1800,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

pittivale, 2023-08-19:
Thank you very much for your help.

I loaded your DDF. The behavior is the following: I turn the light on, the device reports the on state for a few seconds  (less than one minute), thant it starts reporting the Off state forever, but the light stays on. At this point if i send the On command, the light stays on and reports the On state for a few seconds, then back to reporting the Off state forever

Smanar, 2023-08-19:
>I loaded your DDF.

The tuya unlock feature work every 24h or at inclusion, perhaps you need to re-include the device.

But if the attribute is bad, I can do lot of things .... This device don't have the tuya cluster in the cluster list ? (some device have on/off cluster but are using the tuya one)

pittivale, 2023-08-19:
I tried re-inclusion, and unfortunately no change in behavior.

The device has a "Tuya specific" cluster, but with no attribute.

This is the screenshot:
[image]

Smanar, 2023-08-20:
>The device has a "Tuya specific" cluster, but with no attribute.

Tuya haven't attributes ^^, because it's tuya, not standard zigbee.
But can explain the issue, the old version have the tuya cluster too ?

A test you can done, enable log in deconz, deconz / help / debug view, with flag "info" and "info_l2", then press the switch, we will be able to see if it use the cluster 0x0006 or 0xEF00. Lot of tuya device have broken cluster 0x0006 because they are using the tuya one.

pittivale, 2023-08-20:
Thank you for the explanation. The Tuya specific cluster is also present in the old functioning version of the dimmer ( _TZ3210_zxbtub8r  )

Here are the logs regarding the buggy device (_TZ3210_k1msuvg6):

At interval we have this part:

`
11:52:09:251 Websocket 172.30.32.1:43972 send message: {"attr":{"id":"33","lastannounced":null,"lastseen":"2023-08-20T09:52Z","manufacturername":"_TZ3210_k1msuvg6","modelid":"TS110E","name":"Test Dimmer Buggy","swversion":null,"type":"Dimmable light","uniqueid":"a4:c1:38:a4:34:db:8e:bb-01"},"e":"changed","id":"33","r":"lights","t":"event","uniqueid":"a4:c1:38:a4:34:db:8e:bb-01"} (ret = 326)
`

This is what appears when I turn the light on:
`
11:55:23:630 add task 367055 type 14 to 0xA4C138A434DB8EBB cluster 0x0006 req.id 17
11:55:23:631 emit event/tick: 0xa4c138a434db8ebb
11:55:23:635 Websocket 172.30.32.1:43972 send message: {"e":"changed","id":"33","r":"lights","state":{"alert":"none","bri":254,"on":true,"reachable":true},"t":"event","uniqueid":"a4:c1:38:a4:34:db:8e:bb-01"} (ret = 152)
11:55:23:698 	0xA4C138A434DB8EBB force poll (2)
11:55:23:699 Erase task req-id: 17, type: 14 zcl seqno: 173 send time 0, profileId: 0x0104, clusterId: 0x0006
11:55:23:712 ZCL attribute report 0xA4C138A434DB8EBB for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
11:55:23:713 	payload: 00001001
11:55:23:714 SC state/on: synced
11:55:23:714 SC --> StateFinished
11:55:23:714 SC state change finished: a4:c1:38:a4:34:db:8e:bb-01
`

This is what appears in the logs when the device starts reporting the Off state while the light stays on

`
11:55:45:121 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'state/on', 'false', 'dev', 1692525345 FROM sub_devices WHERE uniqueid = 'a4:c1:38:a4:34:db:8e:bb-01'
11:55:45:126 Websocket 172.30.32.1:43972 send message: {"e":"changed","id":"33","r":"lights","state":{"alert":"none","bri":254,"on":false,"reachable":true},"t":"event","uniqueid":"a4:c1:38:a4:34:db:8e:bb-01"} (ret = 153)

`

For completness, these are the logs of the old functioning version ( _TZ3210_zxbtub8r):

Polling at intervals:
`
12:07:17:851 poll node a4:c1:38:87:bc:4e:2d:0c-01
12:07:17:852 Poll light node Pulsante Luci Sottopensili
12:07:17:904 read attributes of 0xA4C13887BC4E2D0C cluster: 0x0006: [ 12:07:17:904 0x0000 12:07:17:905 ]
12:07:17:905 add task 370051 type 19 to 0xA4C13887BC4E2D0C cluster 0x0006 req.id 61
12:07:17:905 Poll APS request 61 to 0xA4C13887BC4E2D0C cluster: 0x0006
12:07:17:990 Poll APS confirm 61 status: 0x00
12:07:17:991 Erase task req-id: 61, type: 19 zcl seqno: 27 send time 0, profileId: 0x0104, clusterId: 0x0006
12:07:18:030 Node data 0xa4c13887bc4e2d0c profileId: 0x0104, clusterId: 0x0006
12:07:18:031 0xA4C13887BC4E2D0C: update ZCL value 0x01/0x0006/0x0000 after 0 s
`

When turning light on:
`
12:07:34:181 add task 370121 type 14 to 0xA4C13887BC4E2D0C cluster 0x0006 req.id 169
12:07:34:184 Websocket 172.30.32.1:43972 send message: {"e":"changed","id":"16","r":"lights","state":{"alert":"none","bri":0,"on":true,"reachable":true},"t":"event","uniqueid":"a4:c1:38:87:bc:4e:2d:0c-01"} (ret = 150)
12:07:34:195 add task 370123 type 14 to 0x680AE2FFFE7F9FB0 cluster 0x0006 req.id 171
12:07:34:241 Websocket 172.30.32.1:43972 send message: {"e":"changed","id":"22","r":"lights","state":{"alert":"none","bri":254,"on":true,"reachable":true},"t":"event","uniqueid":"68:0a:e2:ff:fe:7f:9f:b0-01"} (ret = 152)
12:07:34:258 add task 370126 type 14 to 0xA4C13887BC4E2D0C cluster 0x0006 req.id 174
12:07:34:259 delay sending request 174 dt 0 ms to 0xA4C13887BC4E2D0C, ep: 0x01 cluster: 0x0006 onAir: 1
12:07:34:260 delay sending request 174 ep: 0x01 cluster 0x0006 to 0xa4c13887bc4e2d0c onAir 1
12:07:34:298 delay sending request 174 dt 0 ms to 0xA4C13887BC4E2D0C, ep: 0x01 cluster: 0x0006 onAir: 1
12:07:34:299 delay sending request 174 ep: 0x01 cluster 0x0006 to 0xa4c13887bc4e2d0c onAir 1
12:07:34:304 	0xA4C13887BC4E2D0C force poll (2)
12:07:34:305 Erase task req-id: 169, type: 14 zcl seqno: 43 send time 0, profileId: 0x0104, clusterId: 0x0006
12:07:34:309 Node data 0xa4c13887bc4e2d0c profileId: 0x0104, clusterId: 0x0006
12:07:34:310 0xA4C13887BC4E2D0C: update ZCL value 0x01/0x0006/0x0000 after 0 s
12:07:34:311 ZCL attribute report 0xA4C13887BC4E2D0C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:07:34:311 	payload: 00001001
12:07:34:363 	0xA4C13887BC4E2D0C force poll (2)
12:07:34:363 Erase task req-id: 174, type: 14 zcl seqno: 45 send time 0, profileId: 0x0104, clusterId: 0x0006
12:07:34:397 Node data 0xa4c13887bc4e2d0c profileId: 0x0104, clusterId: 0x0006
12:07:34:397 0xA4C13887BC4E2D0C: update ZCL value 0x01/0x0006/0x0000 after 0 s
12:07:34:399 ZCL attribute report 0xA4C13887BC4E2D0C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:07:34:399 	payload: 00001001
`

When turning light off:
`
12:07:47:140 add task 370181 type 14 to 0xA4C13887BC4E2D0C cluster 0x0006 req.id 7
12:07:47:145 Websocket 172.30.32.1:43972 send message: {"e":"changed","id":"16","r":"lights","state":{"alert":"none","bri":0,"on":false,"reachable":true},"t":"event","uniqueid":"a4:c1:38:87:bc:4e:2d:0c-01"} (ret = 151)
12:07:47:219 	0xA4C13887BC4E2D0C force poll (2)
12:07:47:220 Erase task req-id: 7, type: 14 zcl seqno: 58 send time 0, profileId: 0x0104, clusterId: 0x0006
12:07:47:285 Node data 0xa4c13887bc4e2d0c profileId: 0x0104, clusterId: 0x0006
[... cut ...]
12:07:47:288 ZCL attribute report 0xA4C13887BC4E2D0C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
12:07:47:289 	payload: 00001000
12:07:47:290 Websocket 172.30.32.1:43972 send message: {"attr":{"id":"16","lastannounced":"2023-08-16T20:52:53Z","lastseen":"2023-08-20T10:07Z","manufacturername":"_TZ3210_zxbtub8r","modelid":"TS110E","name":"Pulsante Luci Sottopensili","swversion":"0x00000041","type":"Dimmable light","uniqueid":"a4:c1:38:87:bc:4e:2d:0c-01"},"e":"changed","id":"16","r":"lights","t":"event","uniqueid":"a4:c1:38:87:bc:4e:2d:0c-01"} (ret = 361)
`

Smanar, 2023-08-20:
And I don't understand why this device need a poll.

>This is what appears when I turn the light on:

Physicaly/manually I mean, not using the API, so see normal working mode.

>11:55:23:712 ZCL attribute report 0xA4C138A434DB8EBB for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000

This line is the more important, it's the normal reporting, it's the device that send this information itself, and bad luck it seem it use the cluster 0x0006, so if this one have bad value on attribute 0x0000, I don't see how to have a good one.

pittivale, 2023-08-20:
Do you have an idea of what these lines of the logs mean?

11:55:45:121 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'state/on', 'false', 'dev', 1692525345 FROM sub_devices WHERE uniqueid = 'a4:c1:38:a4:34:db:8e:bb-01'
11:55:45:126 Websocket 172.30.32.1:43972 send message: {"e":"changed","id":"33","r":"lights","state":{"alert":"none","bri":254,"on":false,"reachable":true},"t":"event","uniqueid":"a4:c1:38:a4:34:db:8e:bb-01"} (ret = 153)

The erroneous report starts at this moment...

Smanar, 2023-08-20:
Nothing usefull for you, the interested part is before.

>11:55:45:121 INSERT INTO resource_items (sub_device_id,item,value,source,timestamp) SELECT id, 'state/on', 'false', 'dev', 

Just mean deconz is writing information in the database (so mean it have already the information)

>11:55:45:126 Websocket 172.30.32.1:43972 send message: {"e":"changed","id":"33","r":"lights","state":{"alert":"none","bri":254,"on":false,"reachable":true},"t":"event","uniqueid":"a4:c1:38:a4:34:db:8e:bb-01"} (ret = 153)

Is just the websocket message make by deconz to third app.

AOskam, 2023-09-08:
Exactly the same behaviour here. Can I provide a specific log here that would help?

AOskam, 2023-09-08:
I’m on ZHA via skyconnect. HAOS 10.5
Pve via proxmox.

Mimiix, 2023-09-08:
> I’m on ZHA via skyconnect. HAOS 10.5 Pve via proxmox.

deCONZ is not supporting ZHA or Skyconnect. Your in the wrong place :)

mwitlox, 2023-09-08:
Same problem here. I have more then one. 1ch and 2ch. Tried ZHA and Z2MQTT with Home Assistant. They al report the wrong state some of them wrong on state some of them wrong off state. If i can provide any info to fix let me know

closed 2023-10-07 by github-actions[bot]