dresden-elektronik/deconz-rest-plugin #7187: frient Smart Plug Mini 2 (SPLZB-141)

fadamsen, 2023-08-19:
## Device

- Product name: Smart Plug Mini 2
- Manufacturer: frient A/S
- Model identifier: SPLZB-141
- Device type:
  - Switch

The device shows up as a switch, but sensors for power and consumption are not added.

## Screenshots

### Basic
[image]

[image]

### Simple Metering
[image]

[image]

### Electrical Measurement
[image]

[image]

[image]

[image]

[image]

SwoopX, 2023-08-19:
You could modifiy the current DDF for the old version (file `splzb-131_smart_plug.json` in the frient folder) and change the following lines from

```
  "manufacturername": ["frient A/S", "frient A/S"],
  "modelid": ["SPLZB-131", "SPLZB-134"],
```

to

```
  "manufacturername": ["frient A/S", "frient A/S", "frient A/S"],
  "modelid": ["SPLZB-131", "SPLZB-134", "SPLZB-141"],
```

fadamsen, 2023-08-19:
@SwoopX, I've tried that, but Deconz doesn't pick up the change. I'm running Deconz through Home Assistant as an addon (latest version of both), and despite being able to open the modified DDF file and issuing the Force Reload command, nothing happens.

Smanar, 2023-08-19:
Select the node on the GUI, and select "edit DDF", if it's not the good one that appear there is a problem (you will the path and the name on the window title)

On HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

fadamsen, 2023-08-20:
I managed to make it work! It might have been an issue with line endings, in any case it was solved when I used a copy of `/usr/share/deCONZ/devices/frient/splzb-131_smart_plug.json` and placed it in `/data/.local/share/dresden-elektronik/deCONZ/devices`. That directory survives restarts and updates of the deCONZ addon, so I'm good for now.

I see @SwoopX already went ahead and created #7189 to merge the change, which is great, so everyone else will be able to use the smart plug without having to mess with DDFs.

closed 2023-08-21 by manup