dresden-elektronik/deconz-rest-plugin #7190: ROBB SMARRT Zigbee Dimmer Pro 400W

Preto, 2023-08-20:
## Device

- Product name: Zigbee dimmer 2-draads
- Manufacturer: ROBB smarrt
- Model identifier: ROB_200-011-0
- Device type : 
  - Light
  - Sensor

## Screenshots
### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]

### Diagnostics
[image]

Although the device is is listed in the Supported Devices, it is not working. The light is automatically discovered, but the sensor for Power Measurement is not being found.

I've put a lot of effort in getting it to work myself, but can't seem to figure it out. Please help me to integrate this device so both light and sensor work.

If something is missing or wrong I would appreciate if you don't close this Device Request but let me supply the additional information required.

Mimiix, 2023-08-20:
Did you check phoscon or the api when checking for the power measurement?

If only phoscon, please check the api.

Preto, 2023-08-20:
I've checked the API as it seems that the light sensors never show in Phoscon under sensors.
The API (and Phoscon) show only the light, but not the sensors of that particular light (`ZHAConsumption` or `ZHAPower`).

Smanar, 2023-08-20:
Can try this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"ROBB smarrt",
   "modelid":"ROB_200-011-0",
   "product":"Smart Zigbee Dimmer Pro 400W",
   "sleeper":false,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_DIMMABLE_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x01"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/alert",
               "default":"none"
            },
            {
               "name":"state/bri"
            },
            {
               "name":"state/on"
            },
            {
               "name":"state/reachable"
            }
         ]
      },
      {
         "type":"$TYPE_CONSUMPTION_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0702"
         ],
         "items":[
[... 173 lines cut ...]
               "min":5,
               "max":500,
               "change":"0x00000003"
            },
            {
               "at":"0x0508",
               "dt":"0x21",
               "min":5,
               "max":300,
               "change":"0x0000000F"
            },
            {
               "at":"0x050B",
               "dt":"0x29",
               "min":5,
               "max":300,
               "change":"0x00000003"
            }
         ]
      }
   ]
}
```

There is no capture for the complete node with all cluster deployed , so IDK wich one endpoint it use, I have take the 0x01 by defaut,  you can change them if needed.

Preto, 2023-08-21:
Thanks @Smanar for your time!

I've placed the DDF and restarted deCONZ. It immediately added the sensor automatically, no reset of the device was needed!

With "_no capture of the complete node_" you mean a screenshot was missing? Am I correct that the one bellow is the correct one?

[image]

I can read the following values from the API now:

| Type | Light off | Light on |
|---|---|---|
| ZHAConsumption | Consumption: 1226129 | Consumption: 1236139|
| ZHAPower | Current: 14<br>Power  : 0<br>Voltage: 314 | Current: 127<br>Power  : 201<br>Voltage: 1830 |

The **Voltage** seems a bit high for voltage (V) and I'm not sure about Consumption. What are the all the units here?

Smanar, 2023-08-21:
>Am I correct that the one bellow is the correct one?

Perfect, so we are fine the endpoint 0xf2 have nothing usefull for us.
Hu ? Strange voltage .... IDK what is measuring the device, why this one is moving .....
Nevermind P = U* I > 23 W = 183 V * 0.127 A

23 W is not to bad for a light, you know the power of bulbs ? So the multiplier/divisor are exactly the good one finally

[image]

With corrective the DDF will be

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ROBB smarrt",
  "modelid": "ROB_200-011-0",
  "product": "Smart Zigbee Dimmer Pro 400W",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/bri"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val / 1000"
          },
          "refresh.interval": 365
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 365
        },
        {
          "name": "state/voltage",
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 365
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 1800
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 5,
          "max": 1800,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 5,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 5,
          "max": 500,
          "change": "0x00000003"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 5,
          "max": 300,
          "change": "0x0000000F"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 5,
          "max": 300,
          "change": "0x00000003"
        }
      ]
    }
  ]
}
```

Preto, 2023-08-22:
> 23 W is not to bad for a light, you know the power of bulbs ? So the multiplier/divisor are exactly the good one finally

The lightsource contains 3 times a 7.5 W dimmable LED. So in total that would be around 23 W.

I've loaded your DDF today and it looks as follows from the API:

| Type | Light off | Light on (max) | Light on (min)
|---|---|---|--|
| ZHAConsumption | Consumption: 1249 Wh | Consumption: 1251 Wh | Consumption: 1253 Wh
| ZHAPower | Current: 15 mA<br>Power  : 0 Wh<br>Voltage: 36 V | Current: 127 mA<br>Power  : 21 Wh<br>Voltage: 195 V| Current: 30 mA<br>Power  : 0 Wh<br>Voltage: 22 V

Am I correct on the units now?

Smanar, 2023-08-22:
For me yes, 0.127 * 195 = 24 and uou have 21, for information power is in W not Wh.
21W during 1 hour give 21 more W/h to consumption.

Preto, 2023-08-31:
@Smanar: Again, thanks a lot!

I did found out something weird:

At complete random moments once a day when a light is on, it seems to powercycle. The light turns off and after a second back on. Is it possible we're getting the information too quickly from the dimmer that it reboots itself?

A second issue that I found is that deCONZ or the API turns on the light automatically. This only happens with brightness changes. I can reproduce this as followed:

1. I turn on the light via API with low brightness (`{ "on": true, "bri": 60 }`), and leave it on
2. I send another API request to turn on the light, this time at max brightness (`{ "on": true, "bri": 254 }`), still leaving it on
3. I wait an optional 5 or 10 seconds and turn of the light manually with the physical button and the light turns off as expected.
4. Within a few seconds the light goes on again, without me touching the physical button, or send an API request

When I remove the DDF, both the powercycle issue and the automagically-turn-on issue will no longer occur.

Any clue?

Smanar, 2023-09-01:
You can see in logs if a request is from deconz, or if it's the device that chnage state itself alone ?

Some magic from tuya ? Can try without bind/report, on normal device, you will be able to send zigbee request, but not having return.

This part is for the on/off

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 1800
        }
      ]
    },
```

>When I remove the DDF

Just removing the DDF, no re-inclusion ? (so the setting on the device aren't modified)

Preto, 2023-09-06:
> You can see in logs if a request is from deconz, or if it's the device that change state itself alone ?

Looking at the INFO debug log I see the following for the device in question:
```
21:40:52:939 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:40:58:225 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:41:03:512 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:42:41:149 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:42:46:420 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:42:51:702 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

[ • Enabling light on low brightness • ]

21:42:53:070 emit event/tick: 0xXXXXXXXXXXXXXXXX
21:42:53:209 delay sending request 188 dt 0 ms to 0xXXXXXXXXXXXXXXXX, ep: 0x01 cluster: 0x0008 onAir: 1
21:42:53:256 delay sending request 188 dt 0 ms to 0xXXXXXXXXXXXXXXXX, ep: 0x01 cluster: 0x0008 onAir: 1

21:42:53:258 	0xXXXXXXXXXXXXXXXX force poll (2)
21:42:53:301 SC state/on: synced
21:42:53:303 SC --> StateFinished
21:42:53:304 SC state change finished: XX:XX:XX:XX:XX:XX:XX:XX-01
21:42:53:327 	0xXXXXXXXXXXXXXXXX force poll (2)

21:42:54:032 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:42:54:044 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

[ • Enabling Light on high/full brightness • ]

21:42:55:111 emit event/tick: 0xXXXXXXXXXXXXXXXX
21:42:55:156 delay sending request 216 dt 0 ms to 0xXXXXXXXXXXXXXXXX, ep: 0x01 cluster: 0x0008 onAir: 1

21:42:55:169 	0xXXXXXXXXXXXXXXXX force poll (2)
21:42:55:212 	0xXXXXXXXXXXXXXXXX force poll (2)

21:42:56:082 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:42:56:099 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:42:56:987 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:42:58:957 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

[ • Turning off the light with the physical switch • ]

21:42:58:958 SC state/on: not synced

21:42:59:302 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:43:01:355 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0702, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:43:01:373 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:43:02:273 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:43:07:558 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

21:43:11:067 SC tick --> StateCallFunction
21:43:11:068 SC_SetOnOff()

[ • Now the light turns on magically! This time its quickly, but sometimes it takes a few seconds before it turns on • ]

21:43:11:216 SC state/on: synced
21:43:11:217 SC --> StateFinished
21:43:11:218 SC state change finished: XX:XX:XX:XX:XX:XX:XX:XX-01

21:43:12:119 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:43:12:130 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
21:43:12:845 ZCL attribute report 0xXXXXXXXXXXXXXXXX for cluster: 0x0B04, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
```

If I re-do all the above steps that are shown between brackets, but this time press the physical switch to turn off the light **after** I see the following automatically in the logs, then the light stays off as it should:

```
22:18:01:157 SC state/on: synced
22:18:01:157 SC --> StateFinished
22:18:01:158 SC state change finished: XX:XX:XX:XX:XX:XX:XX:XX-01
```
In some occasions I spot the following (note the one extra logline) but the result is the same (light stays off):

```
22:16:27:944 SC tick --> StateRead state/on, XX:XX:XX:XX:XX:XX:XX:XX-01
22:16:28:012 SC state/on: synced
22:16:28:013 SC --> StateFinished
22:16:28:013 SC state change finished: XX:XX:XX:XX:XX:XX:XX:XX-01
```

It seems at the first brightness change that StateFinished follows up quickly. But after the second brightness change that StateFinished takes a long time. Shutting down the light will then turn on the light again.

Smanar, 2023-09-07:
And you have same result if you use only the command "on" and not the command "bri" ?

The problem is I don't see what the DDF can add to the legacy code to provoke that .... (If it work without problem, without the DDF).

Something you can try is deleting the DDF, deleting the device, re-in cluding the device, so you will have it in deconz but with legacy code, select the node, edit DDF, deconz will generate an incomplete one but based on legacy, save the DDF (remember to set the status to "gold"), and check if with the autogenerated DDF you have same problem (you will miss sensors like that.)

closed 2023-10-06 by github-actions[bot]