dresden-elektronik/deconz-rest-plugin #7191: _TZE200_68nvbio9 ... Zemismart Rechargeable Tuya Zigbee Roller Blinds Tube Motor

Ltek, 2023-08-20:
## Device
Product name: Zemismart Rechargeable Tuya Zigbee Roller Blinds Tube Motor
Manufacturer: _TZE200_68nvbio9
Model identifier: TS0601
Device type: Window Covering

According to this page (https://github.com/zigpy/zha-device-handlers/issues/1528) looks like there are several Tuya compatible shades that are identical.

# For tuya devices 0 = Up/Open, 1 = Stop, 2 = Down/Close
TUYA_COVER_COMMAND = {
    "_TZE200_zah67ekd": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_xuzcvlku": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_3i3exuay": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_nueqqe6k": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_gubdgai2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_zpzndjez": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TYST11_wmcdj3aq": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_yenbr4om": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_5sbebbzs": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_xaabybja": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_hsgrhjpf": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_iossyxra": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_68nvbio9": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_zuz7f94z": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_ergbiejo": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_nhyj64w2": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},
    "_TZE200_uzinxci0": {0x0000: 0x0000, 0x0001: 0x0002, 0x0002: 0x0001},

## Screenshots
[image]
[image]
[image]

Smanar, 2023-08-21:
Hello to make short deconz is not able to support tuya covering using DDF.

On this topic you will found a hack to use them as light and a code to test to make the suport possible https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5960#issuecomment-1616634196

Ltek, 2023-08-23:
> Hello to make short deconz is not able to support tuya covering using DDF.
> 
> On this topic you will found a hack to use them as light and a code to test to make the suport possible [#5960 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5960#issuecomment-1616634196)

DDF was supposed to be 'better' but if it cant support Tuya -- that is 'not better'

Deconz not supporting a well known and very large producer of Zigbee devices is VERY concerning.

Smanar, 2023-08-23:
If you have a linux machine with full OS we can make test to update deconz ?

Ltek, 2023-08-24:
Can you compile it for Windows?  I dont have a PC with Linux 

DDF was supposed to make device support easier so I'm baffled how it seems to not be able to support all device types, properties and parameters -- like Tuya (prob the 2nd largest zigbee manufacturer in the world?)

Looks like both ZHA and Z2M support many Tuya devices natively. I'm invested in Conbee/Deconz and dont want to change, but if Conbee/Deconz has problems supporting devices, sorry to say -- it may just be wiser to look at adopting Z2M.

Mimiix, 2023-08-24:
> Can you compile it for Windows? I dont have a PC with Linux
>
> DDF was supposed to make device support easier so I'm baffled how it seems to not be able to support all device types, properties and [...]

Can you please stay on topic here? You made yourselves clear before and continuing drama isnt making it better. Deconz supports tuya ,just not blinds. If you don't like things, feel free to move.  From now on: please stay on topic. Any comments I recommend opening a topic on the forums.

chbndrhnns, 2023-08-24:
> Deconz supports tuya ,just not blinds

And that's a design limitation or technical problem or it has just has not yet been implemented?

Mimiix, 2023-08-24:
> > Deconz supports tuya ,just not blinds
> 
> And that's a design limitation or technical problem or it has just has not yet been implemented?

That's something for @Smanar to answer. My guess would be a difference tuya created in the zigbee spec and it hasn't been implemented.

Smanar, 2023-08-24:
>it has just has not yet been implemented?

It's not a suppport problem, some devices can work with native code, and the modification in c++ code just take 2 lines to edit, I can give you the line to changes if you want, but you will nee to recompile the code too.

DDF just don't support sending covering request if the device don't respect zigbee standard.

Like I have said before, there is a code to test, I can finish it and propose a PR, but I can't without tests. And IDK if someone else is making something for this problem.

Ltek, 2023-08-24:
> > > Deconz supports tuya ,just not blinds
> > 
> > 
> > And that's a design limitation or technical problem or it has just has not yet been implemented?
> 
Deconz supports tuya ,just not blinds

And that's a design limitation or technical problem or it has just has not yet been implemented?
> That's something for @Smanar to answer. My guess would be a difference tuya created in the zigbee spec and it hasn't been implemented.

Answering these questions are very important from systems perspective (having a 30yr career in computer systems design)... if Deconz is not able to easily (in the DDF) account for variations (Tuya put those in long before Deconz moved to DDF) then its not going ever have universal Zigbee.  Tuya has adding unique parameters (@Mimiix, Tuya has not changed the zigbee spec, they cannot do that) and thus shows that any manufacture, at any time, could do the same and then is no longer compatible with Deconz.  The Deconz hack (which people say works) to make the shade motor look like a light is evidence the issue is likely a shortcoming in Deconz.

btw, this is fully on-topic since the only way to "solve" for this device getting supported is either A) update Deconz native code to account for the Tuya variations -or- B) enhance Deconz DDF format to allow configuring for Tuya variations

Smanar, 2023-08-24:
> Tuya has not changed the zigbee spec, they cannot do that

No but they interpret them, provoke a battery leak if you set a bind like you do on all other device for exemple, or describe a motion sensor as PLUG

Ltek, 2023-08-24:
> > Tuya has not changed the zigbee spec, they cannot do that
> 
> No but they interpret them, provoke a battery leak if you set a bind like you do on all other device for exemple, or describe a motion sensor as PLUG
> 
> Create a special cluster making all Tuya device not compatible with group with all others brands.
> 
> There is tuya and there is all other brands, it's for that for lot of people it's not repecting zigbee standards, even they are not making something "not autorised by zigbee standard"

Yes, this is common -- usually because manufactures (or firmware writers) are lazy. This happens on all industries.
A good system design accounts for this common problem, its no different than common coding practices like error trapping, exception handling, data transformation, etc. 

If Deconz is written so it can let the DDF tell it things like...
--> Overwrite property / parameter XYZ with value ABC
--> Read 'Tuya cluster' and move each value <basic, etc>

This is a common solution even from ike DDF (just like JSON, INI, etc... way back from caveman days of coding).

Smanar, 2023-08-24:
The problem have nothing to see with tuya specialy.
When DDF was created, only standard request was possible (so no tuya at all), later DDF was able to read all possible incoming request (so all sensors can be added), but still missing something for DDF are able to make "custom" request.
We miss the support for covering, but classic one are working because they still use native code
We miss too support for exemple for warning device, we are able to know the device state, but we can't send a siren order.

The problem is in the DDF core, but there is lot of more usefull thing to do.

If you want to help, I m free, and I already have a code to test to have support in DDF for covering. Not sure it will be included, but we will have a code. I just need someone with a full unix OS and a tuya covering.
Generaly users are from HA, so docker OS and this kind of OS is too much complex for them, so not possible to make tests with them.

Ltek, 2023-08-26:
> The problem have nothing to see with tuya specialy. When DDF was created, only standard request was possible (so no tuya at all), later DDF was able to read all possible incoming request (so all Tuya sensors can be added), but still missing something for DDF are able to make "custom" request. We miss the support for covering, but classic one are working because they still use native code We miss too support for exemple for warning device, we are able to know the device state, but we can't send a siren order.
> 
> The problem is in the DDF core, but there is lot of more usefull thing to do.
> 
> If you want to help, I m free, and I already have a code to test to have support in DDF for covering. Not sure it will be included, but we will have a code to propose. I just need someone with a full unix OS and a tuya covering. Generaly users are from HA, so docker OS and this kind of OS is too much complex for them, so not possible to make tests with them.
> 
> [Smanar/deconz-rest-plugin@master...Smanar:deconz-rest-plugin:covering_new](https://github.com/Smanar/deconz-rest-plugin/compare/master...Smanar:deconz-rest-plugin:covering_new)
> 
> And this PR add too the config/reversemode

If you want to send me the Windows code I'll install/test. I have 2 model different Tuya shades I can test.

Smanar, 2023-08-26:
We need a linux OS, but can work on virtual machine (If you know better than me how to manage them ?)
ATM we are trying with @chbndrhnns but he need to make a complete setup.

Smanar, 2023-09-03:
@Ltek Do you still want to make a try ^^.
@chbndrhnns have issue to install the setup and have give up, he is using now new device that use the covering cluster.

Ltek, 2023-09-09:
I the covers worked first try in Z2M. I don't have time to mess around. Seems like you could look at how z2h works and make deconz work for it  too

Mimiix, 2023-09-09:
> I the covers worked first try in Z2M. I don't have time to mess around. Seems like you could look at how z2h works and make deconz work for it too

Please do as Smanar asks. If you can't , and nobody else can I have to close this issue as it has no use if nobody can test.

Ltek, 2023-09-09:
All the info you need is avail since  Z2M clearly has it in the definition for that cover, otherwise it wouldn't work. I spent like 5 hours last time trying to help you get a temp sensor working.. . And you never did. Sorry if I'm not willing to do that again.

Mimiix, 2023-09-09:
We are not putting untested code in the main branch. So if you are not willing to do so, it's not going to happen if nobody else will.

Mimiix, 2023-09-09:
Besides, you didn't do anything for me. I am not a developer, just a volunteer.

@Smanar this can't be done without testing correct?

Ltek, 2023-09-09:
He asked me to put up a Linux system... I don't have one, not going to try to do it. Want me to test compiled windows code...sure can do

Ltek, 2023-09-11:
Will you be providing a compiled windows EXE to test?

Smanar, 2023-09-11:
I don't think the project is compilable on windows.

Mimiix, 2023-09-11:
> I don't think the project is compilable on windows.

In that case we are stuck.

Ltek, 2023-09-11:
Deconz runs on windows...why can't deconz support this Tuya device like it
has with other Tuya devices before?
Is Deconz just not going to support Tuya at all? ...tgus not support the
largest zigbee device maker globally?

closed 2023-10-11 by github-actions[bot]

Ltek, 2023-10-19:
update, hope this helps others... This works perfectly on Zigbee2MQTT. I use both Z2M and Deconz as it seems Deconz will not work with some Tuya devices, Z2M does. But, for me, Deconz works better with Aqara motion sensors (I have a lot of them)