dresden-elektronik/deconz-rest-plugin #7193: Zemismart Smart Vibration Sensor "Zigebee"

Rocket200, 2023-08-21:
- Product name: Zemismart 
- Manufacturer: _TZ3000_fkxmyics
- Model identifier: TS0210
- Device type : Vibration / Movement sensor

Screenshots

[image]

[image]

[image]

I hope I did not forget anything

with the DDF from [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4904#issuecomment-1479595650) i get every Second a Vibration withapout touching the Device

and on Discord i open a [help-thread](https://discord.com/channels/494922323518947329/1142777255949504592/1142777258730344488) with some information but i think useless

Mimiix, 2023-08-21:
Hi, can you add the power config cluster screenshot :)?

Mimiix, 2023-08-21:
Thanks :)

@smanar you able to help out ?

Rocket200, 2023-08-21:
I can try later when i back Home 
What u change all in the DDF ?

Rocket200, 2023-08-21:
i change it to this with the code file can not load 
json file problem

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_fkxmyics",
  "modelid": "TS0210",
  "vendor": "Tuya",
  "product": "Moes vibration sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHAPresence",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/duration",
          "default": 1
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/sensitivity",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0013",
            "cl": "0x0500",
            "ep": 1
          },
          "write": {
            "at": "0x0013",
            "cl": "0x0500",
            "dt": "0x20",
            "ep": 1,
            "eval": "Item.val"
          },
          "parse": {
            "at": "0x0013",
            "cl": "0x0500",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/presence",
          "awake": true,
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}`

but same bevor every second in the phoscom App

Rocket200, 2023-08-22:
[TS0210.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/12406455/TS0210.zip)
this one working i can load the code and Device get all Info an i can Pair it , but same Problem bevor every second 

[TS0210 new.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/12406462/TS0210.new.zip)
this one i can  load it but its not saved and the Device can not Pair with this one

Smanar, 2023-08-22:
And with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_bmfw9ykl", "_TYZB01_3zv6oleo", "_TYZB01_j9xxahcl", "_TYZB01_kulduhbj", "_TYZB01_cc3jzhlj", "_TYZB01_geigpsy4", "_TZ3000_fkxmyics"],
  "modelid": [ "TS0210", "TS0210", "TS0210", "TS0210", "TS0210", "TS0210", "TS0210"]
  "vendor": "Tuya",
  "product": "Moes vibration sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "TYPE_VIBRATION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/sensitivity",
          "range": [
            0,
            6
          ],
          "parse": {
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0013",
            "eval": "Item.val = Attr.val"
          },
          "read": {
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0013"
          },
          "write": {
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0013",
            "dt": "0x20",
            "eval": "Item.val"
          },
          "refresh.interval": 86400
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/vibration",
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

But remove the device before installaing the new DDF, the old one is a presence sensor with state/duration, the new one is a ZHAVibration without duration (the timing is at 1s, so can explain your result)  and with state/vibration.

Mimiix, 2023-08-22:
@Smanar  Your missing a , after the model ID array.

Smanar, 2023-08-22:
> @Smanar Your missing a , after the model ID array.

Arf, right, have corrected the last DDF.

Rocket200, 2023-08-22:
[8/22/2023, 4:25:46 PM] [deCONZ] Presence 8: Status Fault: set to 1 (from 0)
not more come now from Homebridge
and it stays there , always on , in phoscaon web it looks its change every second

Rocket200, 2023-08-22:
17:34:13:942 [INFO] - No button map for: TS0210, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 010400640000, zclSeq: 21

17:35:56:386 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
17:35:57:760 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
17:35:58:995 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
17:36:03:232 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
17:36:04:319 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 

from debug view

Smanar, 2023-08-22:
>[8/22/2023, 4:25:46 PM] [deCONZ] Presence 8: Status Fault: set to 1 (from 0)

With the last DDF it's no more a presence but a vibration ?

>17:34:13:942 [INFO] - No button map for: TS0210, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 010400640000, zclSeq: 21

And this you have it only 1 time ?

Rocket200, 2023-08-22:
It's show as Presence 

No Button map come all the time

Smanar, 2023-08-22:
>No Button map come all the time

So at every vibration or even without vibration ?

>It's show as Presence

Can you take a look in phoscon/help/API INformation/sensors
You will see the device json, it need to be a ZHAVibration, else something have failed, remember to delete the device before adding the new DDF.

Rocket200, 2023-08-22:
> Can you take a look in phoscon/help/API INformation/sensors
> You will see the device json, it need to be a ZHAVibration, else something have failed, remember to delete the device before adding the new DDF.

`},
        "ep": 1,
        "etag": "f6ee4b409de70cecd71f56ed1052c390",
        "lastannounced": "2023-08-22T19:52:59Z",
        "lastseen": "2023-08-22T19:55Z",
        "manufacturername": "_TZ3000_fkxmyics",
        "modelid": "TS0210",
        "name": "TS0210",
        "state": {
            "lastupdated": "2023-08-22T19:56:13.095",
            "lowbattery": false,
            "vibration": true
        },
        "type": "ZHAVibration",
        "uniqueid": "a4:c1:38:b5:35:63:9f:70-01-0500"
    }
}
                    
                `

Rocket200, 2023-08-22:
> > No Button map come all the time
> 
> So at every vibration or even without vibration ?

Yes on INFO_L2 (Debug Info) 
`22:01:24:049 [INFO] - No button map for: TS0210, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 010400640000, zclSeq: 136`

Rocket200, 2023-08-23:
`21:21:50:618 0xA4C138B535639F70: update ZCL value 0x01/0x0500/0x0002 after 0 s
21:21:50:618 [INFO] - No button map for: TS0210, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 010400640000, zclSeq: 136
21:21:50:619 Websocket 192.168.0.166:44118 send message: {"e":"changed","id":"40","r":"sensors","state":{"lastupdated":"2023-08-23T19:21:50.618","lowbattery":false,"vibration":true},"t":"event","uniqueid":"a4:c1:38:b5:35:63:9f:70-01-0500"} (ret = 182)
21:21:50:619 Websocket 192.168.0.146:55487 send message: {"e":"changed","id":"40","r":"sensors","state":{"lastupdated":"2023-08-23T19:21:50.618","lowbattery":false,"vibration":true},"t":"event","uniqueid":"a4:c1:38:b5:35:63:9f:70-01-0500"} (ret = 182)
21:21:50:620 Websocket 192.168.0.146:55582 send message: {"e":"changed","id":"40","r":"sensors","state":{"lastupdated":"2023-08-23T19:21:50.618","lowbattery":false,"vibration":true},"t":"event","uniqueid":"a4:c1:38:b5:35:63:9f:70-01-0500"} (ret = 182)
21:21:50:896 APS-DATA.indication from child 0x633F`

Rocket200, 2023-08-23:
i add "config/tuya_unlock" in the DDF File but nothing changed ...
or i make something wrong ..
my Aqara Vibration work normal

Rocket200, 2023-08-25:
maybee this help ? 
https://community.smartthings.com/t/edge-driver-mc-zigbee-drivers-for-motion-open-close-moisture-smoke-co-sensors-and-others-devices/231689/3430?page=167

or try other ? Z2M ?

Smanar, 2023-08-25:
And not better if your change config/sensitivity ? You need to awake the device for setting change happen faster.

Rocket200, 2023-08-25:
I dont Know how to Change 
i can Change a DDF in a Folder and remove or add the Device

Rocket200, 2023-08-26:
I dont want make anything wrong, change sens. from to? 
or please a DDF I can test it

Smanar, 2023-08-26:
On the GUI.
Select the cluster 0x0500 IAS zone. It will open the cluster panel
Go on bottom, to see the attribute list, select the attribute 0x0013.
Awake the sensor (short press on button) and press Read.
Change the value and press Write when awaking the device.

Rocket200, 2023-08-26:
Vibration 42: logLevel: set to 2
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Name: set to "Vibration 42"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Serial Number: set to "A4C138B535639F70"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Manufacturer: set to "_TZ3000_fkxmyics"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Model: set to "TS0210"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Firmware Revision: set to "0.0.0"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: _TZ3000_fkxmyics TS0210 v0.0.0 (1 resources)
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Name: set to "Vibration 42"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Configured Name: set to "Vibration 42"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Motion Detected: set to false
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Sensitivity: set to 7
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Enabled: set to true
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Status Active: set to true
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42: Status Fault: set to 0
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42 Battery: Name: set to "Vibration 42 Battery"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42 Battery: Configured Name: set to "Vibration 42 Battery"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42 Battery: Battery Level: set to 0%
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42 Battery: Status Low Battery: set to 0
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42 Battery: lowBatteryThreshold: set to 20%
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42 History: Name: set to "Vibration 42 History"
[8/26/2023, 8:44:18 AM] [deCONZ] Vibration 42 History: Configured Name: set to "Vibration 42 History"
[8/26/2023, 8:44:18 AM] [deCONZ] Phoscon-GW: warning: recommended version: deCONZ v2.22.2
[8/26/2023, 8:44:18 AM] [deCONZ] Phoscon-GW: migration: /resourcelinks/1: 26 resources
[8/26/2023, 8:44:19 AM] [deCONZ] Vibration 42: Sensitivity: set to 4 (from 7)
[8/26/2023, 8:44:29 AM] [deCONZ] Vibration 42: Motion Detected: set to true (from false)
[8/26/2023, 8:44:33 AM] [deCONZ] Vibration 42: Sensitivity: set to 7 (from 4)

i change it to valvue 2 
and the log above is from Homebridge 
the Device is always on "Vibration detected"

Rocket200, 2023-08-26:
with Zone Sensitivity Level 5 looks better , 

in Phoscon it looks better now , but always Vibration detected in Homebridge 

when i Push to Button on the Device it changes back to false from true . but why ?

Rocket200, 2023-08-26:
Vibration 28 = Aqara  ( 1 Minute from frue to false )
[8/26/2023, 9:40:24 AM] [deCONZ] Vibration 38: Motion Detected: set to true (from false)
[8/26/2023, 9:41:29 AM] [deCONZ] Vibration 38: Motion Detected: set to false (from true)

Vibration 38 = _TZ3000_fkxmyics 
[8/26/2023, 9:40:16 AM] [deCONZ] Vibration 44: Motion Detected: set to true (from false)
after 5 Minute nothing happend , not change back to false

Rocket200, 2023-08-26:
when i push the Button ( Debug Info ) 

10:48:18:651 a4:c1:38:b5:35:63:9f:70-01-0500/config/battery expression: Item.val = Attr.val / 2 --> 100
10:48:18:651 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:27:994 a4:c1:38:b5:35:63:9f:70-01-0500/config/battery expression: Item.val = Attr.val / 2 --> 100
10:48:27:995 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:30:735 a4:c1:38:b5:35:63:9f:70-01-0500/config/battery expression: Item.val = Attr.val / 2 --> 100
10:48:30:735 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:30:976 00:15:8d:00:03:89:45:e3-01-0b04/state/voltage expression: if (Attr.val != 65535) { Item.val = Attr.val; } --> 233
10:48:30:977 ZCL attribute report 0x00158D00038945E3 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:32:213 00:15:8d:00:03:89:26:03-01/state/on expression: Item.val = Attr.val --> false
10:48:32:213 ZCL attribute report 0x00158D0003892603 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:37:751 a4:c1:38:b5:35:63:9f:70-01-0500/config/battery expression: Item.val = Attr.val / 2 --> 100
10:48:37:752 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:43:330 a4:c1:38:b5:35:63:9f:70-01-0500/config/battery expression: Item.val = Attr.val / 2 --> 100
10:48:43:330 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:44:974 00:15:8d:00:03:89:45:e3-01-0b04/state/voltage expression: if (Attr.val != 65535) { Item.val = Attr.val; } --> 231
10:48:44:975 ZCL attribute report 0x00158D00038945E3 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:45:210 a4:c1:38:b5:35:63:9f:70-01-0500/config/battery expression: Item.val = Attr.val / 2 --> 100
10:48:45:210 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:48:48:998 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:49:01:190 ZCL attribute report 0x00158D0003892603 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:49:08:331 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:49:10:345 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:49:11:969 ZCL attribute report 0xA4C138B535639F70 for cluster: 0x0001, ep: 0x01, frame control: 0x18, mfcode: 0x0000

Rocket200, 2023-08-26:
10:56:38:059 0xA4C138B535639F70: update ZCL value 0x01/0x0500/0x0002 after 0 s
10:56:38:060 [INFO] - No button map for: TS0210, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 010800640000, zclSeq: 238
10:56:38:061 Websocket 192.168.0.146:51822 send message: {"e":"changed","id":"44","r":"sensors","state":{"lastupdated":"2023-08-26T08:56:38.060","lowbattery":false,"vibration":true},"t":"event","uniqueid":"a4:c1:38:b5:35:63:9f:70-01-0500"} (ret = 182)
10:56:38:062 Websocket 192.168.0.166:37030 send message: {"e":"changed","id":"44","r":"sensors","state":{"lastupdated":"2023-08-26T08:56:38.060","lowbattery":false,"vibration":true},"t":"event","uniqueid":"a4:c1:38:b5:35:63:9f:70-01-0500"} (ret = 182)

Rocket200, 2023-08-29:
this device i get today , here all normal ,but  new one from i start Device req. need new config .... 

**_TZ3000_bmfw9ykl**
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: logLevel: set to 2
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Name: set to "Presence 33"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Serial Number: set to "A4C1382B3ED579AC"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Manufacturer: set to "_TZ3000_bmfw9ykl"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Model: set to "TS0210"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Firmware Revision: set to "0.0.0"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: _TZ3000_bmfw9ykl TS0210 v0.0.0 (1 resources)
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Name: set to "Presence 33"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Configured Name: set to "Presence 33"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Motion Detected: set to false
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Duration: set to 5s
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Enabled: set to true
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Status Active: set to true
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33: Status Fault: set to 0
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33 Battery: Name: set to "Presence 33 Battery"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33 Battery: Configured Name: set to "Presence 33 Battery"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33 Battery: Battery Level: set to 0%
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33 Battery: Status Low Battery: set to 0
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33 Battery: lowBatteryThreshold: set to 20%
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33 History: Name: set to "Presence 33 History"
[8/29/2023, 1:56:12 PM] [deCONZ] Presence 33 History: Configured Name: set to "Presence 33 History"

Smanar, 2023-08-29:
>after 5 Minute nothing happend , not change back to false

If I m right, it's normal, vibration sensor doesn't have true/false state.
They make a new notification at every detection, but only if you have a vibration.

Rocket200, 2023-08-29:
> If I m right, it's normal, vibration sensor doesn't have true/false state.
> They make a new notification at every detection, but only if you have a vibration.

the _TZ3000_bmfw9ykl  change back to false after 1 sec. ( "same" Vibration Sensor ) here work all fine see 2 Post bevor
[_TZ3000_bmfw9ykl_vibration_sensor.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/12469309/_TZ3000_bmfw9ykl_vibration_sensor.zip)

Smanar, 2023-08-30:
So it's good for you ?

Rocket200, 2023-08-30:
The req. one not work , not solve the problem for deconz or other user get this one…

but for me i have a Solution the new Device „MOES“ with Same DDF working perfect!

closed 2023-09-28 by github-actions[bot]

Rocket200, 2023-09-28:
Problem not Solved , but without help i can do nothing !

Smanar, 2023-09-28:
>the _TZ3000_bmfw9ykl change back to false after 1 sec

This can be normal, what is your problem ?