dresden-elektronik/deconz-rest-plugin #7194: ROVF (TMZ02L-16A-B2) Mini Switch Module 16A No Neutral: _TZ3000_hkkkuk7k TS0012

str31ok, 2023-08-21:
## Device

- Product name: ROVF (TMZ02L-16A-B2)
- Manufacturer: _TZ3000_hkkkuk7k
- Model identifier: TS0012
- Device type :  
  - Switch

## Screenshots

[image]
### Basic
[image]
### Identify
[image]
### Groups
[image]
[image]
### Scenes
[image]
[image]
[image]
[image]
### On/Off
[image]
[image]
[image]
### Unknown
[image]
### Tuya specific
[image]
### OTAU
[image]
### Time
[image]
### On/Off
[image]
[image]

str31ok, 2023-08-22:
It seems to me that it works.  At least the relay appeared in phoscon and i could add it in openhab. Thank you very much for your help

BabaIsYou, 2023-08-23:
My pleasure 😇

PR #7196 (2023-08-31): DDF add support for ROVF (TMZ02L-16A-B2) Mini Switch
Close #7194
closed 2023-08-31 by manup