dresden-elektronik/deconz-rest-plugin #7197: LIVARNO home Automatik-Verdunkelungsrollo

chbndrhnns, 2023-08-23:
## Device

- Product name: LIVARNO home Automatik-Verdunkelungsrollo
- Manufacturer: _TZB000_42ha4src
- Model identifier: TS030F
- Device type : Please remove all unrelated device types. 
  - Other:  Blinds

[image]

## Screenshots

### Basic

[image]

### Identify

[image]

### Alarms

### Device Temperature

### Groups

[image]

[image]

### Scenes

[image]

[image]

[image]

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

#### Window Covering

[image]

#### Power Configuration

[image]

#### OTAU

[image]

BabaIsYou, 2023-08-25:
Hi. Can you please just complete the Windows covering screenshot, some attributes are missing.

chbndrhnns, 2023-08-25:
@BabaIsYou I added the missing attributes.

BabaIsYou, 2023-09-02:
Seems that this device use standard ZCL command and not Tuya. Then could try this DDF 
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZB000_42ha4rsc",
  "modelid": "TS030F",
  "product": "Livarno blinds",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift"
        },
        {
          "name": "state/open"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0001",
          "0x0102"
        ]
      },
      "items": [
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "parse": {
            "cl": "0x0001",
            "at": "0x0021",
            "eval": "Item.val = Attr.val / 2"
          },
          "read": {
            "cl": "0x0001",
            "at": "0x0021"
          },
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x01"
        }
      ]
    }
  ]
}
```

chbndrhnns, 2023-09-02:
Just tried it and it seems to work fine at first glance! Thanks much!

chbndrhnns, 2023-09-02:
Would this DDF also be the right place to define a custom maximum limit value for the cover?

Smanar, 2023-09-02:
You haven't a calibration mode ?

chbndrhnns, 2023-09-06:
> You haven't a calibration mode ?

The manual doesn't mention a calibration mode, unfortunately. Am I out of luck now?

Smanar, 2023-09-06:
No, but there is so much possible working mode.
On your capture on the GUI, can you try on the cluster 0x0102 covering cluster, the attribute 

```
          <attribute id="0x0017" name="Mode" type="bmp8" default="0x0000" access="rw" required="o">
            <value name="Reversed" value="0"></value>
            <value name="Calibration Mode" value="1"></value>
            <value name="Maintenance Mode" value="2"></value>
            <value name="LED feedback" value="3"></value>
          </attribute>
```

or 

`          <attribute id="0xf003" name="Calibration time (tenth of a second)" type="u16" default="0x0000" required="m" access="rw"></attribute>`

For the second one, will be easy, you just need to change the value
For the first it will just enable the calibration mode, after you need to do"things" described in a manual, and it will probably update the value present in the second one.

chbndrhnns, 2023-09-17:
Thanks for the idea with the calibration channel, I will give it a try!

In home assistant, I see such a diagram for the battery state and I am wondering if the 100 % values are something that can be fixed using the DDF file, as well?

[image]

chbndrhnns, 2023-09-17:
Just for reference, here is the manual for the remote control with features that LIDL is not advertising:
- set min/max limit
- pair one remote control to more than one cover

 https://device.report/manual/4093622

Smanar, 2023-09-18:
Ha yes, so can use direclty the remote for calibration, usefull.

For the battery issue, IDK, what is the good value ? 100 or 30 ? Need to find something specific to reconize bad value and add a test in

`"eval": "Item.val = Attr.val / 2"`

There is a natural bind/report for it 

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x01"
        }
      ]
    }
```

And a pool (when bind don't work)

`"refresh.interval": 3600`

Perhaps one of then is not working ?

chbndrhnns, 2023-09-27:
It seems to me that `"refresh.interval": 3600` is not working: when using the RF remote, the position of the blinds never updates in Home Assistant, not even after a day.

Smanar, 2023-09-28:
`"refresh.interval": 3600` is just for the battery ?

You mean the position is updated when using HA but not when you use the local remote ?

state/open  and state/lift are not updated ?

chbndrhnns, 2023-09-30:
I can only observe updates in home assistant as the deconz UI does not show the covers.
So when I set the "lift" attribute and the covers moves, it seems the state does not update in home assistant (ever or in a timely fashion). It shows the last state from "2 days ago".

When subscribing to the websocket feed, I found these events related to the covering device.

Is it something I should bring up with the HA team, instead?

```
  {
    "attr": {
      "id": "32",
      "lastannounced": null,
      "lastseen": "2023-09-30T10:57Z",
      "manufacturername": "_TZB000_42ha4rsc",
      "modelid": "TS030F",
      "name": "Window covering device 32",
      "swversion": null,
      "type": "Window covering device",
      "uniqueid": "70:ac:08:ff:fe:3e:b4:3c-01"
    },
    "e": "changed",
    "id": "32",
    "r": "lights",
    "t": "event",
    "uniqueid": "70:ac:08:ff:fe:3e:b4:3c-01"
  },
  {
    "e": "changed",
    "id": "32",
    "r": "lights",
    "state": { "lift": 56, "open": false, "reachable": true },
    "t": "event",
    "uniqueid": "70:ac:08:ff:fe:3e:b4:3c-01"
  },
  {
    "e": "changed",
    "id": "32",
    "r": "lights",
    "state": { "lift": 47, "open": false, "reachable": true },
    "t": "event",
    "uniqueid": "70:ac:08:ff:fe:3e:b4:3c-01"
  }
```

Smanar, 2023-09-30:
The first one is a "attr" websocket, not usefull for the problem, but right the 2 others contain the good data.
`"state": { "lift": 56, "open": false, "reachable": true },`

"Lift" is the covering position, if you can see it in websocket event, HA can see it too.
Can be a "synchronisation" problem, if I m right there is a way to remove a device ONLY from HA, and then at next launch the device will be re-created (because still visible in deconz).

Are you sure having a recent version of the deconz plugin ? we was using on/bri previously for covering and now open/lift, the change is recent, long time we had support for both, but now on/bri is deprecated.

chbndrhnns, 2023-09-30:
I think I am on 2.22.0. Is this recent enough?

I cannot remove an entity in home Assistant which is provided by deconz:

> You can only remove 0 of the selected 1 entities. Entities can only be removed when the integration is no longer providing the entities. Sometimes you have to restart Home Assistant before you can remove the entities of a removed integration. Are you sure you want to remove the removable entities?

Smanar, 2023-09-30:
And you can't try to re-include the device ?

Kane610, 2023-09-30:
> @Kane610 there isn't a way to "reset" the device in the plugin HA database ?

There is very little static information kept in the integration. Not sure what you're asking but restarting the integration will reload all device information

chbndrhnns, 2023-09-30:
if it’s off-topic for deconz, where could be continue the conversation?

@Kane610 if I may jump in, I notice that home assistant does not seem to have an idea about the last state update of my covers. It always reports them as 2 days ago. Also, the battery state is strange, it is reported as 100 percent of the time. 

I’ll add two pictures here.

[image]
[image]

Kane610, 2023-09-30:
> if it’s off-topic for deconz, where could be continue the conversation?
>
> @Kane610 if I may jump in, I notice that home assistant does not seem to have an idea about the last state update of my [...]

All you see is as reported by deconz. There are no conversions or assumptions in the values. 

Best thing to do is to enable debug logging as detailed in the deConz integration documentation and see what data is sent from Deconz over websocket. Which is the typical cause of devices not updating in home assistant

chbndrhnns, 2023-09-30:
As for the battery value, deconz actually sends 100 % but the real value is 19 %. It gets sent as well, but most of the time, 100 % is sent.

```
[pydeconz.websocket] {"e":"changed","id":"59","r":"sensors","state":{"battery":100,"lastupdated":"2023-09-30T22:10:15.938"},"t":"event","uniqueid":"70:ac:08:ff:fe:3e:98:03-01-0001"}
```

Kane610, 2023-10-01:
> As for the last updated value in the UI, I just moved the cover position and it's reported correctly via an event so I suspect Home Assistant to be the problem here

And you saw that websocket event in hass debug logs? Can you share a full log?

Smanar, 2023-10-01:
```
  {
    "e": "changed",
    "id": "32",
    "r": "lights",
    "state": { "lift": 56, "open": false, "reachable": true },
    "t": "event",
    "uniqueid": "70:ac:08:ff:fe:3e:b4:3c-01"
  },
  {
    "e": "changed",
    "id": "32",
    "r": "lights",
    "state": { "lift": 47, "open": false, "reachable": true },
    "t": "event",
    "uniqueid": "70:ac:08:ff:fe:3e:b4:3c-01"
  }
```

IDK for HAss log, but he can see the websocket event in phoscon, so no reason for it is not in HA too ? I suspect more a problem in device creation, for exemple a device created with bri/on instead of open/lift, but IDK if the HA plugin is able to update that after a device creation ?

chbndrhnns, 2023-10-01:
I just used the RF remote to change the cover position. 
One event is created when the end position is reached or I stop the cover.
The UI in HA updates the position correctly but still but the timestamp in the UI stays at "12 hours ago" or something and updates much less often.

```
2023-10-01 10:13:23.505 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"32","r":"lights","state":{"lift":25,"open":false,"reachable":true},"t":"event","uniqueid":"70:ac:08:ff:fe:3e:b4:3c-01"}
```

chbndrhnns, 2023-10-01:
> For the battery issue, can be the device itself that send good and bad value, I don't see something special in DDF, can ignore the 100 value ? You will miss the 100% but will be fine after for 99% and other ? 

I am happy to not see 100 % ever if the values are reliable apart from the 100 % value.
I will try your two suggestions.

Kane610, 2023-10-01:
> IDK for HAss log, but he can see the websocket event in phoscon, so no reason for it is not in HA too ? I suspect more a problem in device creation, for exemple a device created with bri/on instead of open/lift, but IDK if the HA plugin is able to update that after a device creation ?

I can't debug possible integration issues without data directly from the integration, Phoscon does a lot of other things that differ so there is no truth for me in looking at any data not from the integration.

> The UI in HA updates the position correctly but still but the timestamp in the UI stays at "12 hours ago" or something and updates much less often.

That is nothing I have control over, it should reflect any state changes seen in the HA state machine. As you say state changes properly I don't understand why that value doesn't change, have you cleared browser cache and everything else to make sure it has latest versions active?

chbndrhnns, 2023-10-04:
> That is nothing I have control over, it should reflect any state changes seen in the HA state machine. As you say state changes properly I don't understand why that value doesn't change, have you cleared browser cache and everything else to make sure it has latest versions active?

I checked that I am running the latest frontend version.

The logbook also seems to be missing lots of events. For the last week it only contains one close event although the state changes multiple times a day:

[image]

I just trigger the cover and these are the related events that appear in the HA logs:

```
2023-10-04 10:27:59.725 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'open': False}" to "http://core-deconz:40850/api/56478788EA/lights/34/state"
2023-10-04 10:27:59.755 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/34/state/open': False}}]
2023-10-04 10:28:08.416 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'stop': True}" to "http://core-deconz:40850/api/56478788EA/lights/34/state"
2023-10-04 10:28:08.430 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/34/state/stop': True}}]
2023-10-04 10:28:09.236 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"34","lastannounced":null,"lastseen":"2023-10-04T08:28Z","manufacturername":"_TZB000_42ha4rsc","modelid":"TS030F","name":"Window covering device 34","swversion":null,"type":"Window covering device","uniqueid":"70:ac:08:ff:fe:3e:92:d2-01"},"e":"changed","id":"34","r":"lights","t":"event","uniqueid":"70:ac:08:ff:fe:3e:92:d2-01"}
2023-10-04 10:28:09.239 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"34","r":"lights","state":{"lift":29,"open":false,"reachable":true},"t":"event","uniqueid":"70:ac:08:ff:fe:3e:92:d2-01"}
2023-10-04 10:28:09.274 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"60","lastannounced":null,"lastseen":"2023-10-04T08:28Z","manufacturername":"_TZB000_42ha4rsc","modelid":"TS030F","name":"Battery 60","swversion":null,"type":"ZHABattery","uniqueid":"70:ac:08:ff:fe:3e:92:d2-01-0001"},"e":"changed","id":"60","r":"sensors","t":"event","uniqueid":"70:ac:08:ff:fe:3e:92:d2-01-0001"}
```

However, this does not end up in the logbook or the "last updated" text

Smanar, 2023-10-04:
>However, this does not end up in the logbook or the "last updated" text

Ha yes, I have take a look on my side, it seem "light" haven't state/lastupdated

`2023-10-04 10:28:09.239 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"34","r":"lights","state":{"lift":29,"open":false,"reachable":true},"t":"event","uniqueid":"70:ac:08:ff:fe:3e:92:d2-01"}`

And the lift value need to be updated at 29.

chbndrhnns, 2023-10-05:
> Ha yes, I have take a look on my side, it seem "light" haven't state/lastupdated

@Smanar Thanks! Does it mean I need to update the DDF to make it work?

Smanar, 2023-10-05:
No this is normal, "light" device haven't state/lastupdated.

About your issue, I don't see why HA don't manage the websocket return. For me something bad happen when you have included the device the fist time, before using the DDF.
If you delete the device and re-include it, the HA plugin will perhaps "reset" his data for this device.

chbndrhnns, 2023-10-27:
I'd like to keep it open until the DDF that was posted above becomes part of deconz.

Smanar, 2023-10-27:
Oups, some one have made a PR ?
@chbndrhnns do you want to make it, or want someone make it ?

chbndrhnns, 2023-10-30:
@Smanar I added one: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7325
Can you help with the details if necessary?

Smanar, 2023-10-31:
So this device with defaut value

```
        {
          "name": "state/lift"
        },
        {
          "name": "state/open"
        },
```

We don't need to reverse one ?

If you can just add on your first post

> 
> -   Product name: LIVARNO home Automatik-Verdunkelungsrollo
> -    Manufacturer: _TZB000_42ha4rsc
> -    Model identifier: TS030F
> -    Device type : Please remove all unrelated device types.
>         Other: Blinds

Else all is fine, thx.

chbndrhnns, 2023-10-31:
I see a position of 0 in home assistant when the cover is down. Is this correct?

Smanar, 2023-11-01:
He nope ^^

**shutter closed**
(on = true)
(bri = 255)
open = false
lift = 100%

Open is reversed too ?
if you send a lift=0 it open or close it ?

chbndrhnns, 2023-11-02:
I just tested it: When I send `{"lift": 0}`, it opens the cover.

Smanar, 2023-11-02:
Ok so nice, was the harder part, so the command works fine, and the return ?
You have send `{"lift": 0}` so after the mouvement, you need to have same on the device json, else need to reverse the return (it's easy)

Kane610, 2023-11-02:
> I see a position of 0 in home assistant when the cover is down. Is this correct?

Home assistant has inverted values compared to Deconz "100 - self._device.lift"

chbndrhnns, 2023-11-03:
When I send `{"lift": 0}`, I get this response:

```
[
  {
    "success": {
      "/lights/34/state/lift": 0
    }
  }
]
```

After the movement, this is the status of the device:

```
{
  "config": {
    "groups": []
  },
  "etag": "0f76b1a60fefb02c1bcc4741bc6f8c03",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2023-11-03T08:45Z",
  "manufacturername": "_TZB000_42ha4rsc",
  "modelid": "TS030F",
  "name": "Window covering device 34",
  "state": {
    "lift": 0,
    "open": false,
    "reachable": true
  },
  "swversion": null,
  "type": "Window covering device",
  "uniqueid": "70:ac:08:ff:fe:3e:92:d2-01"
}
```

Smanar, 2023-11-03:
It seem fine, the reverse situation is probably due to HA convention (from @Kane610  comment)
So your DDF is fine.

Just add on your first post

>  -       Product name: LIVARNO home Automatik-Verdunkelungsrollo
>  -       Manufacturer: _TZB000_42ha4rsc
>  -       Model identifier: TS030F
>  -       Device type : Please remove all unrelated device types.
>         Other: Blinds

And for me all is fine.

chbndrhnns, 2023-11-04:
>Pls just add on your first post

That's already in the first post, right?

Smanar, 2023-11-04:
On the first post on your PR ^^.
For the search engine.

Smanar, 2023-11-26:
PR not yet valided, in wiating list.

chbndrhnns, 2023-12-18:
What needs to happen to merge the related PR?

Smanar, 2023-12-18:
Good question, all seem fine for me
Swoop have added the flag "Device request" 1 week ago.

Can depend of the next plugin version, beta or stable.

chbndrhnns, 2024-01-10:
Another post to keep the issue alive...

closed 2024-01-10 by Mimiix

Mimiix, 2024-01-10:
> Another post to keep the issue alive...

Its merged :)