dresden-elektronik/deconz-rest-plugin #7198: Sunricher SR-ZG9092A Thermostat

Vies123, 2023-08-25:
## Device

- Product name: Sunricher SR-ZG9092A
- Manufacturer: Sunricher
- Model identifier: ZG9092
- Device type : 
  - Thermostat

## Screenshots
[image]
[image]

### Basic
[image]
### Identify
[image]
### Time
[image]
### Thermostat
[image]
[image]
[image]
[image]
### Thermostat user interface configuration
[image]
### Simple Metering
[image]
### Electrical measurements
[image]
[image]
[image]

BabaIsYou, 2023-08-25:
Hi. Seems that some cluster’s screenshots are missing  ?

BabaIsYou, 2023-08-25:
Read somewhere that it could be the same as behavior as Namron 4512737/4512738 model.

could begin with this DDF then (adapted from already supported Namron), will see to add perhaps other items later (don’t forget to complete screenshots)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Sunricher",
  "modelid": "ZG9092",
  "product": "Thermostat SR-ZG9092A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "default": 0
        },
        {
          "name": "config/mode",
         "default": "off"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/temperature",
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "description": "The measured current (in A).",
          "refresh.interval": 300,
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val / 1000; }"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "default": 0
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300,
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 0,
            "eval": "if (Attr.val != 65535) { Item.val = Attr.val * 100 ; }"
          },
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
    "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0201",
     "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 300,
          "max": 3600,
          "change": "0x00000032"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

closed 2023-09-24 by github-actions[bot]