dresden-elektronik/deconz-rest-plugin #7202: Legrand / BTicino Master remote SW Home / Away

CAgnesi, 2023-08-28:
## Device

- Product name: BTicino Living Now K4570CW
- Manufacturer: Legrand
- Model identifier: Master remote SW Home / Away
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch

## Screenshots

### Basic
[image]

### Identify

[image]

CAgnesi, 2023-08-28:
Hi Smanar! This one I bought by itself. From a previous post of yours I understood the one that is bundled with the gateway is very hard (if not impossible) to get working, right?

Mimiix, 2023-08-28:
Hi. Please add all requested screenshots to the first post.

Mimiix, 2023-08-28:
Looks good, thanks

Smanar, 2023-08-29:
>Hi Smanar! This one I bought by itself. From a previous post of yours I understood the one that is bundled with the gateway is very hard (if not impossible) to get working, right?

No, dangerous ^^, because you can make all the Legrand Gateway out of order.

Can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Legrand",
  "modelid": "Master remote SW Home / Away",
  "product": "Master remote SW Home / Away",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x000f"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

It will not work, will be normal. But can you check in deconz / help / debug view with flag "info_l2", the error message when you use it ? Something about a Buttonmap.

CAgnesi, 2023-08-29:
Thanks Smanar! I am now able to see the switch in the Phoscon App! Howere I do get something in the debug view: 
`22:40:02:014 DB save zll database items 0x00000081
22:40:02:017 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('3', 'Master remote SW Home / Away', 'ZHASwitch', 'Master remote SW Home / Away', 'Legrand', '00:04:74:00:01:3b:1b:95-01-000f', '003f', '{"buttonevent":null,"lastupdated":null}', '{"battery":0,"on":true,"reachable":true}', '', 'normal', '1', '2023-08-29T20:39Z', '2023-08-29T20:39:01Z')
22:40:02:034 DB saved in 19 ms
22:40:02:124 Websocket 192.168.1.123:49804 send message: {"e":"scene-called","gid":"65527","r":"scenes","scid":"1","t":"event"} (ret = 70)
22:40:02:125 Websocket 192.168.1.123:52956 send message: {"e":"scene-called","gid":"65527","r":"scenes","scid":"1","t":"event"} (ret = 70)
22:40:02:126 [INFO] - No button map for: Master remote SW Home / Away, broadcast to: 0xFFF7, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: F7FF010000, zclSeq: 3`

CAgnesi, 2023-08-30:
Hi Smanar! Unfortunately it doesn't seem to work...
Now i get the following messages in debug mode:
For the **Home** button
`23:43:20:086 Websocket 192.168.1.123:37878 send message: {"e":"scene-called","gid":"65527","r":"scenes","scid":"1","t":"event"} (ret = 70)
23:43:20:087 [INFO] - No button map for: Master remote SW Home / Away, broadcast to: 0xFFF7, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: F7FF010000, zclSeq: 0
23:43:20:088 Websocket 192.168.1.123:37878 send message: {"e":"changed","id":"4","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2023-08-30T21:43:20.085"},"t":"event","uniqueid":"00:04:74:00:01:3b:1b:95-ff-000f"} (ret = 164)
23:43:20:409 Websocket 192.168.1.123:37878 send message: {"attr":{"id":"4","lastannounced":"2023-08-30T21:43:20Z","lastseen":"2023-08-30T21:43Z","manufacturername":"Legrand","modelid":"Master remote SW Home / Away","name":"Master remote SW Home / Away","swversion":null,"type":"ZHASwitch","uniqueid":"00:04:74:00:01:3b:1b:95-ff-000f"},"e":"changed","id":"4","r":"sensors","t":"event","uniqueid":"00:04:74:00:01:3b:1b:95-ff-000f"} (ret = 372)`
For the **Away** button
`23:39:19:227 APS-DATA.indication from child 0xBD9B
23:39:19:228 Websocket 192.168.1.123:37878 send message: {"e":"scene-called","gid":"65526","r":"scenes","scid":"1","t":"event"} (ret = 70)
23:39:19:229 Websocket 192.168.1.123:58006 send message: {"e":"scene-called","gid":"65526","r":"scenes","scid":"1","t":"event"} (ret = 70)
23:39:19:230 [INFO] - No button map for: Master remote SW Home / Away, broadcast to: 0xFFF6, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: F6FF010000, zclSeq: 2
23:39:19:231 Websocket 192.168.1.123:37878 send message: {"e":"changed","id":"4","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2023-08-30T21:39:19.228"},"t":"event","uniqueid":"00:04:74:00:01:3b:1b:95-ff-000f"} (ret = 164)
23:39:19:232 Websocket 192.168.1.123:58006 send message: {"e":"changed","id":"4","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2023-08-30T21:39:19.228"},"t":"event","uniqueid":"00:04:74:00:01:3b:1b:95-ff-000f"} (ret = 164)`
(You were right about the payload values though).
I should also tell you that there is a Scenes cluster for the devices. However it is grey compared to the other clusters that are blue... Please see the attached screenshot.
[image]

Smanar, 2023-08-31:
>I should also tell you that there is a Scenes cluster for the devices. However it is grey compared to the other clusters that are blue... 

Yep, I have see it after, blue are input cluster, grey are output.

>Hi Smanar! Unfortunately it doesn't seem to work...

It work

`"state":{"buttonevent":1002,`

`{"buttonevent":2002,"`

You haven't this value updated ? You can see it for exemple in phoscon/help/API information/event

CAgnesi, 2023-09-01:
Dear Smanar, This is what I get in Phoscon app/ help / API information / REST API: 
`"scenes": {},
    "schedules": {},
    "sensors": { ...
        "3": {
            "config": {
                "on": true,
                "reachable": true
            },
            "etag": "01122ca2e24679979711dbd2f26889ee",
            "lastannounced": "2023-09-01T06:16:45Z",
            "lastseen": "2023-09-01T06:17Z",
            "manufacturername": "Legrand",
            "mode": 1,
            "modelid": "Master remote SW Home / Away",
            "name": "Master remote SW Home / Away",
            "state": {
                "buttonevent": null,
                "lastupdated": "none"
            },
            "swversion": "003f",
            "type": "ZHASwitch",
            "uniqueid": "00:04:74:00:01:3b:1b:95-ff-000f"
        }
    }`
Instead in phoscon/help/API information/event I do not see any entries for the "Master remote SW Home / Away" (nor for switches in general).
[image]

Smanar, 2023-09-01:
Ha ? the buttonevent is missing ....
Strange because on log we can see it in the request

192.168.1.123:37878 send message: {"e":"changed","id":"4","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2023-08-30T21:43:20.085"},"t":"event","uniqueid":"00:04:74:00:01:3b:1b:95-ff-000f"} (ret = 164)

23:39:19:232 Websocket 192.168.1.123:58006 send message: {"e":"changed","id":"4","r":"sensors","state":{"buttonevent":2002,"lastupdated":"2023-08-30T21:39:19.228"},"t":"event","uniqueid":"00:04:74:00:01:3b:1b:95-ff-000f"} (ret = 164)

And why it's not the same device id ? (4 on previous logs and 3 on last ones)

CAgnesi, 2023-09-01:
Sorry, it is my fault, I had to reset the device and re-install it and did not update the DDF with the current version. Is there a way of making this DDF the default?
It seems to work more or less. Now I can see the button pressed event being triggered in Phoscon and also in HomeAssistant.
However I still cannot create automatons using the switch as an input... (the last step before total success)

Smanar, 2023-09-02:
>Is there a way of making this DDF the default?

I m not sure to understand, you mean waiting for deconz add it officially ?

>Minor observation: the battery level also doesn't report correct values

It show 0 ?
For that we have a problem, on other Legrand device we are reading battery state everytime you press the button, but IDK how to do that using DDF.
And if you add a 24h refresh on the DDF ?

```
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "refresh.interval": 86400,
          "default": 0
        },
```

CAgnesi, 2023-09-02:
I'll give you a bit of context on my system, maybe you could give me some advice. I have a Raspberry Pi with Home Assistant Core installed together with the RaspBee II. To test the DDF I have to close deCONZ.service via terminal and open up the deCONZ app, and load the DDF via the editor. However having the deCONZ app makes Home Assistant incapable of controlling the ZigBee devices. I have to close the deCONZ app but this reverts the DDF back to default, loosing the Button Events... Should I recompile deCONZ with the DDF added? I already did something similar to get percentage control on Home Assistant for the Bticino Living Now rolling shutter, as you Smanar proposed in another thread I can't find. Thanks!

Smanar, 2023-09-02:
No need to recompile code for the moment, DDF are just text file.
You have deconz on a dedicated docker (the officila one ), or using the HA integration ?
For the second one, you need to put the DDF in `/data/.local/share/dresden-elektronik/deCONZ/devices/ `

https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

CAgnesi, 2023-09-14:
Apart from the battery status that is not properly reported, the DDF works perfectly fine and has allowed me to integrate the remote switch to Home Assistant. Thank you Smanar for your help.

Mimiix, 2023-09-15:
Hi,

Please keep this open so the DDF can be merged in the codebase for others to use :).

Can any of you please make a PR?

Smanar, 2023-09-15:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7228

closed 2023-09-24 by manup