dresden-elektronik/deconz-rest-plugin #7223: ZG-205ZL Human Presence and Illuminance Sensor with Siren Alarm

robidog, 2023-09-13:
## Device

- Product name: Tuya ZG-205ZL Human Presence and Illuminance Sensor with Siren Alarm
- Manufacturer: _TZE200_hl0ss9oa
- Model identifier: TS0025
- Device type: 
  - Presence Sensor
  - Light Sensor
  - Siren

(Zigbee2MQTT work-in-progress: https://github.com/Koenkk/zigbee2mqtt/issues/18497)

https://zigbee.blakadder.com/Tuya_ZG-205ZL.html

[image]

## Screenshots

### Basic
[image]
[image]
[image]

### Identify
[image]

### IAS Zone
[image]
[image]

### Tuya Specific (1)
[image]

### Tuya Specific (2)
[image]

### Illuminance Measurement
[image]

### Node Info
[image]

robidog, 2023-09-19:
Sorry there was a wrong screenshot in device request for Identify cluster. Fixed now.

robidog, 2023-10-11:
Bump. I noticed that with the device joined to the network, there were issues with Philips dimmer switches being disconnected all of a sudden and certain lag with other operations. Is it possible this sensor is overly chatty out out-of-the box and flooding the network?

ebaauw, 2023-10-11:
You didn’t include the screenshot of the node with endpoints and clusters, but from the _Node Descriptor_, this device is a Zigbee router (yellow node).   Technically, it’s very possible that it screws up your Zigbee network, not so much by being overly chatty, but by not correctly forwarding messages to other routers and/or by not correctly handling end devices that (try to) use it as parent.

Confirming this might be challenging; ideally you’d sniff the Zigbee traffic going into and out of this router.  You might check its neighbour table (or the lines in the deCONZ GUI) to check whether the Hue dimmer switches use it as parent, and the coordinator’s neighbour table to check whether the sensor is in it.  You might also check the deCONZ log for Zigbee errors.

Years ago, I had two innr bulbs close to my coordinator, that screwed up my network.  The only remedy was to remove these bulbs from the network.

robidog, 2023-10-11:
Thanks for the info. I will follow your advice and provide more data if possible.

closed 2023-11-09 by github-actions[bot]

BabaIsYou, 2023-11-29:
Could try with this DDF. Doesn,t cover all the stuff this device can do but it's a beginning ;-)

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZE200_hl0ss9oa",
    "modelid": "TS0225",
    "product": "Tuya ZG-205ZL Human presence sensor",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_LIGHT_LEVEL_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0400"
            ],
            "items": [
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "refresh.interval": 86400,
                    "read": {
                        "at": "0x0001",
                        "cl": "0x0000",
                        "ep": 1,
                        "fn": "zcl:attr"
                    },
                    "parse": {
                        "at": "0x0001",
                        "cl": "0x0000",
                        "ep": 1,
                        "fn": "zcl:attr",
                        "script": "tuya_swversion.js"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/reachable"
                },
                {
                    "name": "state/lastupdated"
                },
                {
                    "name": "state/dark"
                },
                {
                    "name": "config/tholddark"
                },
[... 286 lines cut ...]
                },
                {
                    "name": "state/presence",
                    "read": {
                        "fn": "tuya"
                    },
                    "parse": {
                        "dpid": 1,
                        "eval": "Item.val = Attr.val;",
                        "fn": "tuya"
                    },
                    "default": false
                }
            ]
        }
    ],
    "bindings": [
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0xEF00"
        }
    ]
}
```

robidog, 2023-12-03:
Awesome, thanks! Will test this and provide feedback as soon as possible. I just re-joined the device using your DDF and hopefully can get more info about it.