dresden-elektronik/deconz-rest-plugin #7225: [Device Request] Elko Smart ZB Thermostat 16 A

gladhorn, 2023-09-14:
## Device

- Product name: Smart ZB Thermostat 16 A
- Manufacturer: Schneider Electric
- Model identifier: EKO07259
- Device type:
  - Thermostat

There is some general information here:
https://proff.elko.no/elko-smart/ELKO%20-%20Smart%20ZB%20Thermostat%2016%20A_Device%20User%20Guide/English/ELKO%20-%20Smart%20ZB%20Thermostat%2016%20A_Device%20User%20Guide_0000930613.html

## Screenshots

Node Info:
[image]

[image]

## First Device: Thermostat

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups
[image]
[image]

### Thermostat
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Thermostat User Interface Configuration Cluster
[image]

### Diagnostics
[image]

### OTAU
[image]

## Second Device: Temperature Sensor
[image]

### Identify
[image]

### Temperature Measurement
[image]

## Third Device
### Basic
[image]

### Identify
[image]

### Temperature Measurement
[image]
(presumably not connected in my case)

## Fourth Device (05) Consumption Awareness Device
[image]

### Identify
[image]

### Groups
[image]
[image]

### Simple Metering
[image]

## Green Power
doesn't give any information

gladhorn, 2023-09-14:
I'm happy to experiment/implement things if that helps. Need some pointers though, this seems rather complex.

Smanar, 2023-09-16:
There is so much stuff on this device, can try this DDF to start

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "EKO07259",
  "product": "Elko Smart ZB Thermostat 16 A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "default": 0
        },
        {
          "name": "config/unoccupiedheatsetpoint",
          "default": 0
        },
        {
          "name": "config/heatsetpoint",
          "default": 0
        },
        {
          "name": "config/mode"
[... 244 lines cut ...]
          "max": 300,
          "change": "0x00000032"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 5,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

I haven't set all bind/report because on lot of thermostat this is native, have set only the temperature heatpoint and the local temperature.
You will have 3 temperatures, 2 Temperature sensor + the thermostat one, not sure all will be usefull ?

gladhorn, 2023-09-17:
Great start, the devices show up and I seem to get reasonable temperature readings 💯 I need to look at all the details in the next days as time permits, but this is looking very promising 🥇

Smanar, 2023-09-18:
For Information I will be afk for a week, but I think another devs will come help you if needed.

gladhorn, 2023-09-20:
Setting the heat setpoint seems to work, which is nice.

I'm using home assistant and I am still unsure about some points:
All thermostats always claim to be "idle", even when the physical display indicates heating. I'm not sure where the heat on/off state would come from though. In the manufacturer manual, it's clearly shown as on/off.

auto/heat/off - in home assistant by default new thermostats are shown with all of these greyed out. Once I set one of them, I cannot go back to "none of these selected". Not sure whether that's a home assistant thing and whether it makes any difference on the device.

The device supports different setups, which type of temperature sensor is used. This would be nice to have, although it shouldn't change usually:
https://proff.elko.no/elko-smart/ELKO%20-%20Smart%20ZB%20Thermostat%2016%20A_Device%20User%20Guide/English/ELKO%20-%20Smart%20ZB%20Thermostat%2016%20A_Device%20User%20Guide_0000930613_files/Managingdevicesettings_ELKO_CFMT_ZBCPT_0000993194.html#ManagingDeviceSettings-A81F4319
(I'm mostly wondering whether my thermostats are configured correctly)

Smanar, 2023-09-25:
> All thermostats always claim to be "idle", even when the physical display indicates heating

Can you show the device json to guess where HA is reading this information ? (you can have it in phoscon/help/API Information/sensor)
You are not speaking about the auto/heat/off mode ?

>auto/heat/off - in home assistant by default new thermostats are shown with all of these greyed out. Once I set one of them, I cannot go back to "none of these selected". Not sure whether that's a home assistant thing and whether it makes any difference on the device.

The value is probably "unknow" at start, but one time you have selected one, you are forced to be one of the tree one, so for me it's normal.

>The device supports different setups, which type of temperature sensor is used. This would be nice to have, although it shouldn't change usually:

Sure there is for exemple "config/externalsensortemp".
Do you have more information ? I can see 2 temperatures sensor on the device, there are both used ? 1 for air and the second for floor ?
I think we can do something with the attribute 

```
          <attribute id="0x001a" name="Remote Sensing" type="bmp8" default="0" access="rw" required="o">
            <value name="Local temperature sensed remotely" value="0"></value>
            <value name="Outdoor temperature sensed remotely" value="1"></value>
            <value name="Occupancy sensed remotely" value="2"></value>
          </attribute>
```

For the childlock you can add to the DDF (on thermostat sensor)

```
        {
          "name": "config/locked",
          "refresh.interval": 3660
        },
```

Need too add in bind
```
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0204",
      "report": [
        {
          "at": "0x0001",
          "dt": "0x30",
          "min": 1,
          "max": 3600
        }
      ]
    },
```

gladhorn, 2023-09-25:
One small update, I just tried using deCONZ with VNC and poked around some more.

The value I'm currently most interested in is:
0201 Thermostat Cluster, 0x0008 PI Heating Demand. That is an integer value between 0 and 100, returning how much the heater is on. When I set a high target temperature, it returns 100, when I am setting something very low, it's 0 and when target temperature is close to actual, it will return something like 50 it seems.
From what I can tell, this corresponds to how much the thermostat is heating.

Smanar, 2023-09-25:
Yep, it's the attribute we are using for state/on with 

```
        {
          "name": "state/on",
          "parse": {
            "at": "0x0008",
            "cl": "0x0201",
            "ep": 1,
            "eval": "Item.val = Attr.val > 3;",
            "fn": "zcl"
          },
          "read": {
            "fn": "none"
          }
        },
```

It return true if the value is > 3, and can be your "idle" issue, it's for that I m asking.

And need to add it in the bind table too

```
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        },
```

I can remake a DDF if you want, just tell me the attribute you want visibles.
But better to try first wthout bind table, I can't explain why, but lot of thermostat have some attribute with native report. If the bind/report not work, the value will move just if you read it in deconz, if the value is updated automatically mean you don't need to set in the bind table (take care if the bind is done, it's permanent, removing it from the DDF will not have a big impact)

gladhorn, 2023-09-28:
Might actually be nice with a complete DDF. I didn't get the idle issue working. In the original ddf on top, state/on wasn't handled specifically from what I can tell. I think home assistant uses something called hvac_action by default, but I have no idea how things are connected and whether on would be used as fallback. Even when I change the update frequency to 1 second and set the heating to a super hot temperature and PI_heating_demand returns 100, I still don't ever see a thermostat as anything but "idle" in home assistant. It would be nice to visualise how much the floor heating was active over time.
On the other hand, the DDF is already great in that I can actually set the desired temperature and read the sensor, which is of course the most important thing.

Smanar, 2023-09-28:
Have found that on HA code

```
    @property
    def hvac_action(self) -> HVACAction:
        """Return current hvac operation ie. heat, cool.

        Preset 'BOOST' is interpreted as 'state_on'.
        """
        if self._device.mode == ThermostatMode.OFF:
            return HVACAction.OFF

        if self._device.state_on or self._device.preset == ThermostatPreset.BOOST:
            if self._device.mode == ThermostatMode.COOL:
                return HVACAction.COOLING
            return HVACAction.HEATING
        return HVACAction.IDLE
```

So it's a mix with config/mode, state/on and config/preset.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "EKO07259",
  "product": "Elko Smart ZB Thermostat 16 A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
[... 286 lines cut ...]
          "max": 300,
          "change": "0x00000032"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 5,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

With this DDF you will have the config/locked and the state/on.
I m using legacy code for config/mode (it use the attribute 0x001C), so if it work you need to have in config/mode heat, off or auto.

gladhorn, 2023-09-28:
Awesome, this seems to work, I'll test with it a bit :) I now realise my problem, I had the cl and at mixed up for the on state in my experiments. Very nice!

gladhorn, 2023-09-29:
This is really great, I can see at what time in the night heating turns on and can make better decisions for when to change target temp etc. Thanks!

Smanar, 2023-09-29:
Tell me if you want some more changes.
There is so much thing to do with this device, but not sure all is usefull.

You can do the PR to add the DDF if you want, or ask for I make it if it's finished for you.

gladhorn, 2023-09-29:
Sure, I think it's best to get it in and it's always easier to iterate on top later on, when bugs are identified. I can do the pull request, although I feel you did all the work :)

Smanar, 2023-09-29:
It take more time at you to test DDF than me to write it ^^.

closed 2023-10-29 by github-actions[bot]