dresden-elektronik/deconz-rest-plugin #7226: Hue Secure Contact Sensor

ebaauw, 2023-09-15:
## Device

- Product name: Hue Secure Contact Sensior
- Manufacturer: Signify Netherlands B.V.
- Model identifier: SOC001
- Device type : On/Off Sensor

Device pairs with deCONZ without any issue, but doesn't send anything.  I suppose it needs to be configured first.

It also pairs with the Hue bridge without any issue, but it doesn't show in the v1 API.  That's unexpected, since they've had a CLIPOpenClose sensor since forever.  I would have expected a ZLLOpenClose sensor, but then again, I've been expecting this for years.  The v2 API reports:
```
        {
            "id": "3e7448d6-f769-43f8-8ff1-02298715a95d",
            "product_data": {
                "model_id": "SOC001",
                "manufacturer_name": "Signify Netherlands B.V.",
                "product_name": "Hue secure contact sensor",
                "product_archetype": "unknown_archetype",
                "certified": true,
                "software_version": "2.67.1",
                "hardware_platform_type": "100b-125"
            },
            "metadata": {
                "name": "Contact Sensor",
                "archetype": "unknown_archetype"
            },
            "identify": {},
            "services": [
                {
                    "rid": "cef1f4fb-dbea-40ca-b008-56346260cc0c",
                    "rtype": "zigbee_connectivity"
                },
                {
                    "rid": "92682235-d62c-4545-bf98-c3456ec64a10",
                    "rtype": "contact"
                },
                {
                    "rid": "8ed84318-09d7-4039-9bc0-d20d14ba9cdf",
                    "rtype": "tamper"
                },
                {
                    "rid": "c7dbe4ea-bd59-4d53-a3b5-3b45d120b540",
                    "rtype": "device_power"
                }
            ],
            "type": "device"
        },
        {
            "id": "cef1f4fb-dbea-40ca-b008-56346260cc0c",
            "owner": {
                "rid": "3e7448d6-f769-43f8-8ff1-02298715a95d",
                "rtype": "device"
            },
            "status": "connected",
            "mac_address": "00:17:88:01:0d:81:ab:4c",
            "type": "zigbee_connectivity"
        },
        {
            "id": "92682235-d62c-4545-bf98-c3456ec64a10",
            "owner": {
                "rid": "3e7448d6-f769-43f8-8ff1-02298715a95d",
                "rtype": "device"
            },
            "enabled": true,
            "contact_report": {
                "changed": "2023-09-15T11:21:16.240Z",
                "state": "contact"
            },
            "type": "contact"
        },
        {
            "id": "8ed84318-09d7-4039-9bc0-d20d14ba9cdf",
            "owner": {
                "rid": "3e7448d6-f769-43f8-8ff1-02298715a95d",
                "rtype": "device"
            },
            "tamper_reports": [
                {
                    "changed": "2023-09-15T11:48:01.826Z",
                    "source": "battery_door",
                    "state": "tampered"
                }
            ],
            "type": "tamper"
        },
        {
            "id": "c7dbe4ea-bd59-4d53-a3b5-3b45d120b540",
            "owner": {
                "rid": "3e7448d6-f769-43f8-8ff1-02298715a95d",
                "rtype": "device"
            },
            "power_state": {
                "battery_state": "normal",
                "battery_level": 100
            },
            "type": "device_power"
        }
```
## Screenshots
[image]

So this is interesting, it identifies as a controller (_On/Off Sensor_) rather than a proper sensor.  I would expect it to send _On_ and _Off_ commands from the client _On/Off_ cluster, which would suggest the sensor is actually stateless (i.e. it reports state changes, "opening" and "closing", but not the current state, "open" or "close").  I suppose/hope the (server) 0xFC06 cluster does report current state.

The Hue app reports _Opened_ and _Closed_.  I tried to change the state while the Hue bridge was powered down, to check whether it would report current state, but the sensor reported _Unreachable_ when the bridge was restarted.  I had to change the state for the bridge to find the sensor again.  I suppose we need some serious sniffing here.

[image]

No _Receiver On When Idle_, but it is a light sleeper - it does react to _Idenfity_.

### Basic
[image]
[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

ebaauw, 2023-09-15:
The sensor does send _Report Attributes_ from 0xFC06 to the Hue bridge:
```
ZigBee Application Support Layer Data, Dst Endpt: 65, Src Endpt: 2
    Frame Control Field: Data (0x00)
    Destination Endpoint: 65
    Cluster: Manufacturer Specific (0xfc06)
    Profile: Home Automation (0x0104)
    Source Endpoint: 2
    Counter: 72
ZigBee Cluster Library Frame, Mfr: Philips (0x100b), Command: Report Attributes, Seq: 216
    Frame Control Field: Profile-wide (0x0c)
    Manufacturer Code: Philips (0x100b)
    Sequence Number: 216
    Command: Report Attributes (0x0a)
    Attribute Field, Enum8: 1
        Attribute: 0x0100
        Data Type: 8-Bit Enumeration (0x30)
        Uint8: 1 (0x01)
    Attribute Field, Uint32: 0
        Attribute: 0x0101
        Data Type: 32-Bit Unsigned Integer (0x23)
        Uint32: 0 (0x00000000)
    Attribute Field, Enum8: 0
        Attribute: 0x0102
        Data Type: 8-Bit Enumeration (0x30)
        Uint8: 0 (0x00)
    Attribute Field, Uint32: 12425
        Attribute: 0x0103
        Data Type: 32-Bit Unsigned Integer (0x23)
        Uint32: 12425 (0x00003089)
```
The attributes are reverse engineered simply enough:
[image]

The _Last ... Changed_ times are in 0.1s.

And after setting up binding and attribute reporting, the sensor also sends reports to deCONZ.

closed 2023-09-16 by ebaauw

mguaylam, 2025-04-06:
How's the battery on this device so far @ebaauw ? I wonder how do you guys configure reporting for it in Deconz. On ZHA I suddenly see a sharp decrease for some reason, trying to understand if it's something specific on ZHA side.

ebaauw, 2025-04-06:
I wouldn't know about the battery usage; I haven't installed this device in my home.  It's been a while since we added support for this device and I'm not sure if I copied the report configuration from the Hue bridge.  I need to check.

> if it's something specific on ZHA side.

Make sure you're not polling the device.  When the device initiates the communication with the gateway, the type of gateways shouldn't impact the rate of the battery drain.

It could of course always be something in your network: the signal quality to the parent router; interference, congestion, and other conditions causing resends, etc.  And, of course, the type of battery.  Different device types like different battery types.