dresden-elektronik/deconz-rest-plugin #7231: Device name

Idaho947, 2023-09-16:
## Device

- Product name: Nous door\windows sensor E3
- Manufacturer: _TZ3000_v7chgqso
- Model identifier: TS0203
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots
[image]

### Basic
[image]

[image]

### Identify
[image]

### Groups
[image]

[image]

### Scenes
[image]

[image]

### Power
[image]

### IAS zone
[image]

Smanar, 2023-09-17:
Hello, I don't find information for this sensor, but probably a new clone
I have just edited the devices\tuya\_TZ3000_TS0203_door_sensor.json

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_rgchmad8", "_TZ3000_bzxlofth","_TZ3000_au1rjicn", "_TZ3000_4ugnzsli", "_TZ3000_decxrtwa", "_TZ3000_2mbfxlzr", "_TZ3000_6zvw8ham", "_TZ3000_v7chgqso"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203","TS0203","TS0203","TS0203", "TS0203", "TS0203", "TS0203"],
  "vendor": "Tuya",
  "product": "Smart Zigbee Door Window Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Idaho947, 2023-09-23:
Perfect as usual.

Smanar, 2023-09-25:
@Idaho947 so all is working ? I can make the PR ?

Idaho947, 2023-09-25:
Yes youncan

Smanar, 2023-09-25:
Thx, PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7260

closed 2023-09-28 by manup